lumapu/ahoy #981: Don´t forget to enable Radio (NRF24L01+) after update to 0.7.1

rmayergfx, 2023-06-08:
Nach dem Update von Release 0.6.15 auf 0.7.1 meldete mir das System nach dem Update das kein NRF24L01+ verfügbar sei. Hat kurzzeitig zuer erhöhten Ausschüttung von Adrenalin gesorgt. In den Settings war dann der Grund jedoch auch gleich sichtbar:
[image]
@lumapu 
Bitte die Firmware so umstellen, das hier das WLAN Modul per Default auf on steht, d.h. der Haken gesetzt ist.

lumapu, 2023-06-08:
kann ich nur für den ESP8266 machen. Das Problem entsteht nur bei upgrades von alten Versionen, leider wird hier automatisch ein `false` gelesen. Werde heute noch `0.7.2` veröffentlichen, was dieses Problem für den ESP8266 behebt.

rmayergfx, 2023-06-09:
> kann ich nur für den ESP8266 machen. Das Problem entsteht nur bei upgrades von alten Versionen, leider wird hier automatisch ein `false` gelesen. Werde heute noch `0.7.2` veröffentlichen, was dieses Problem für den ESP8266 behebt.

Dann sollten wir diese Info irgendwo ganz oben anpinnen damit jeder der ein ESP32 updated danach manuell wieder aktiviert. Wenn man es nicht weiß ist der Schockmoment erstmal groß. Daher wäre eine Info praktisch, dürfte ja dann alle Versionsupdates von 0.6.x auf 0.7.1 und höher treffen, nehme an die Funktion wurde erst ab 0.7.x implementiert. Warum eigentlich?

lumapu, 2023-06-09:
stimmt, man könnte eine Warnung anzeigen. Hintergrund sind die neu unterstützten HMS und HMT Wechselrichter. Es gibt inzwischen auf Setups, die kein NRF24 mehr haben, da nur der CMT2300A für die neueste Generation verwendet wird.

closed 2023-08-05 by lumapu