lumapu/ahoy #982: Prevent sending not updated data over MQTT - Die Zweite !

knickohr, 2023-06-09:
Ich dachte wir haben das „Rauskotzen“ aller Werte mal mit Issue #826 schon mal gelöst ?  Nun scheint es aber das es wieder da ist 😵

Ist mir heute aufgefallen als ich mal zufällig den Datenstroms angeschaut habe. Bin fast vom Hocker gefallen als ich diesen Datenwulst gesehen habe,

[image]

Bitte wieder reparieren und auch nur Daten senden die upgedated wurden,, keine alten Daten wiederholen.

lumapu, 2023-06-09:
der Einfachheit halber ist dieses Phänomen wohl geschuldet. Aktuell wird nur das angefragte Kommando an MqTT übergeben - dieses 'loopt' über alle Inverter und gibt deren Werte aus. Ich schaue mal was sich hier optimieren lässt.

lumapu, 2023-06-09:
biite erneut testen, ich glaube es 'erwischt' zu haben

knickohr, 2023-06-10:
Geht nicht, der Build ging in die Hose 😲 Außerdem kann ich erst testen wenn es auch wieder mit dem S3 Fusion klappt 😇

knickohr, 2023-06-10:
Hab extra wieder das gute alte „DukeWolle“ Board raus gekramt 😉

Ja, jetzt klappt’s wieder mit der Nachbarin 😎

[image]

Interessanter Nebeneffekt :  Mein Node-Red Dashboard ist jetzt auch wieder flüssiger. Klar, wenn es nicht immer von Daten überschüttet wird 👍

lumapu, 2023-06-10:
welcher issue beschreibt das Problem mit s3 fusion?

knickohr, 2023-06-10:
Es gibt kein Issue, aber offenbar gehört PR #961 dazu.

knickohr, 2023-06-11:
Kann man eigentlich zu machen, das Ding ist durch 👍

closed 2023-06-11 by knickohr