lumapu/ahoy #983: [Bug] Network: not connection if the SSID is hidden, data is manually entered.

mmattel, 2023-06-09:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.6.9 (stable) and 0.7.2 (dev)

### Github Hash

abc

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Not relevant as it is a Network / SSID issue.

### Debug Serial Log output

```shell
scanning APs with SSID xyz
reconnect in 3 seconds
reconnect in 2 seconds
reconnect in 1 seconds
scanning APs with SSID xyz
...
```

### Error description

The SW runs fine, I get data from the HM-800, BUT

When setting up the network, it will not connect to it. Using both versions which show the same behaviour, I found out the following:

* When the SSID is open (not hidden), you can scan and select it + pwd or manually add the data and the connection is set up. I tested this via my iPhone via providing a hotspot and got a connection.
* When manually defining the SSID + pwd because the network is hidden, no connection can be established. This is also shown in the console.

lumapu, 2023-06-09:
can you try the latest version which was uploaded some seconds ago

mmattel, 2023-06-10:
I did, this comes from the log:
```
try to connect to AP with BSSID: b4 14 89 83 5b 90
reconnect in 19 seconds
reconnect in 18 seconds
I: connectionEvent
reconnect in 17 seconds
reconnect in 16 seconds
I: connectionEvent
reconnect in 15 seconds
reconnect in 14 seconds
I: connectionEvent
reconnect in 13 seconds
reconnect in 12 seconds
reconnect in 11 seconds
I: connectionEvent
reconnect in 10 seconds
reconnect in 9 seconds
I: connectionEvent
reconnect in 8 seconds
reconnect in 7 seconds
I: connectionEvent
reconnect in 6 seconds
reconnect in 5 seconds
I: connectionEvent
reconnect in 4 seconds
reconnect in 3 seconds
reconnect in 2 seconds
I: connectionEvent
reconnect in 1 seconds
reconnect in 0 seconds
try to connect to AP with BSSID: f0 51 36 42 87 13
reconnect in 19 seconds
```

mmattel, 2023-06-10:
Note that the following:
* The WLAN pwd used is working because I fully removed another client from the network and added it back manually typing all the required data. (just to be sure not to mistype the pwd...)
* The *first* MAC addresses printed in BSSID (b4 14 89 83 5b 90) is the correct one of the access point. The others printed like (f0 51 36 42 87 13) are not.

From my Windows notebook, maybe this holds additional valuable info:
```
netsh wlan show interfaces

There is 1 interface on the system:

    Name                   : WLAN
    Description            : Intel(R) Dual Band Wireless-AC 8265
    GUID                   : 2a0fb744-cfc2-4a23-b939-cd449f74b344
    Physical address       : 48:f1:7f:74:1f:8f
    State                  : connected
    SSID                   : xyz
    BSSID                  : b4:14:89:83:5b:90
    Network type           : Infrastructure
    Radio type             : 802.11n
    Authentication         : WPA2-Personal
    Cipher                 : CCMP
    Connection mode        : Auto Connect
    Channel                : 13
    Receive rate (Mbps)    : 144.4
    Transmit rate (Mbps)   : 144.4
    Signal                 : 97%
    Profile                : xyz
```

lumapu, 2023-06-15:
so I've to dig deeper into this. First I need to setup a hidden WiFi on my side 😅. I hope to fix this soon.

stefan123t, 2024-10-27:
@lumapu this may have to do with the 802.11 k/r/v Standards. You know I like the `s` in Standards :wink: 

Here are the links to the documentation for roaming 802.11k and management 802.11v collectively known as "Mesh Steering"
https://github.com/tbnobody/OpenDTU/issues/2202#issuecomment-2289412118

Actually both 802.11k and 802.11r are necessary for seamless Basic Service Set (BSS) transitions aka roaming:

802.11k Basic Service Set (BSS) transitions: [IEEE 802.11k-2008](https://en.wikipedia.org/wiki/IEEE_802.11k-2008)
802.11r Fast BSS Transition (FT): [IEEE 802.11r-2008 or fast BSS transition (FT)](https://en.wikipedia.org/wiki/IEEE_802.11r-2008)
802.11v Wireless Network Management: [IEEE 802.11v-2011](https://en.wikipedia.org/wiki/IEEE_802.11v-2011)