lumapu/ahoy #987: [Bug] MQTT sendet kein P_AC wenn Inverter im Status 1

knickohr, 2023-06-11:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.7.3

### Github Hash

bdf43fd

### Build & Flash Method

ESP Tools (flash)

### Setup

MQTT, Nighttime, Display enabled.

### Error description

Wir haben uns mit dem Fixen von Issue #980 bzw. #927 offenbar ein neues Problem eingefangen :

MQTT sendet über total/P_AC keine Werte mehr wenn ein Inverter in den Status <INVERTER_NAME_FROM_SETUP>/available = 1 wechselt.

Auf der WebGuI und im Display werden weiterhin die Werte ausgegeben, aber über MQTT bleibt der letzte Wert hängen als noch alle Inverter available = 2 waren.

(Siehe Fotos)

[image]

[image]

[image]

teletapi, 2023-06-11:
Das ist mir tatsächlich heute Abend ebenfalls aufgefallen.  Ich war aber am überlegen das bis morgen abend noch zu beobachten

knickohr, 2023-06-12:
Siehe dazu auch Issue #929 , ähnliches Phänomen 😲

knickohr, 2023-06-13:
Bis 0.6.15 war es noch OK. Ich würde sagen sogar bis 0.7.2.

Development Changes

0.7.3 - 2023-06-09

fix hidden SSID scan #983
improved NRF24 missing message on home screen #981
fix MqTT publishing only updated values #982
0.7.2 - 2023-06-08

fix HMS-800 and HMS-1000 assignments #981
make nrf enabled all the time for ESP8266
fix menu item active highlight for 'API' and 'Doku'
fix MqTT totals issue #927, #980
reduce maximum number of inverters to 4 for ESP8266, increase to 16 for ESP32
0.7.1 - 2023-06-05

enabled power limit control for HMS / HMT devices
changed NRF24 lib version back to 1.4.5 because of compile problems for EPS8266
0.7.0 - 2023-06-04

HMS / HMT support for ESP32 devices

lumapu, 2023-06-13:
ich glaube der Fehler liegt einfach an der ID Abfrage, neu ist der Teil mit `mSendTotals = false`. Die Abfrage dafür müsste lauten `isAvail()`

https://github.com/lumapu/ahoy/blob/development03/src/publisher/pubMqttIvData.h#L122-L126

knickohr, 2023-06-14:
Ahhh, verstehe 😲 Dann ist das erst durch Issue #982 aufgekommen. Klar, bis dahin wurden ja alle Daten raus gedonnert bis auch der letzte WR offline war.

Schnell fixen 😉

lumapu, 2023-06-15:
oh da stand 'schnell fixen' - bin jetzt dran

knickohr, 2023-06-15:
Noch nicht fertig ? 🤪

knickohr, 2023-06-16:
Teilweise gefixed 😉
 YieldDay bleibt immer noch hängen u d ich schätze YieldTotal auch. Wahrscheinlich das gleiche Problem.

knickohr, 2023-06-18:
Jepp, jetzt kommt auch das YieldDay und YieldTotal wieder rüber 👍

Nur leider rechnet er die Summe noch falsch zusammen ☹️ Siehe Issue #927 . Er summiert nur die der noch online Inverter. Komischerweise macht es das WebGUI aber richtig 🫣

knickohr, 2023-06-18:
Ich glaube wir machen hier zu, der P_AC funktioniert ja, alles weitere in den anderen Issues 😉

closed 2023-06-18 by knickohr