lumapu/ahoy #989: hms-2000 no communication

mirek-z, 2023-06-12:
Hello
I have a problem with connecting to the hms-2000 inverter in the system tab I have this data ...
radio nrt24l01+ is connected 
data rate: 250 kbps
power level: max
and then only 0
nothing is uploaded or downloaded.

gmed-ed, 2023-06-12:
if it's 8266-based i would set power level to low.
the 3.3Volts aren't strong enough to handle wifi+max-level radio.

Even esp32 may struggle.

If one really need radio-level up at max, you should setup a more stable 3.3V source separattly.

Ollipop030, 2023-06-13:
Correct me if i´m wrong, but are the HMS/HMT Inverters not only supportet by ESP32 with CMT2300A? Or do I miss here something?

gmed-ed, 2023-06-13:
You're right
i didn't read correctly

mirek-z, 2023-06-15:
fact my mistake I read in the manual that it should be esp32. I will start experimenting but I have a question about this module with cmt2300a, do you have any checked?

lumapu, 2023-06-15:
what do you mean by 'do you have any checked?'.
We recommend the part from ebyte `E49 900M20S`:
https://github.com/lumapu/ahoy/wiki/RF-Module-CMT2300A---E49-900M20S
If you need a PCB to assemble this module let me know (contact on https://ahoydtu.de)

closed 2023-08-05 by lumapu