lumapu/ahoy #991: [Bug] kein Datenaustausch mehr möglich seit neueren Versionen: Datenrate schaltet auf 2 MBps, nachdem mit 250 kBps gestartet wurde

ralfsfb, 2023-06-14:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.6.9

### Github Hash

Webinstaller

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

 siehe beigefügte Sceenshots

### Debug Serial Log output

```shell
.
```

### Error description

In früheren  Versionen konnten ganz normal Daten mit den Wechselrichtern ausgetauscht werden, seit den neueren Versionen wird kurz nach dem Neustart eine Datenrate von 250 kBps angezeigt und wechselt dann nach 2 MBps, wodurch kein Datenaustausch mehr möglich ist. Um Fehler in der Konfiguration auszuschließen, habe ich den Webinstaller verwendet.
Was kann ich noch versuchen, um auch aktuelle Versionen nutzen zu können?
Besten Dank für eure Ideen im Voraus sendet
ralf

fsck-block, 2023-06-14:
Ich habe bei mir auch diesen Effekt.
Nach dem ersten Senden von Daten wechselt die Datenrate von 250 kbps auf 2M bps.
Um genauer zu sein in `HmRadio::sendPacket` nach aufruf von ` mNrf24.stopListening();` wechselt die Datenrate auf 2M bps.
Habe schon unterschiedliche esp8266 Module probiert. Allerdings am gleichen nrf24l01+ Modul

rmayergfx, 2023-06-15:
Kann auch am NRF24l01+ Modul liegen.

ralfsfb, 2023-06-16:
> Kann auch am NRF24l01+ Modul liegen.

Ist natürlich möglich, doch bei früheren Versionen hat die selbe Hardware gut funktioniert und schaltet inzwischen immer wieder auf 2 Mbps

rmayergfx, 2023-06-17:
Es soll durchaus auch vorkommen das ein Modul defekt geht und zufälligerweise genau mit dem Update. Das hat nichts mit dem Update an sich zu tun, das ist Murphy. Daher würde ich erst einmal ein anderes Modul ausprobieren und genau prüfen das die Verkabelung passt und keinen Wackler hat. Elko ist ja schon verbaut, vielleicht auch noch mal ein anderes USB Kabel und Netzteil ausprobieren, nicht das es noch Probleme mit der Stromversorgung generell gibt.

fsck-block, 2023-09-16:
Ich habe mal mein NRF24l01+ Modul getauscht und der Fehler war weg.
Lag' wohl doch an am Modul

lumapu, 2023-09-16:
jetzt haben wir noch einen Issue mit gleichem Problem #1149