lumapu/ahoy #992: Wemos D1 Mini ESP32 und NRF verbinden

dtuuser, 2023-06-14:
Hallo zusammen,

habe mir zum Testen einen Wemos D1 Mini ESP32 geholt. Leider ein Mini ESP32. Finde keine Verdrahtung dazu. 
Kann hier vielleicht jemand helfen?

Gruß

dknubben, 2023-06-14:
Wo liegt Dein Problem?
Die Verdrahung ist doch weitgehend frei wählbar. Ich habe einen Wemos D1 Mini ESP32 wie im Bild angeschlossen.
Das 2. Bild zeigt das Pinout des Moduls.
[image]
[image]

rmayergfx, 2023-06-15:
> Hallo zusammen,
> 
> habe mir zum Testen einen Wemos D1 Mini ESP32 geholt. Leider ein Mini ESP32. Finde keine Verdrahtung dazu. Kann hier vielleicht jemand helfen?
> 
> Gruß

Was heist hier leider? Ein ESP32 ist doch besser als ein ESP8266 was die Leistung betrifft.
https://smarthome-blogger.de/blog/esp32-projekte/esp32-vs-esp8266

Preislich ist der Unterschied jetzt auch nicht so groß.

Was ich aber leider immer noch nicht verstehen kann ist warum man in Gottes Namen sich nicht einfach an die Empfehlungen des Entwicklers hält und bereits getestete und funktionierende Boards kauft. Spart Zeit und Geld und erleichtert den Support!
Da teilt einem der Entwickler schon mit was man kaufen kann https://ahoydtu.de/getting_started/ und dann kommen immer wieder Leute um die Ecke die Unterstützung für ein anderes Board haben möchten.  Warum kauft man sich so etwas wenn man nicht weiß wie das ganze funktioniert?

dtuuser, 2023-06-15:
@dknubben 
Danke für die Konfig. Hat nach der Verdrahtung auf Anhieb geklappt. Topp

@rmayergfx
Ich nutze den USB Ausgang eines Accesspoints mit einem Y Kabel für eine USB Kamera und einen Wemos. Bei der USB Buchse vom Wemos habe ich nur Masse und die 5 Volt angeschlossen.
Diesen Wemos tausche ich gegen den ESP32 aus. Ich habe mich für den ESP32 Mini entschieden, da er angeblich weniger Strom als ein normaler ESP32 verbraucht.
Und ich jedes mA für die Kamera benötige.

closed 2023-06-15 by dtuuser