lumapu/ahoy #993: /api/record/live not showing every inverter although /api/inverter/list does

fummmmel, 2023-06-15:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

nothing

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.6.9

### Github Hash

15ec6a0

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

MQTT installed
Display installed
Added 4 Inverters
Sunrise&Sunset enabled
Inverter Poll interval 5s
Max retry 0

### Error description

Hi guys,

I was trying to get the HoyMilesZeroExport Script up and running which relies on AHOY-DTU API Data.
The Script is failing because it is missing the fourth inverter under /api/record/live:
[image]
The inverter is in the inverter list under /api/inverter/list 
[image]
All four Inverters are availlable in the UI and on MQTT. 

Any idea why it is missing on the live endpoint?
Thank you!

fummmmel, 2023-06-15:
[ahoy.txt](https://github.com/lumapu/ahoy/files/11757254/ahoy.txt)
This is my config, I xxxed names and values.

lumapu, 2023-06-15:
I think the problem is the entire size of the payload. In my case with 2 HMS inverters the payload is `3074` bytes long. Due to  limited memory ressources it is not possible to allocate that much memory. 
For the same reason I refactored some time ago the live view. First the endpoint `/api/live` is read and the boolean array `[iv]` is checked for enabled inverters (`true`). In a second stage an iteration over all enabled inverters occurs and the endpoint `/api/inverter/id/0` where the last digit is the id of the inverter - in your case `0 .. 1`

reserve85, 2023-06-16:
Thank you, i´ll update the script to fit this. Thanks for clarify.

fummmmel, 2023-06-16:
Thank you @reserve85, it's working fine now. All Inverters are found and can be controlled by the script:
[image]

lumapu, 2023-06-16:
it should now scale to infinity number of inverters 😄
Really cool that you adjusted that fast

reserve85, 2023-06-16:
@lumapu i would suggest to remove the endpoint /api/record/live in the upcoming versions so everyone have to implement the proofed api and will not run into the same problem. Just my opinion. Thank you for the fast help.

ncqgm, 2023-06-18:
maybe return "deprecated" from this endpoint for a while ... ;-)

closed 2023-08-05 by lumapu