lumapu/ahoy #994: esp8266-release Build schlägt fehl

rbutenuth, 2023-06-16:
## Hardware
  * [x] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [x] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [x] circuit board
* [ ] external antenna (SMA)

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [x] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: main
Github Hash: https://github.com/lumapu/ahoy/commit/dbb352801e9076f50679efd0a7f432b9fa34dbd4

### Build & Flash Method:
* [ ] AhoyDTU Webinstaller
* [x] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

Hallo!

Ich versuche gerade, das Projekt in Platform IO zu übersetzen, scheitere aber. Muss man irgendwelche besonderen Einstellungen in Platform IO machen? 

Meine Schritte:
* clone von github (main)
* In Code File -> Open Workspace from File
* PlatformIO -> project tasks -> esp8266-release -> general -> build

Das gibt dann die untenstehende Fehlermeldung. Die esp32-Projekte lassen sich übersetzen.

Meine Umgebung: Windows 10, aktuelles VSCode (1.79.1), aktuelles Platform IO (IDE: 3.2.0, Core: 6.1.7, Home: 3.4.4)

Hier die Fehlermeldungen:

Executing task in folder src: C:\Users\butenuth\.platformio\penv\Scripts\platformio.exe run --environment esp8266-release 

Processing esp8266-release (platform: espressif8266; board: esp12e; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
Firmware Revision: gdbb3528
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp12e.html
PLATFORM: Espressif 8266 (4.2.0) > Espressif ESP8266 ESP-12E
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif8266 @ 3.30102.0 (3.1.2)       
 - tool-esptool @ 1.413.0 (4.13)
 - tool-esptoolpy @ 1.30000.201119 (3.0.0)
 - toolchain-xtensa @ 2.100300.220621 (10.3.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 48 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ESP Async WebServer @ 1.2.3+sha.1dde9cf
|-- RF24 @ 1.4.7
|-- Time @ 1.6.1
|-- espMqttClient @ 1.4.2+sha.06c150c
|-- ArduinoJson @ 6.21.2
|-- Timezone @ 1.2.4+sha.7cf1425
|-- U8g2 @ 2.34.18
|-- GxEPD2 @ 1.5.2
|-- ESP8266WiFi @ 1.0
|-- Adafruit GFX Library @ 1.11.5
|-- SPI @ 1.0
|-- DNSServer @ 1.1.1
|-- LittleFS @ 0.1.0
|-- ESPAsyncTCP @ 1.2.2
|-- Adafruit BusIO @ 1.14.1
|-- Hash @ 1.0
|-- Wire @ 1.0
Building in release mode
Compiling .pio\build\esp8266-release\src\app.cpp.o
Compiling .pio\build\esp8266-release\src\main.cpp.o
In file included from .pio\libdeps\esp8266-release\RF24/RF24.h:18,
                 from app.h:11,
                 from app.cpp:6:
.pio\libdeps\esp8266-release\RF24/RF24_config.h:186:26: error: expected ';' at end of member declaration
  186 |         #define printf_P Serial.printf
      |                          ^~~~~~
.pio\libdeps\esp8266-release\ESP Async WebServer\src/AsyncWebSocket.h:170:12: note: in expansion of macro 'printf_P'
  170 |     size_t printf_P(PGM_P formatP, ...)  __attribute__ ((format (printf, 2, 3)));
      |            ^~~~~~~~
.pio\libdeps\esp8266-release\RF24/RF24_config.h:186:32: error: expected unqualified-id before '.' token
  186 |         #define printf_P Serial.printf
      |                                ^

lumapu, 2023-06-16:
ja das liegt leider an der NRF24 library. 
Ändere für den `main`-branch folgendes lokal bei dir oder baue den `development03`-branch:
https://github.com/lumapu/ahoy/blob/fd330995224899639ed972704bfa6022b89c3943/src/platformio.ini#L27

rbutenuth, 2023-06-21:
Danke für die schnelle Antwort, das war's tatsächlich.
Wäre es nicht besser, im main-Branch die Version von NRF24 fest auf 1.4.5 zu setzen, bis das Problem dort behoben ist?
Aus Zeitmangel bisher nicht gelötet/installiert.

closed 2023-08-05 by lumapu