lumapu/ahoy #995: OTA Update Failed (Ahoy 05.66 -> 06.9)

georg-x, 2023-06-16:
## Hardware
  * [x] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [ ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [x] circuit board
* [ ] external antenna (SMA)

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: _._.__
Github Hash: _______

### Build & Flash Method:
* [x] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)
----
I only get this message:
----
Update: failed

rebooting ... auto reload after 20s
----

How do I find out what's wrong?

File used:
230419_ahoy_0.6.9_15ec6a0_esp8266.bin

Previous Upgrades OTA worked.

rmayergfx, 2023-06-16:
Don´t waste your time with debugging the update, do a clean install. This is faster and you´ll get the matching version. Take a screenshot of your settings before factory reset (pinout and other changes you made, also useful for the future).

knickohr, 2023-06-16:
Turn Off inverters and MQTT. In some old versions, the OTA is struggling.

lumapu, 2023-06-16:
which version do you have installed?
In general it's possible to install a newer version using our online installer (https://ahoydtu.de/web_install) without loosing settings. I really like the sugestion of @rmayergfx to have a seperate backup of your settings somewhere else.

georg-x, 2023-06-16:
I'm still a little buffeled ... I like the idea of the online installer, but didn't get it to work anymore. Maybe since Windows 11, but I really don't know. It always says, no serial device found.
I've updated multiple times via the Webinterface until 05.66. By now Update just says failed for any newer version.
I don't care much about settings. Entering the data is not a big deal for me. Already tried a factory reset. Still simply no update.
How could I see what's wrong?

georg-x, 2023-06-16:
Sorry for my questions, another try. Webinstaller works now. I think has to do with different USB cables...

closed 2023-06-16 by georg-x