lumapu/ahoy #997: Question: Looking for a PCB for Wemos D1 mini, nRF24L01+, Nokia display

TomTom101, 2023-06-18:
## Hardware
  * [x] ESP8266

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [x] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode

### Antenna:
* [x] circuit board

### Power Stabilization:
* [x]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module

Ordering a single PCB board to be etched is either costly or when ordered in China takes forever and is hardly economical.

Does anyone have a tip where to get one for max €10 incl. shipping? 

Alternatively, would we find enough users to make a bulk order?

Peertje62, 2023-06-18:
Easy,
https://shop.blinkyparts.com/en/OpenDTU-Your-evaluation-for-your-balcony-solar-system/blink237542

Regards

CBFOUR, 2023-06-18:
Jepp, > German Area
https://www.makershop.de/?s=esp&post_type=product
Cheers

lumapu, 2023-07-03:
maybe you should ask in our Discord channel. Otherwise it's not that hard to use a breadboard for that.

stefan123t, 2024-10-27:
@TomTom101 is this still relevant or can we close this issue ?

closed 2024-10-28 by TomTom101