lumapu/ahoy #998: wiring cmt2300a esp32 devkit

001oc, 2023-06-19:
## Hardware
  * [ ] ESP8266
  * [x] ESP32
  * [ ] Raspberry Pi

Modelname: ESP32 DevKit
Retailer URL: ______

### nRF24L01+ Module
* [ ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [x] external antenna (SMA)

### Power Stabilization:
* [x]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: _._.__
Github Hash: _______

### Build & Flash Method:
* [ ] AhoyDTU Webinstaller
* [x] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

I am looking for a wiring diagram to connect a cmt2300a to a esp32 devkit. I want to connect solely a cmt21300a (no nrf24l01+). Connecting the csb and fscb is straight forward but where is irq connected to. As the systems settings have solely 3 input fields (csb, fcsb, irq) where to connect the sclk ping of the cmt2300a.

Thanks

001oc, 2023-06-19:
maybe that is important too. I have used the self compiled current hms branch.

lumapu, 2023-07-03:
that branch is not updated any more, please use `development03`. I will delete the `hms` branch in near future

001oc, 2023-07-04:
@lumapu - thanks for your advice, I have moved to develeopment branch shortly after I having started my question. But this is not an answer to my question. I am looking for a wiring diagram for solely having a CMT2300A connected to the ESP - idealy I would like to have an advice how to config this inside the software. Thanks

mirek-z, 2023-07-04:
Hello
I join the question about esp32+cmt2300a configuration

lumapu, 2023-07-04:
Have you checked the issue #904?

001oc, 2023-07-04:
Yes, I checked that before, thanks. The question there is if and how to operate nrf24l01 and cmt2300a both together parallel - which is working at all. My question is how to run solely a cmt2300a without a nrf24l01. I dont habe any HM, I only have HMS and HMT running.

lumapu, 2023-07-04:
I did a quote on the interesting part, the pinning. Ahoy will support both generations, it depends on your settings which module will be used.

@mirek-z do you need a translation of my post before?

mirek-z, 2023-07-04:
definitely yes, because the soft has CSB, FCSB and IRQ settings and CMT has SCLK SDIO CSB FCSB, how to set it in the soft?

jmeister79, 2023-07-05:
Same Problem here.
Wired up CMT to an ESP32 and cannot get it working.
What does IRQ mean? 
The Boards are wired as follows:

CMT:
GPIO12 - SCLK
GPIO14 - SDIO
GPIO27 - CSB
GPIO26 - FCSB

nrf:

GPIO4 - CE
GPIO5 - CS
GPIO18 - SCK
GPIO23 - MOSI
GPIO19 - MISO
GPIO16 - IRQ

However, I cannot bring the NRF to work: "your NRF24 module can't be reached, check the wiring, pinout and enable"

CMT says:
15:34:05 I: TX 865.00Mhz \| 15 80 42 26 55 81 00 17 65 80 0b 00 64 9e d9 cd 00 00 00 00 00 00 00 00 7c c5 8b
5:34:05 W: CMT TX failed, code: 2

And I am not sure what to select for IRQ of the CMT.

Nils

lumapu, 2023-07-05:
for the CMT part I use the following pinout with an ESP32-30 Pin module:
[image]

SDIO has to be connected to D23-VSPI-MOSI and cannot be placed somewhere else. Same for SLCK, it has to be connected to D18-VSPI-SCK

For IRQ (Interrupt Request) you need a interrupt capable pin of you ESP, I use D34 for that.

---

For NRF (optional) you can use the following pins:
D14: SPI-CLK
D12: SPI-MISO
D13: SPI-MOSI

for CS, CE and IRQ you have free choice.

mirek-z, 2023-07-06:
I have it exactly as described, but in the console I have it all the time: 
19:56:43 I: TX 868.00Mhz | 56 81 03 73 23 81 00 17 65 02 15 21 14 14 41
19:56:43 W: CMT TX failed, code: 2
i only have cmt 2300 and esp32 module. I am trying to connect to HMS-2000 4T

mirek-z, 2023-07-06:
and shouldn't the microinverter be disconnected from the dtu? or something? wires well connected and no contact with hms

lumapu, 2023-07-07:
would be nice if you post the serial output from booting up Ahoy

mirek-z, 2023-07-08:
20:15:36 I: enqueued cmd failed/timeout
20:15:36 I: resetPayload: id: 0
20:15:36 I: (#0) Requesting Inv SN xxxxxxxxxxxxx
20:15:36 I: (#0) prepareDevInformCmd
20:15:36 I: TX 865.00Mhz | 15 81 03 73 23 81 00 17 65 80 0b 00 64 a9 a7 c8 00 00 00 00 00 00 00 00 86 7c e7 
20:15:36 W: CMT TX failed, code: 2

lumapu, 2023-07-08:
I meant the log during boot up - once the CMT module was initialized. I think the ESP can't communicate with your radio

mirek-z, 2023-07-09:
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
E (488) esp_core_dump_flash: No core du▒▒partition found!
E (488) esp_core_dump_flash: No core dump partition found!
I: Initializing FS ..
I:  .. done
I: Settings valid: true

---------
         AhoyDTU Info:
Version: 0.6.15
Github Hash: 2c1a32f

---------
         AP MODE
                SSID: AHOY-DTU
PWD: esp_8266
IP Address: http://192.168.4.1
---------

Wifi event: 0
Wifi event: 2
Wifi event: 10
Wifi event: 11
Wifi event: 10
connect to network 'DOM 29' ...
I: added inverter HMSxxxxxxxxxx
I: added inverter HMSyyyyyyyyyy
I: resetPayload: id: 0
I: resetPayload: id: 1
I: resetPayload: id: 2
I: resetPayload: id: 3
I: resetPayload: id: 4
I: resetPayload: id: 5
I: resetPayload: id: 6
I: resetPayload: id: 7
I: resetPayload: id: 8
I: resetPayload: id: 9
scanning APs with SSID DOM 29
reconnect in 8 seconds
reconnect in 7 seconds
reconnect in 6 seconds
I: Iv: 0
I:
I: Iv: 1
I:
Wifi event: 1
reconnect in 5 seconds
BSSID 0: 14 eb b6 af 8 4
reconnect in 1 seconds
reconnect in 0 seconds
try to connect to AP with BSSID: 14 eb b6 af 8 4
reconnect in 19 seconds
reconnect in 18 seconds
I: Iv: 0
I:
I: Iv: 1
I:
reconnect in 17 seconds
Wifi event: 5
I: connectionEvent
Wifi event: 4
I: connectionEvent

[WiFi] Connected
Wifi event: 7
I: connectionEvent

--------------------------------
Welcome to AHOY!

point your browser to http://192.168.1.112 (Station)
to configure your device
--------------------------------

[WiFi] AP disabled
Wifi event: 11
Wifi event: 10
Wifi event: 11
I: [NTP]: 2023-07-09 07:07:03 UTC
I: resetPayload: id: 1
I: (#1) Requesting Inv SN 114483188222
I: TX 868.00Mhz | 56 83 18 82 22 81 00 17 65 02 15 21 14 14 A8
I: resetPayload: id: 0

lumapu, 2023-07-22:
sieht eigentlich gut aus, weiß nicht wo es hakt.

mirek-z, 2023-07-25:
maybe from the beginning.
is this diagram ok?
[image]

lumapu, 2023-07-25:
your pinnumbering is wrong, check https://www.studiopieters.nl/esp32-pinout/
3.3V for the ESP32-38 is pin 2. Then P34 is pin 6 and P26 is pin 11.
The other connections are correct.

DejanBukovec, 2023-08-20:
@jmeister79 did you make it work ? I have ebay board which have same configuration(OpenDTU pinout):
"clk": 12,
"cs": 27,
"fcs": 26,
"sdio": 14,

Im set this setting:
[image]

But can't test it because still waiting inverter... When Im insert some serial of HMS inverter and check console I get this:
```
11:50:37 I: resetPayload: id: 0
11:50:37 I: (#0) prepareDevInformCmd 0x0b
11:50:37 I: TX 865.00Mhz | 15 73 51 13 41 81 00 17 65 80 0b 00 64 e1 e1 ed 00 00 00 00 00 00 00 00 7d 47 ae 
11:50:37 W: CMT TX failed, code: 2
11:50:42 I: resetPayload: id: 0
11:50:42 I: (#0) enqueCommand: 0x01
11:50:42 I: (#0) enqueCommand: 0x0b
11:50:42 I: (#0) prepareDevInformCmd 0x01
11:50:42 I: TX 865.00Mhz | 15 73 51 13 41 81 00 17 65 80 01 00 64 e1 e1 f2 00 00 00 00 00 00 00 00 47 c1 07 
11:50:42 W: CMT TX failed, code: 2
11:50:47 I: resetPayload: id: 0
11:50:47 I: (#0) prepareDevInformCmd 0x0b
11:50:47 I: TX 865.00Mhz | 15 73 51 13 41 81 00 17 65 80 0b 00 64 e1 e1 f7 00 00 00 00 00 00 00 00 1d f4 67 
11:50:47 W: CMT TX failed, code: 2
11:50:52 I: resetPayload: id: 0
11:50:52 I: (#0) enqueCommand: 0x01
11:50:52 I: (#0) enqueCommand: 0x0b
11:50:52 I: (#0) prepareDevInformCmd 0x01
11:50:52 I: TX 865.00Mhz | 15 73 51 13 41 81 00 17 65 80 01 00 64 e1 e1 fc 00 00 00 00 00 00 00 00 27 8d 25 
11:50:52 W: CMT TX failed, code: 2
11:50:57 I: resetPayload: id: 0
11:50:57 I: TX 868.00Mhz | 56 73 51 13 41 81 00 17 65 02 15 21 14 14 e3 
11:50:57 W: CMT TX failed, code: 2
```

jmeister79, 2023-08-20:
I switched to opendtu

erikteelen, 2023-09-03:
Hallo zusammen,
Ich habe das gleiche Problem. Ich bekomme es nicht zum Laufen. Habe bereits nachgeschaut und alles versucht. Erhalte die gleiche Meldung.
[image]
[image]
Ich habe nur HMS-Konverter
ich habe einen esp32wroom 32u und einen e49 900m20s(cmt2300a) presolded en alleen ahoydtu
Alles wurde vom Niederländischen ins Deutsche übersetzt, daher kann es zu Fehlern kommen.

18:20:22 I: resetPayload: id: 0
18:20:22 I: (#0) Requesting Inv SN ------
18:20:22 I: TX 868.00Mhz | 56 90 14 89 93 81 00 17 65 02 15 21 14 14 0d 
18:20:22 W: CMT TX failed, code: 2
18:20:52 I: enqueued cmd failed/timeout
18:20:52 I: resetPayload: id: 1
18:20:52 I: (#1) Requesting Inv SN -----
18:20:52 I: TX 868.00Mhz | 56 90 15 07 01 81 00 17 65 02 15 21 14 14 10 
18:20:52 W: CMT TX failed, code: 2

aditiyagio, 2023-11-23:
I have done this, but the inverter data is not yet available, do I need certain settings again?

I: TX 865.00Mhz | 15 83 76 30 78 81 00 17 65 80 0B 00 65 5E CE BC 00 00 00 00 00 00 00 00 A1 1C 24
I: enqueued cmd failed/timeout
I: resetPayload: id: 0
I: (#0) Requesting Inv SN 114483763078
I: (#0) enqueCommand: 0x01
I: (#0) enqueCommand: 0x0B
I: (#0) prepareDevInformCmd 0x01
I: TX 865.00Mhz | 15 83 76 30 78 81 00 17 65 80 01 00 65 5E CE DA 00 00 00 00 00 00 00 00 09 95 69 
I: enqueued cmd failed/timeout
I: resetPayload: id: 0
I: (#0) Requesting Inv SN 114483763078
I: (#0) prepareDevInformCmd 0x0B
I: TX 865.00Mhz | 15 83 76 30 78 81 00 17 65 80 0B 00 65 5E CE F8 00 00 00 00 00 00 00 00 A2 1F 60 
I: enqueued cmd failed/timeout
I: resetPayload: id: 0
I: (#0) Requesting Inv SN 114483763078
I: (#0) enqueCommand: 0x01
I: (#0) enqueCommand: 0x0B
I: (#0) prepareDevInformCmd 0x01
I: TX 865.00Mhz | 15 83 76 30 78 81 00 17 65 80 01 00 65 5E CF 16 00 00 00 00 00 00 00 00 C9 C1 30

aditiyagio, 2023-11-23:
@lumapu 
I have done this, but the inverter data is not yet available, do I need certain settings again?
what is the correct configuration?

My Trial :
I : enqueued cmd failed/timeout
I : resetPayload: id : 0
I : (#0) Requesting Inv SN 114483763078
I : (#0) enqueCommand: 0x01
I : (#0) enqueCommand: 0x0b
I : (#0) prepareDevInformCmd 0x01
I : TX 865.00Mhz | 15 83 76 30 78 81 00 17 65 80  0b 00 65 5d c8 53 00 00 00 00 00 00 00 00 ce 7b c6

mo0815, 2024-07-05:
i have desolderd the Nrf modul now it workes fine

stefan123t, 2024-10-27:
@001oc @mirek-z do you still have an issue with wiring the CMT2300A connections ?

001oc, 2024-10-27:
As @mo0815 statet after removing the NRF modul it seams to be working - i believe it is a problem with either the poor quality modul or the power supply or both of them. 

I have done a couple dozend of opendtus and from my expieriences its always a problem with either cheap NRF modules or more often poor powersupplies. Most of my problems were solved when using a reliable and proven capable power supply. If that does not helped, the problem was typically a cheap chinese rebuild of an NRF module. Mostly they are functioning in general but have difficulties to establish a reliable radio signal which mostly ends up in (at least) unstable function. After using good quality power supply and spending some bucks more per NRF modul these problems never come up again. When following this simple rules you will see that opendtu is a very stable and reliable project.