lumapu/ahoy #999: Hoymiles HM 1500 keine Verbindung

graueresel, 2023-06-19:
Hallo,
soweit ist alles Super erklärt und habe es genauso gemacht. Bei mir sind keine Live Daten zu sehen und das Limit kann ich auch nicht einstellen. Die Ahoy DTU ist in der Fritzbox zu sehen und ich kann von dort auch den Wechselrichter erreichen. Was mache ich Falsch? Vielleicht kann mir jemand helfen.

 
[image]

[image]

## Hardware
  * [ ] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [ ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [x ] circuit board
* [ ] external antenna (SMA)

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: _._.__
Github Hash: _______

### Build & Flash Method:
* [ ] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

rmayergfx, 2023-06-19:
Solange AhoyDTU keine Verbindung zum Wechselrichter hat kann man natürlich auch nichts einstellen. Wurde die Seriennummer richtig in der Konfiguration eingetragen? Sind die IRQ richtig definiert und angeschlossen. Stehen alle LED auf off? Welche Ahoy Version ist installiert ? Bitte mal Screenshots der Settings posten und die Seriennummer nach der 3ten stelle schwärzen!

graueresel, 2023-06-19:
Die Seriennummer passt, dass habe ich geprüft.

[image]
[image]
[image]
[image]
[image]

graueresel, 2023-06-19:
Ich hoffe das passt soweit von dern Info,

mroenne2022, 2023-06-19:
Meines Erachtens müssen die beiden Schalter für LED0 und LED1 auf off gestellt werden, überschneiden sich ja mit dem IRQ (GPIO0). Dann läuft es sehr warscheinlich.

graueresel, 2023-06-19:
ICh habe mal die LEDs wie beschrieben auf Off gestellt.
Die blaue LED ist nun nicht mehr dauerhaft an und blinkt immer wieder mal.

[image]

graueresel, 2023-06-19:
[image]
[image]
[image]
[image]

mroenne2022, 2023-06-19:
Die LED wird alle 30sec blinken weil er dann versucht den WR zu erreichen. 
In Kürze ist Dunkelheit und Sense für heute, dann erreicht man den WR nicht. 
Tip: NRF Modul angucken ob ein + 
Entfernung zum WR?
ggf Sendeleistung etwas erhöhen zum Test. 
Netzteil stark genug für Versorgung?

graueresel, 2023-06-19:
WWenn das blinken richtig ist, dann warte ich mal ab und melde mich Morgen wieder. Aktuell kommen noch ca. 80 W vom Dach.
Habe gerade nochmal ein stärkeres Netzteil angehängt. Entfernung zum Wechselrichter max. 5m durch eine Außenwand.

mroenne2022, 2023-06-19:
Woher weißt du das 80W kommen?
Siehst du nun bei LiveDaten etwas ?
Das blinken ist richtig, aber das macht der esp sowieso, auch wenn kein WR in der Nähe ist. Das ist ja so eingestellt und sagt nichts aus. 
Entfernung sollte null Problem sein.

graueresel, 2023-06-19:
[image]
 
NRF +

mroenne2022, 2023-06-19:
Das kann niemand erkennen, du musst mit Lupe auf den schwarzen Cip kontrollieren ob ein + dahinter steht. Ich befürchte das es keines mit + ist

graueresel, 2023-06-19:
Langsam kommt nichts mehr vom Dach. Dann müssen wir morgen weitermachen

graueresel, 2023-06-19:
Die Leistung vom Dach wird mit Shelly gemessen

mroenne2022, 2023-06-19:
Ah Okay - mit Shelly. Grundsätzlich muss der WR erreicht werden , muss in der Übersicht bei Systeminfo auch so sichtbar sein . Wenn das (morgen)  nicht der Fall ist , dann kommen keine Daten. Das ist die Grundvorrausetzung, wenn nichts kommt alles halt nochmals prüfen , PinOut,Kabelverbindungen ...usw

graueresel, 2023-06-19:
Okay, das machen wir so

mroenne2022, 2023-06-19:
Ich kann da nichts machen, hab eigentlich entsprechend die Tips angesprochen, akribisch kontrollieren musst du das morgen. Ich drücke die Daumen. Ich glaube der Punkt LED prirority muss auch auf off,  bin da nicht so sicher, ggf kann das jemand anders bestätigen. Ich hab zur Zeit OpenDTU bei mir in Betrieb, daher hab ich keinen Screenshot parat von Ahoy

graueresel, 2023-06-19:
Vielen Dank für die Hilfe und melde mich wieder

graueresel, 2023-06-20:
Ich habe gerade noch die vorbeschriebenen Einstellungen geprüft und eingestellt. Nach kurzer Zeit mit etwas höherer Sendeleistung kommen nun die LIVE Daten an. Vielen Dank für die schnelle und kompetente Hilfe.

mroenne2022, 2023-06-20:
@graueresel : Glückwunsch 👍😃

graueresel, 2023-06-26:
Hallo, ich habe heute mal die Leistungsbegrenzung ausprobiert. Leider kann ich kein Limit einstellen und bekomme folgende Meldung. Vielleicht kann mir jemand helfen?

[image]

lumapu, 2023-07-03:
@graueresel die Fehlermeldung sagt aus, dass der Wechselrichter das Powerlimit nicht akzeptiert hat. Deine Uptime ist sehr kurz, funktioniert die Funkverbindung überhaupt zum Wechselrichter? Sehr interessant wäre natürlich auch er Output der Konsole weiter oben. Dafür wäre es noch hilfreich die Einstellung `Serial Debug` unter `System Settings` zu aktivieren.

graueresel, 2023-07-04:
Hallo, danke für die Hilfe. Jetzt funktioniert auch die Einstellung des Limit.

closed 2023-08-05 by lumapu