lumapu/ahoy #1000: AhoyDTU Instabilitäten mit >= v0.7.3

nexulm, 2023-06-22:
Hallo,
nachdem ich seit einigen Tagen die v0.7.3 im Einsatz habe, möchte ich folgende Beobachtungen teilen:
1) v0.7.3 hat min. 1x täglich einen Hardware-Watchdog reset ausgeführt. Möglicherweise ist dies der Grund warum die MQTT-Werte Übermittlung für mich seit Tagen gut aussieht. Ein möglicher 20s andauernder Reset fällt mir in meiner Beobachtung kaum auf.
2) Heute habe ich auf den aktuellsten Stand v0.7.6 aktualisiert. Dieser hat schon nach kürzester Zeit den Versand der MQTT-Nachrichten eingestellt. Warum? Ein erneuter "Send" in den Settings hat dies temporär gelöst, mit dem nächsten automatischen Update nach 30s wurde der MQTT-Versand jedoch wieder eingestellt.
3) Jetzt läuft die Vorgängerversion v0.7.5 (bisher ohne MQTT-Aussetzer). Jetzt werde ich beobachten, ob der Hardware-Watchdog in den nächsten Stunden/Tagen wieder zuschlägt oder ob diese Version diesbezgl. stabiler als die v0.7.3 läuft.

nexulm, 2023-06-23:
Update:
v0.75 lief gestern ebenfalls instabil, lieferte am Ende des Tages nur noch 8Wh als TotalYield anstelle der >2400Wh, die produziert wurden.
Danach habe ich sicherheitshalber wieder die v0.73 aufgespielt, die zuvor Tage zuverlässig (bis auf die beobachteten Hardware-Watchdog Resets) gearbeitet hat.
Doch damit waren plötzlich wieder die alten Bekannten da, dass das NRF24+ Modul nicht mehr erkannt wurde. :-(
---
Heute Morgen dann immer noch, deshalb gleich mal einen Neuflash mit nodeFlasherPy "from-the-scratch" via USB und "wipe all flash" durchgeführt. Und siehe da, danach wird das NRF24+ wieder erkannt und alles läuft wie zuvor. 
Warum???????
---
Mit diesem Stand fahre ich nun erstmal in Urlaub da er Hardware-Watchdog Reset eher positiv war und dazu führte, das die AhoyDTU tagelang lieferte. Alle anderen Versuche der letzten Wochen seit v0.6.0 waren leider immer mehr oder weniger instabil und liefen nur für max. 4-5 Tage. Ohne Einsatz vor Ort und ggf. Komplett-Reset und Neuflash (wipe all) ging dann oftmals nichts mehr.
---
Hoffentlich kann in den nächsten Wochen die Stabilität verbessert werden und wieder ein Release >v0.6.9 bereitgestellt werden. :-)

lumapu, 2023-07-03:
Ich nehme mal an du meinst `YieldDay`, das Problem welches du beschreibst wird auch im issue #848 behandelt.

welche Hardware wird eingesetzt ESP8266?

nexulm, 2023-07-03:
Oops, habe TotalYield in YieldDay in meinem Kommentar angepasst. Wenn ich es richtig verstanden habe, machte der Reset des Wertes doch Sinn da nach einem Neuflash dieser Wert bet den Tag wieder neu akkumuliert wird und mich vom Mikro-Wechselrichter gelesen wird, oder?
Ja, ich nutze einen ESP8266.
—-
Meine AhoyDTU mit v0.73 läuft seit 2 Wochen ( s. letzter Kommentar) stabil. Nur der regelmäßige Hardware-Watchdog Reset stört geringfügig. 😉

lumapu, 2023-07-03:
YieldDay wird vom Wechselrichter gelesen, wenn es aber am Abend wolkig ist kann es sein, dass der Wechselrichter kurz ausgeht und wenig später wieder an. Dann sind die Werte auf Wechselrichterseite zurückgesetzt und werden von Ahoy auch so weitergegeben. Leider hat sich irgendwo noch ein Fehler bei der Übertragung der Yield-Werte eingeschlichen, daher der Verweis auf #848

rejoe2, 2023-07-28:
Weiß nicht, ob das dasselbe ist:

Habe in der letzten Zeit immer mal wieder versucht, aktuelle binaries via OTA auf den ESP32 zu bekommen. Das klappt zwar, aber danach startet der ESP alle 12-15 Sekunden neu - zurück auf 0.6.15 und das Ding läuft wieder stabil.

Jetzt habe ich es endlich geschafft, VSCode auf dem neuen Laptop zu installieren und das auch selbst komplett neu zu bauen (0.7.19) - flash via USB, factory reset, json wieder drauf => selber Effekt mit den ständigen Neustarts.... 

Jemand eine Idee dazu?

nexulm, 2023-07-28:
Interessante Beobachtung. Die Fehlerbilder hängen aber nicht direkt mit einander zusammen.
---aber---
Ich habe heute auch mal einen v0.7.19 stand selbst gebaut und via OTA auf die Laufende v0.7.15 geflasht. Und sie da, bei mir auch dein beschriebenes Fehlverhalten. Im Browser sehe ich noch, dass er keine Zeitinfo per NTP erhält, ein Update über das verfügbare Icon Sync Time via Browser hat  den permanenten Reset alle 12-15s nicht verhindern können.
Ich habe denn die gleiche selbstgebaute Firmware via USB nodeMCY pyFlasher aufgespielt und alles funktioniert stabil ohne Reset. Sicherheitshalber habe ich dann nochmal per USB mit "Wipe All data" geflasht und alles from the scratch schnell neu konfiguriert. Jetzt läuft wieder alles stabil seit einigen Minuten.
---
BTW: Das von mir ursprünglich beschrieben Phänomen konnte ich seit v0.7.11 nicht mehr beobachten. Damit lief meine AhoyDTU -bis auf die gelegentlichen Hardware-Watchdog Resets- über einige Wochen stabil.

lumapu, 2023-07-28:
ist evtl. das neu eingeführte NTP Intervall zu klein, Standard sollte 720 sein

nexulm, 2023-07-28:
Zumindest bei mir war der Default = 720 eingetragen. Es sei denn der wurde durch das OTA-Update auf v0.7.19 überschrieben.
Der Flash via USB hat zumindest wieder den Default-Wert (=720).

rejoe2, 2023-07-28:
Hmmm, habe eben nochmal die 0.7.19 drauf gemacht, 720 war da als Intervall im Web-Interface zu sehen, und der ntp-Server ist auch erreichbar, die Zeit wird gesetzt.

Alles from the scratch neu einzugeben ist mir eigentlich grade zu blöd, kann man nicht wenigstens den Inverter-Teil per (abgespecktem) json-blob laden?

lumapu, 2023-07-29:
meinst du den Import der Wechselrichter?

rejoe2, 2023-07-29:
Das hat nicht so einfach geklappt, habe dann nach und nach die Inverter von Hand eingetragen.

Die 0.7.19 läuft jetzt seit einigen Minuten, ABER: den letzten Inverter habe ich erst mal auf "disabled" gesetzt. Das ist der MI-600.... Klingelt da was?

rejoe2, 2023-07-29:
Das hat nicht so einfach geklappt, habe dann nach und nach die Inverter von Hand eingetragen.

Die 0.7.19 läuft jetzt seit einigen Minuten, ABER: den letzten Inverter habe ich erst mal auf "disabled" gesetzt. Das ist der MI-600.... Klingelt da was? 

Nachtrag: Der Neustart erfolgte "zuverlässig" immer dann, wenn #5 in "serial" durch war... Dürfte also schon an #6 liegen - entweder ist es die Menge, oder der MI-Typ...

lumapu, 2023-07-30:
das müssen wir jetzt rausfinden, ich nehme an du hast einen ESP32, beim ESP8266 sind nur 4 zugelassen.
Hast du eine Möglichkeit den ESP per USB und serieller console zu betrachten?

rejoe2, 2023-07-31:
Ja, ist ein ESP32. 
Werde ausgangs der Woche mal screen darauf ansetzen. Gibt es irgendwas, auf das ich speziell achten sollte?

PS: Mit manipulierter Nummer (1141 statt 1041) stört sich der ESP nicht an dem zusätzlichen Inverter, es scheint also wirklich am MI-Typ zu liegen. Bei 1041 ist die letzte Info im serial des Web-Interfaces die Anfrage an den davorstehenden Inverter (auf die noch keine Antwort kommt).

lumapu, 2023-08-05:
bitte nochmal das neueste Release `0.7.25` testen!

nexulm, 2023-08-06:
Ursprungsthema ist aus meiner Sicht behoben, sodass das Thema geschlossen werden kann.
Den gelegentlichen Hardware-Watchdog Reset habe ich zumindest bis <= v0.7.19 gesehen, zeigt sich aber eher als unkritisch.

closed 2023-08-06 by nexulm

rejoe2, 2023-08-07:
Seit eben läuft 0.7.26. Die mag den MI auch noch nicht...

Wir machen einen separaten Thread auf, oder?

lumapu, 2023-08-07:
ja genau, hoffentlich hijacked keiner den neuen issue 😇