lumapu/ahoy #1001: keine Verbindung zu Ahoy-Dtu nach Einbindung ins lokale Wlan

Frontplatte, 2023-06-22:
## Hardware
  * [ ] ESP8266
  * [x ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [ x] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [x ] external antenna (SMA)

### Power Stabilization:
* [ x]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: 0_.6_.9__
Github Hash: _______

### Build & Flash Method:
* [x ] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)
Hallo miteinander,

habe folgendes Problem:

nach dem Flashen der DTU ist diese über  die Browseroberfläche unter der bekannten IP erreichbar.
Wird nun über den Menüpunkt Netzwerk das lokal Wlan-Netz eingetragen (SSID und Password) und anschließend mit save die 
Konfig gespeichert und die DTU neu gestartet.
Nach einem Blick in die Fritzbox sieht man nun die der DTu zugewiesene IP-Adresse des lokalen Netzwerkes.
Versucht man nun diese IP-Adresse im Browser (egal ober FF oder Edge) aufzurufen, ist die Browseroberfläche der DTu nicht mehr 
erreichbar.
Ebenfalls bleibt auch ein anpingen der IP-Adresse nicht von Erfolg gekrönt. 100% Verlust der Daten.

Langsam gehen mir die Ideen aus, an was das liegen kann.
Vlt hat jemand von euch einen Tip

Vielen Dank im voraus.

Norbert

jpwiedenhoff, 2023-06-23:
habe das Problem auch... 

Bei mir ist es ein ESP8266.

Vorallem: Vor einem halben Jahr ging alles Wunderbar... dann wurden die PageLoads immer langsamer... bis zu 5 Minuten für den Aufruf aus dem Browser. Dannach ging gar nix mehr - daher jetzt über den Webinstaller einmal komplett zurück gesetzt. 

Bei mir bleibt aber auch dauerhaft das AHOY DTU WLAN aktiv... - Nur bei einer testweise einwahl ins Gästenetzwerk ist das dann mal verschwunden. Wieder geflasht - im Regulären Netzwerk frisst die Fritzbox das ding nicht

Frontplatte, 2023-06-23:
Hallo, 
ja das ist irgendwie seltsam.
Wenn ich das Wlan in der Ahoy einrichte, das ganze speichere und neu starte, dann ist der AP Ahoy-DTU disabled und die DTu bekommt eine lokale Adresse. Diese ist aber nicht per Browser erreichbar. :worried:
Der MQTT der im gleichen Netz ist empfängt die Daten von der DTu und wertet diese auch richtig aus.

Werde heute mal versuchen die DTU mit einem Linuxmaschine zu erreichen. Vlt geht da noch was.
Ansonsten...........:cursing_face: probiere ich es mal mit opendtu

mtet2, 2023-06-25:
Hallo,
die Browser mogeln manchmal in das "http://ahoy-dtu/" ein s "https://ahoy-dtu/" und dann wird sie nicht gefunden. Nur so eine Idee.

lumapu, 2023-07-03:
> Hallo, die Browser mogeln manchmal in das "http://ahoy-dtu/" ein s "https://ahoy-dtu/" und dann wird sie nicht gefunden. Nur so eine Idee.

Das finde ich extrem nervig. Vor allem wenn ich eine lokale IP aufrufen will werde ich in den seltensten Fällen ein gültiges SSL Zertifikat haben.
Also immer prüfen, egal ob per Namen oder IP aufgerufen wird, ob sich evtl. ein `s` eingeschlichen hat.

Oeldruck, 2023-07-26:
Ich habe genau dasselbe Problem mit der identischen Konfiguration - nur ohne externe Antenne - wie Frontplatte in seinem Eingangspost.
Die AHOY-DTU ist mit WLAN und IP-Adresse in der Fritzbox sichtbar und als Aktiv markiert, aber deren Weboberfläche ist im Browser nicht erreichbar, Fehlermeldung TIMEOUT beim Verbindungsaufbau.
Die DTU reagiert also nicht auf den Versuch zum Verbindungsaufbau.

Oeldruck, 2023-07-28:
Ich denke, ich habe dieses Problem gelöst:
In der Fritzbox habe ich im Menü WLAN / Sicherheit einen Haken gesetzt bei "Aktive WLAN-Geräte dürfen untereinander kommunizieren".
Und dann konnte ich vom Laptop aus die Eboberfläche der AHOY-DTU sofort erreichen.

Grüße

Daniel

jpwiedenhoff, 2023-07-28:
Hi,

ne das ist zumindest bei mir leider keine Lösung. Die Option hatte ich vorher schon aktiviert.

Viele Grüße
Jan Wiedenhoff

closed 2023-08-05 by lumapu

vito-user, 2023-09-07:
Hallo Jan,
konntest du mittlerweile das Problem lösen.
Ich habe jetzt genau das gleiche Verhalten.
Viele Grüße, Uwe

jpwiedenhoff, 2023-09-07:
Nein, probiere allerdings derzeit auch nicht mehr zu lösen.

Oeldruck, 2023-09-08:
Hallo zusammen, auf der Ahoy Webseite schaut es so aus,als ob da an der Verkabelung was geändert wurde und  auch Pinbelegungen ?
Ñeue SW gibt's auch.
Grüße
Daniel

vito-user, 2023-09-08:
Nachdem ich noch die beiden Anschlüsse D3&D4 gedreht und das Setting angepasst habe, schien alles wieder zu gehen. Die Freude dauerte aber nur kurz, und es war alles wieder wie vorher. Auch ein anschließendes Update von 0.6.0 auf 0.7.36 brachte keine Änderung.
Da mir hier nichts besseres weiter einfällt, werde ich mir wahrscheinlich mal eine ESP32 (anstatt aktuell Wemos D1) zusammenstecken und probieren.

vito-user, 2023-09-12:
Noch ein kurzer Nachtrag, weil jetzt alles wieder zu gehen scheint.
Die Ping-Abfrage zur DTU von der Kommandozeile meines Win-Rechners lieferte meistens Zeitüberschreitung. Die Pingabfrage von meinem Linux-Rechner wiederum war da aber kulanter und zeigte Zugriffszeiten von durchschnittlich 103 ms an ohne vorher abzubrechen! Also war die DTU generell schon erreichbar, nur war es recht langsam. Ein Neustart meines Routers bewirkte, dass jetzt wieder normalere Zugriffszeiten anstehen (durchschnittlich 22 ms)!