lumapu/ahoy #1003: ePaper Display Pinout

skender85, 2023-06-23:
## Hardware
  * [ ] ESP8266
  * [x] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [x] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [x] circuit board
* [ ] external antenna (SMA)

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [x] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: 0.7.6
Github Hash: 0b763df

### Build & Flash Method:
* [ ] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [x] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

Hallo zusammen

Erstmal dankeschön für die super Software und eure Arbeit. Die neu installierte Balkonanlage macht dank euch mindestens doppelt so viel Spass!! :)

Ich habe lange nach einem Pinout für die AhoyDTU, ESP32 und das Waveshare 1.54" ePaper Display, jedoch leider nichts gefunden. Schlussendlich habe ich das ePaper Display mittels Waveshare Manual an den ESP32 angehängt (https://www.waveshare.com/wiki/E-Paper_ESP32_Driver_Board) und in den AhoyDTU Settings die gleichen Pins angegeben, nämlich:
<html>
<body>

Pin | ESP32 | Description
-- | -- | --
VCC | 3V3 | Power input (3.3V)
GND | GND | Ground
DIN | P14 | SPI MOSI pin, data input
SCLK | P13 | SPI CLK pin, clock signal input
CS | P15 | Chip selection, low active
DC | P27 | Data/command, low for commands, high for data
RST | P26 | Reset, low active
BUSY | P25 | Busy status output pin (means busy)

</body>
</html>

Natürlich habe ich VCC mit 5V, statt dem 3V3 verbunden, da der 3V3 bereits durch das nRF24L01+ Modul belegt ist.

In den AhoyDTU Einstellungen fürs ePaper Display habe ich sowohl die Einstellung der Tabelle oben mit den Pins eingestellt, wie auch die Einstellungen "off / default" und jeweils neu gestartet. 

Mein Display zeigt jedoch nur ein willkürliches Muster / "Ameisenhaufen" an:
[image]

Beim Display handelt es sich um die Version 2 (Rev2.1):
[image]

Nun meine Frage: Ich vermute, dass ich das Display wohl falsch am ESP32 angeschlossen habe. Gibt es eine Instruktion, wie dieses anzuschliessen ist? Im Wiki habe ich nur eine leere Seite gefunden.

Besten Dank für eure Hilfe!!

knickohr, 2023-06-23:
[image]

dknubben, 2023-06-24:
"Natürlich habe ich VCC mit 5V, statt dem 3V3 verbunden, da der 3V3 bereits durch das nRF24L01+ Modul belegt ist."

Das ist ja sehr optimitisch gedacht... Ich lese in den Specs bei Waveshare "Power input (3.3V)" für VCC, dann muss das so auch angeschlissen werden. Im schlimmsten Fall wird das Display und auch der ESP durch die zu hohe Spannung zerstört (wenn die GPIOs die 5V-Spannung abbekommen).

rmayergfx, 2023-06-25:
Nur mal so am Rande. Wenn ich mir die Refresh Zeiten der diversen Displays ansehe und wie lange diese teilweise beim Endanwender halten nehme ich gerne Abstand davon. Auch der hohe Preis schreckt mich ab. Möchte ich wirklich Strom sparen, dann klemme ich kein Display an und nehme ein Board das möglichst sparsam ist. Ja ein ePaper sieht schick aus mit schwarz auf weiß und im Idealfall funktioniert es auch richtig gut, sie sind jedoch generell eigentlich nicht dafür gebaut worden in regelmäßigen kurzen Abständen die Anzeige zu ändern. Das sieht man auch an den Rezensionen zu diesen ePaper, teils hohe Werte bei den Refresh Zyklen oder nach wenigen Monaten defekt/ Teildefekt, so das man ein neues Display benötigt.

Loetnase, 2023-06-26:
[image]

zu aller Info
Jetzt funzt es mit 3,3V

knickohr, 2023-06-26:
Dann macht zu das Ding.

closed 2023-07-03 by lumapu