lumapu/ahoy #1007: Your NRF24 module can't be reached, check the wiring, pinout and enable.But NRF24 is disabled

Sascha27, 2023-06-28:
My Error is :
Your NRF24 module can't be reached, check the wiring, pinout and enable.

But NRF24 is disabled and CMT2300A is Enabled and Working
What´s wrong

Installed V 0.7.6

lumapu, 2023-06-30:
that you use a developer version where this was not fixed for now. Thank you for the hint.

Sascha27, 2023-06-30:
Is it not fixed since 0.7.3 ?

⁣BlueMail for Android herunterladen ​

dtuuser, 2023-07-08:
Bei mir sind es zwei fehlende Bauteile. Bevor ich da  Lötzinbrücken reingebraten habe, war das Modul nicht erreichbar. Dann ist mir beim Vergleichen aufgefallen, dass die beiden SMD Bauteile fehlen.
Weiß jemand, was da fehlt? 

[image]

closed 2023-08-05 by lumapu