lumapu/ahoy #1008: Idea: USB Adapter for NRF24L01+ for RPi / Docker /...

MartinBachmannHD, 2023-06-30:
I am happily using Ahoy with 8266 and HM-600 on 7.6.3. This works really great and I am a huge fan! But I was wondering the following:
- I am using an RPi with Docker for Raspberrymatic, OpenHab, Mosquitto, rtl_433, zigbee2mqtt, ...
- Because of Raspberrymatic CCU (which is using all RPi-GPIO-Pins) I cannot connect the NRF24L01+ as described already to the RPi
- There are USB-converters available for NRF24L01+ (random Google search: https://www.makerfabs.com/usb-adapter-for-nrf24l01.html)
- why not also support this USB-version in a Docker image? Then via a Docker image you could use Ahoy also on e.g. Synology, PCs, ...

Advantages: 
- most 'early-adaptors' anyway have a small server running for OpenHab, Mosquitto or similar -> running Ahoy there
- one power adapter less
- no issues with the limited resources of 8266
- easier backup, monitoring, ...

Feedback welcome!

lumapu, 2023-07-03:
aktuell läuft der code nur auf den ESPs, das liegt vor allem an Funktionen wie `millis()`. Mir fehlt schlicht die Zeit sowas aktuell umzusetzten, aber evtl. will sich jemand dem antun. Ich finde es sehr interessant, dass es einen gesonderten Adapter hierfür gibt.

avendretter, 2023-07-29:
Habe mich gestern mal etwas auf die Suche begeben und aktuell stehe ich vor der Herausforderung AHOY mit cmake oder g++ zu kompilieren. Da ich aber nicht verstehe wie Abhängigkeiten installiert werden, stehe ich grad ein bischen wie der "Och vorm Berg".

Das Dockerfile an sich sollte recht einfach sein und auch das mounten der Pins beim RaspberryPi geht wenn man diesem Thread glauben schenken darf: https://stackoverflow.com/questions/30059784/docker-access-to-raspberry-pi-gpio-pins

Wenn also jemand mir helfen kann, wie ich g++ richtig konfiguriere, dass er alle nötigen Pakete findet (z.B. `Arduino.h`), wäre das toll und würde uns ein Stück weiter bringen :)

stefan123t, 2024-10-27:
@MartinBachmannHD @ghost arbeitet ihr noch an diesem Issue oder können wir das schließen. 
Es gibt einen RPi Code unter `/tools/rpi` falls Ihr die Kommunikation prinzipiell auf dem RPi zum Laufen bekommen wollt.
Eure Variante den C++ Source Code von AhoyDTU (eigentlich für ESP8266/32) nativ auf dem RPi zu compilieren sollte prinzipiell auch gehen ist aber eine Herausforderung.

PaeserBastelstube, 2024-10-28:
Hallo @stefan123t, hallo @MartinBachmannHD, 
auf meinem Raspi-2b und dem Raspi-4 konnte ich den `USB-NRF24L01+` nicht zum Laufen bringen. Zu meiner Schande muss ich gestehen, dass ich den Issue aus den Augen verloren hatte. Ich nehme mit noch einmal bis Mitte November 2024 Zeit um weitere Erfahrungen zu sammeln. Vielleicht kann man(n) das Thema auch in anderen Projekten gebrauchen.
Viele Grüße aus Päse

MartinBachmannHD, 2024-10-28:
@stefan123t @PaeserBastelstube Super dass ihr hier nochmal nachhakt! Leider kann ich selbst gar nicht technisch unterstützen, aber wenn es was zum testen geben würde dann kaufe ich mir gerne den USB Adapter und teste fleißig :-).

stefan123t, 2024-10-29:
@PaeserBastelstube Knuti ich habe Dich mal als Bearbeiter zugewiesen und es mit RPi getagged, Du kannst Dir also alle Zeit der Welt lassen um das in aller Ruhe und ausgiebig zu testen :stuck_out_tongue_winking_eye:

PaeserBastelstube, 2025-03-24:
Hier mal ein Zwischenstand:
den NRF24L01-USB-Adapter kann ich via Python (Raspberry und Win11) ansprechen. Alle Kommandos (z.B.: AT?) werden leider mit den folgenden beiden Zeilen kommentiert.
```
你发送的信息为:AT?	--> Die von Ihnen gesendete Nachricht ist: AT?
NRF24L01发送失败		--> NRF24L01 Das Senden ist fehlgeschlagen
b'\xc4\xe3\xb7\xa2\xcb\xcd\xb5\xc4\xd0\xc5\xcf\xa2\xce\xaa\xa3\xbaAT?'
b'NRF24L01\xb7\xa2\xcb\xcd\xca\xa7\xb0\xdc'
```
Folgende Probleme wurden erkannt:
1. jede Zeile muss mit "\r\n" (Carriage-Return dann erst ein Line-Feed) abgeschlossen werden, nicht umgekehrt "\n\r" !
2. Wenn ihr Putty verwendet:
    auf den automatisch eingerichteten COM-Port konfigurieren
    Change Settings: Windows / Appearance / Font Settings / Change /
				Schriftart = NSimSun
				Script von "Westlich" auf "Chinesisch GB2312"
    Windows / Translation / Character set translation 
				Remote Character set: von "UTF-8" auf "Use font encoding"
    Das NRF24L0 Modul baucht immer ein "CR dann ein LF" bei Zeilenabschluss: 	Enter drücken - dann CTRL+J drücken
		"Enter"  macht ein Carriage Return(“\r”)
		"Ctrl+J" macht ein Line Feed(“\n”)
3. bein Verbinden des "SWIM-PIN" mit "GND" wird zwischen Empfangsmodus und Sendemodus umgeschaltet. Das Modul quittiert mit:
```
-  NRF24L01接收模式	--> NRF24L01 Empfangsmodus
-  NRF24L01发送模式	--> NRF24L01-Sendemodus
```
4. Herstellerinformationen: http://www.chinalctech.com/cpzx/Programmer/Serial_Module/268.html
5. Die USB-Adapter gibt es mit einer "gelben 4x2 Buchse" für das NRF24L01 Radio Modul, alternativ mit einer "schwarzen 4x2 Buchse". Ich habe die "schwarze" Version verwendet.
.
Wer kann meine Ergebnisse mit einem "gelben" USB-Adapter überprüfen?
Kennt jemand weitere Steuercodes? - "!c02" oder "!s0A" funktionieren nicht.