lumapu/ahoy #1009: [Bug] Kein Einloggen ins WLAN wenn mit Version 0.7.6 gestartet

kiu77, 2023-07-01:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.7.6

### Github Hash

0b763df

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

aktiviertes NRF24L01+ Modul und OLED Display.
Zwei Wechselrichter (HM-800 und HM-300).
Eingabe von WLAN-SSID und PW.
Geographische Daten zur Nachabschaltung.

### Debug Serial Log output

```shell
ESP32 30-Pin mit NRF24L01+ von eByte (externe Antenne) und einem 1,5" OLED Display mit SSD1309.
WLAN-Einlog-Probleme unter 0.7.6, die mit 0.6.9 nicht auftreten
```

### Error description

unter 0.6.9 war alles okay, auch nachdem ich via Menü  -> Update auf die neueste Developer Release 0.7.6 upgedatet habe (und den Haken bei NRF24L01+ gesetzt habe).
Heute habe ich aus Versehen bei Experimenten meine DTU komplett gelöscht.
Als ich dann erst die 0.6.9 über den Web-Installer gepaart mit dem vorherigen Löschen draufgeladen habe und dann via Menü -> Update  ie neuste Developer Release 0.7.6 drüber gebügelt und alle Daten neu eingegeben hatte, merkte ich, dass die DTU zwar via WLAN-Zugriff auf "AHOY_DTU" und der IP-Adresse 192.168.4.1 funktioniert, dass sie sich dann aber trotz korrekter Eingabe von SSID und PW meines WLANs nicht in dieses einloggen wollte. Egal was ich probiert habe.
Auch das Einlesen der zuvor unter 0.7.6 gemachte Einstellungen und neue Eingabe des WLAN-PW hat nicht geholfen.

Dann habe ich kurzerhand die letzte stabile Firmware 0.6.9 draufgeladen, die Daten neu eingegeben und schwupps! hat sich die DTU wie erwartet ins WLAN eingeloggt und alles ist gut. Ich könnte nun wieder auf die neueste Firmware updaten, dann wird das weiterhin klappen, aber sie muss 1 x mit der 0.6.9 eingeloggt gewesen sein.

Hypothese: Die WLAN-Einstellungen werden bei der Eingabe unter 0.7.6 nicht korrekt abgelegt. Wurden sie unter 0.6.9 angegeben und erst dann auf 0.7.6 upgedatet, funktioniert die Sache.

Kann das bitte jemand überprüfen und validieren, ob das wirklich ein Bug ist und ob sich das von mir beschriebene Verhalten von jemand anderes reproduzieren lässt? Es gibt ja auch die Möglichkeit, dass ich etwas übersehen habe, und ich will nicht die Pferde scheu machen ohne Grund.

lumapu, 2023-07-03:
beim ESP32 kann ich mir das fast nicht vorstellen. Beim ESP8266 könnte es durchaus möglich sein durch ein Heap-Problem, welches besonders nach einem OTA Update zum tragen kommt. 
Ich würde behaupten, dass die `0.7.6` genauso funktioniert wie die releaste `0.6.9`

kiu77, 2023-07-03:
Hm, mittlerweile habe ich es durch einen anderen User bestätigt. Aber ich werde morgen mal einen ESP löschen, und dann erst die 0.6.9 und dann die 0.7.6 drauf machen, bevor ich die Daten eingebe. Ich teile dann mit, ob es sich auch bei mir mit einem anderen Aufbau reproduzieren lässt. Mein einer Bau ist ein 30-poliger ESP32 mit eByte NRF24L01+ mit externer Antenne. Ich habe noch eine DTU mit 38-poligem ESP32 aufgebaut und gleichem Nordic-Modul.

kiu77, 2023-07-05:
Heute habe ich es mit 0.7.7 probiert: Das funktioniert perfekt!
Morgen suche ich nochmal nach der 0.7.6 und teste das.

kiu77, 2023-07-09:
Sodele, ich konnte es zweimal verifizieren.
Das Vorgehen:
1. DTU-Hardware per Webinstaller löschen und mit 0.6.9 flashen.
2. Einloggen ins WLAN des DTU (SSID = AHOY-DTU)
3. Aufrufen der DTU-Webseite unter 192.168.4.1
4. Per Update auf Version 0.7.6 updaten.
5. Über Settings alle Daten inkl. Network (SSID meines WLAN und PW) und Inverter eingeben.
6. Neustart der DTU.
=> Sie loggt sich nicht in mein WLAN ein.

Wenn ich jetzt die Eingabe des PW für mein WLAN bei Network wiederhole, klappt das Einloggen ins eigene WLAN.

Nichtsdestotrotz ist das ja mittlerweile völlig überflüssig anzumerken (ich wollte nur wissen, ob ich mich getäuscht habe), denn die neueren FW-Versionen ab 0.7.7 aufwärts haben das Problem ja nicht. Insofern eindeutig "solved".

kiu77, 2023-07-09:
@lumapu 
Kann gelschlossen oder gelöscht werden, da kein weiterer Nährwert mehr enthalten...

closed 2023-07-22 by lumapu