lumapu/ahoy #1010: [Bug] MQTT funktioniert nicht wenn "Sonderzeichen" im WR-Namen

DReffects, 2023-07-02:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

board

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.7.9

### Github Hash

weiß leider nicht wo das zu finden wäre^^ build action 5300414822

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

3 hm-1500 WR

### Error description

Die MQTT Topics können nicht publiziert werden wenn die Wechselrichter wie folgt benannt sind:

HM-1500 #1
HM-300
HM-1500 #2

HM-300 wird veröffentlicht, die anderen beiden nicht.

Werden sie in 
HM-1500 1
HM-1500 2

umbenannt passt wieder alles

kiu77, 2023-07-02:
Das hat aber nun gar nichts mit Sonderzeichen zu tun, sondern damit, dass Sie 2 Wechselrichtern den gleichen Namen gegeben haben. Ändern Sie die Namen, geht es. Ich weiß daher nicht so recht, ob das Label "Bug" adäquat ist.
Und eine Version 0.7.9 gibts (noch) nicht.
Etwas mehr Sorgfalt ist gelegentlich hilfreich...

lumapu, 2023-07-02:
das Symbol `#` ist bei MqTT ein Wildcard, kann daher nicht in Namen vorkommen.

kiu77, 2023-07-02:
Ah, jetzt erst habe ich es verstanden, DReffects: Du hast tatsächlich die Strings "HM-1500 #1" und "HM-1500 #2" als Namen eingegeben (ich habe fälschlicherweise angenommen, Du hättest zweimal "HM-1500" angegeben und die #1 und #2 wären Deine Kommentare gewesen)? 
Ja dann einfach nicht machen ;-)

closed 2023-08-05 by lumapu