lumapu/ahoy #1011: [Bug] Fehlermeldung "enqueued cmd failed/timeout"

rbutenuth, 2023-07-02:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.6.9

### Github Hash

15ec6a0

### Build & Flash Method

VSCode - Platform IO (build & flash)

### Setup

D1 Mini (8266), NRF+, Hoymiles 400, 1m Distanz, alles Standard

### Debug Serial Log output

```shell
17:14:58 I: (#0) enqueued cmd failed/timeout
17:14:58 I: (#0) resetPayload
17:14:58 I: (#0) Requesting Inv SN <removed>
17:14:58 I: (#0) enqueCommand: 0x01
17:14:58 I: (#0) enqueCommand: 0x0b
17:14:58 I: (#0) prepareDevInformCmd 0x01
17:14:58 15 pid: 80
17:14:58 I: TX 27B Ch3 | 15 84 71 84 30 81 33 13 20 80 01 00 64 a1 94 72 00 00 00 00 00 00 00 00 7b e7 eb
```

### Error description

Ich habe einen D1 Mini mit dem NRF-Modul verbunden (Standardverdrahtung), über PlatformIO Branch development-03 installiert. Nach Einrichten von zwei Wechselrichtern hat alles funktioniert. Dann habe ich den Aufbau in ein Gehäuse aus Depron verpackt.

Nach dem nächsten Einschalten (bzw. Verbindung mit USB-Stromversorgung) bekomme ich nur noch die Fehlermeldung aus dem Log. Ausprobiert mit diversen Versionen: 0.6.0, 0.6.9, development-03: Immer gleiches Ergebnis.

Mir ist auch nicht ganz klar, wie die Fehlermeldung zu deuten ist: Schlägt schon die Übertragung des Datenpakets vom ESP zum NRF-Modul fehl? Oder erst die Funkstrecke? Ich habe nochmals ins Gehäuse geschaut, Kabel sind alle noch dran. Als Power-Level habe ich LOW und und MIN ausprobiert. Entfernung NRF <-> Inverter ist ca. 1m, ohne Wand dazwischen. 

Spricht das für ein "Montagsmodul", dass nur wenige ein/aus-Zyklen überlebt hat?

kiu77, 2023-07-02:
Mach mal ein Foto von Deinem Aufbau und stelle es ein.

rbutenuth, 2023-07-02:
[image]

In der Vergrößerung möchte man seine Lötstellen eigentlich nicht wirklich sehen...

Den Kondensator könnte ich noch näher an das NRF-Modul packen. Wenn die Fehlermeldung keiner kennt, werde ich morgen ein anderes Modul mit Kondensator näher dran ausprobieren.

lumapu, 2023-07-02:
Der Fehler bezieht sich auf die Funkstrecke. Es bedeutet, dass der Wechselrichter nicht in der vorgegebenen Zeit geantwortet hat.
Wird auf der Index Seite die Version der Wechselrichter angezeigt?
Power-Level sind richtig, beide können verwendet werden.

Kondensator muss hier im System sein, bei den Kabelquerschnitten und der Größe des Elkos würde ich nichts ändern. Das du gelötet hast würde ich den Verbindungen trauen.

Als Versuch wäre es wert IRQ und CE zu vertauschen, sowohl Hardware als auch in den Einstellungen.

kiu77, 2023-07-02:
Wenn ich es richtig sehe, entspricht Deine Verdrahtung nicht ganz dem Standard, rbutenuth.
Ich hab mal versucht, das herauszuschreiben. Wenn ich mich nicht täusche, sind D5 und D6 beim ESP8266
bzw. SCK und MISO beim NRF24L01+ vertauscht, oder?
Und dann ist mir nicht klar, wo GND beim ESP8266 angeschlossen ist.
Sehe ich das richtig oder habe ich mich getäuscht?
[image]

rbutenuth, 2023-07-03:
@kiu77, danke für die Mühe des Nachverfolgens!
Die Verdrahtung sollte passen, hat ja auch einmalig funktioniert.
D5 (auch bei mir blau): Geht zum dritten Pin in der gleichen Reihe wie GND (Platinenkante)
D6 (bei mir schwarz) geht an den letzten Pin in der gleichen Reihe.
GND ist beim ESP am GND-Pin angeschlossen, sieht man auf dem Foto nur nicht, da er vom Pin aus nach unten geht, unter der ESP-Platine durch, wird links durch den roten Draht verdeckt. 
Ich tausche jetzt mal testweise CE und IRQ (wie von @lumapu vorgeschlagen). Wenn das nicht geht, kommt ein neues NRF-Modul rein. Wozu bestellt man gleich fünf Stück. ;-)

rbutenuth, 2023-07-03:
So, Erfolgsmeldung, aber nicht so einfach: CE und IRQ getauscht, ausprobiert, alles funktioniert.
NRF-Modul wieder im Gehäuse eingeklebt, funktioniert nicht mehr (dabei ist kein Murmeltiertag...). Vorher hat es gelegen, danach gestanden. 
Gehäuse um 90° gedreht: Funktioniert.
Ich habe dazu nur zwei Erklärungen:
1. Es ist irgendwo ein Wackelkontakt und die Schwerkraft hilft.
2. Das Modul ist bezüglich Antennenausrichtung extrem empfindlich.
Ich vermute es ist 2.. Dabei ist die Entfernung NRF-Modul <-> Inverter maximal 1m mit nichts außer Luft dazwischen.
CE/IRQ hätte ich vermutlich nicht mal tauschen müssen.
Danke für die Hilfe bei der Fehlersuche.

rbutenuth, 2023-07-03:
done

closed 2023-07-03 by rbutenuth

kiu77, 2023-07-03:
Bei mir klappen diese NRF24L01+ Module mit aufgedruckter Antenne auch 15m durch ein Dach mit Ziegeln durch problemlos. Vielleicht probierst Du es besser mit einem Modul mit externer Antenne wie das von eByte. Das scheinen die besten zu sein.

rbutenuth, 2023-07-03:
Geht jetzt ja. Das muss ein ganz komischer Effekt gewesen sein. Eventuell auch _zu nah_.  Wechselrichter hängt an der Wand, Gehäuse mit ESP + NRF steht darunter, wie gesagt, weniger als 1m.

mthoms80, 2023-08-04:
ich klinke mich hier einfach mal mit ein, denn ich habe leider das gleiche Problem. tatsächlich habe ich hier 2 verschiedene AhoyDTU´s, einmal auf Basis eines ESP8266 und einmal ein ESP32. Jeweils mit eigener NRF24L01+

beide geben mir in der Serial Console "enqueued cmd failed/timeout" wieder, bzw im Startbildschirm "Inverter #0: Hoymiles (v0) is not yet available"

Verkabelung ist beim ESP32 korrekt (mehrfach geprüft).

[image]

rbutenuth, 2023-08-04:
Wie ist die Entfernung zum Inverter? Ich habe die Einstellung der Sendeleistung nicht im Kopf, gab es da nicht ein Setting mit noch weniger als "LOW"? Der Kondensator ist auch relativ weit weg vom RF-Modul, mit mehreren Kontakten dazwischen. Dazu noch das Display (auch an der 3.3V Leitung?) könnte den Spannungswandler überfordern.

Ich würde mit niedrigster Sendeleistung nah am Inverter probieren, ohne Display, verschiedene Antennenausrichtungen.

mthoms80, 2023-08-04:
ja, das lässt sich noch auf MIN einstellen... habe ich soeben gemacht.
tatsächlich bin ich jetzt gerade relativ weit weg vom Inverter. ich würde die AhoyDTU gleich noch einmal in der Garage platzieren, auf dessen Dach die Anlage liegt.

Display habe ich jetzt mal abgeklemmt und auch in der Software deaktiviert.

Kondensator:
also sagst auch du, ich sollte den Kondensator vielleicht einfach mal direkt mit an den NRF anlöten?

kiu77, 2023-08-04:
Der Spannungsregler ist nie überfordert, da er 1A liefern könnte.
Wenn etwas ein Problem darstellt, dann ist es der fliegende Aufbau.
Bei Instabilität steht das an erster Stelle.

Also besser entweder Litzen direkt anlöten oder eine Platine verwenden.

@rbutenuth 
Falls Du eine getestete Platine verwenden willst, könnte ich Dir Layoutdateien dafür
zukommen lassen. Ich kann sie nur im Moment noch nicht öffentlich einstellen.

mthoms80, 2023-08-04:
kenne mich mit Elektronik null aus^^... aber ich entnehme deiner Aussage, dass es nicht an der Position des Kondensator liegen dürfte, ja?

wegen Steckplatine... sind die nicht genau dafür gemacht? aber ok, bekomme ich sicherlich auch ordentlich verlötet. wäre dann mein nächster Schritt

mthoms80, 2023-08-04:
sooo... ich habe die Steckplatine nun entfernt, sowie den Kondensator direkt an die NRF24 angelötet.
Durchgangstest jeder Verbindung habe ich geprüft - passt.

leider gleiches Ergebnis.

kiu77, 2023-08-04:
Hast Du gesteckte Käbelchen verwendet oder Litzen direkt an die Pins gelötet?

mthoms80, 2023-08-04:
direkt gelötet und sitzt auch bombenfest

kiu77, 2023-08-04:
Gut. Und Du hast letztlich auch beim ESP32 keine gute Verbindung zum Inverter?
Kannst Du mal ein Foto von der Oberseite des NRF24 einstellen?
Welche Firmware hast Du aktuell drauf?

mthoms80, 2023-08-04:
ich habe gar keine Verbindung ;o)
Firmware war die Stable 0.6.9, ich habe es aber nun auch mal mit der aktuellen Developer probiert.

Hier mal meine Konfiguration, vielleicht fällt da ja was auf.
[image]

NRF24 - ja, ist ein "+"-Modul ;o)
[image]

kiu77, 2023-08-04:
Also auch keine Verbindung ins eigene WLAN?
Ist die DTU bei der Liste verbundener Geräte Deines Router aufgeführt?

Das NRF24-Modul ist übrigens nur das "zweitbeste".

die von eByte sollen besser sein:
[image]

mthoms80, 2023-08-04:
nene, verbindung ins eigene WLAN ist kein Problem... nur zum Wechselrichter eben nicht