lumapu/ahoy #1012: I2C-Adresse der Displays

kiu77, 2023-07-02:
## Hardware
ESP32

### nRF24L01+ Module
nRF24L01+ verified  **Plus model** from eByte

### Antenna:
external antenna (SMA)

### Power Stabilization:
100uF Electrolytic Capacitor 

### Build & Flash Method:
AhoyDTU Webinstaller

Frage:
Da ich mit unterschiedlichen Displays "rumspiele" (darunter auch welche, die nicht gehen),
wäre ich an der erwarteten I2C-Adresse der OLED-Displays interessiert. Für die Info
wäre ich dankbar.

Ich habe einen I2C-Scanner mit einem Arduino Uno aufgesetzt und sehe,
das manche Displays entweder x3c (default) / x3D oder x78 (default) / x7A haben.

knickohr, 2023-07-02:
Das ist die Standard-Adresse für die gängigen OLEDs : 0x3C wenn ich mich nicht irre.

kiu77, 2023-07-02:
D.h. man müsste sie um Quellcode ändern, wenn man ein untypisches Display hat.

knickohr, 2023-07-03:
Noch viel schlimmer. Eher in der Lib des Displays. Außer man kann es per Parameter mitgeben 😉

Müßte man in der u8g2 nachschauen.

kiu77, 2023-07-03:
Danke Knickohr für den Hinweis!
Dann muss ich mir das mal anschauen.

kiu77, 2023-07-03:
Auf Github hat jemand beschrieben, wie man die I2C-Adresse eines Displays für die u8g2lib  festlegt.
[https://github.com/olikraus/u8g2/wiki/u8g2reference#seti2caddress](url)

Leider bin ich zu inkompetent dazu.
Wenn (ich trau mich kaum zu fragen) @lumapu erwägen könnte, dazu einen Eintrag in Settings -> Display
zu machen und den default Wert 0x3C voreinzustellen, wäre das mehr als super...

[image]

lumapu, 2023-07-03:
ist das nicht eher ein individuelles Problem? Wenn ich was einbaue, dann sollte es doch der Allgemeinheit weiterhelfen. In diesem Fall würde ich dir raten, einen Fork meines Repositories zu erstellen, deine Änderung bei dir zu commiten (du brauchst ja nicht mal die WebUI verändern, sondern direkt den Aufruf der u8g2). Jedes mal wenn ich was veröffentliche kannst du bei dir mergen und github baut dir die neue Firmware mit deiner Änderung.

kiu77, 2023-07-03:
Das kann man so sehen. Ich denke, dass 95% der User nicht auf die Idee kommen, ein anderes Display als die 0,96" oder 1,3" OLED-Teile einzusetzen.
Auf der anderen Seite benutzt auch kaum jemand ein ePaper-Display.
Ich würde das ja selber machen (und Dir schicken), wenn ich könnte, aber ich verstehe schon den Umgang mit dem Code in Github nicht. Ich kann zwar grundsätzlich in diversen Sprachen programmieren, aber hier scheitere ich an der Umgebung und ein C-Crack war ich noch nie. Vielleicht sollte ich mich da schlauer machen...

Auf jeden Fall schadet es nicht, wenn die I2C-Adresse einstellbar ist. Wenn schon die Pin-Belegung mit Funktionen weitgehend konfigurierbar ist. Und es gibt tatsächlich Displays mit anderen Diagonalen, die sich nicht an die 0x3C/0x3D-Regel halten. Wenn jemand ein größeres Display einsetzen möchte, hätte er mit änderbarer Adresse die Möglichkeit dazu. 
Ich habe ein 1,54"-Display mit SSD1309 (statt SSD1306) und das funktioniert prima, da es die Standard-Adresse hat. Es gibt aber auch Versionen mit 2,42" und sogar 2,7" und manche hätten es gerne größer, auch wenn dann as Display teurer als der ganze Rest der Hardware ist ;-)
Aber wie gesagt: Du musst das nicht machen. Nur wenn Du das interessant findest und "eh grade dabei" bist.
Und falls ich es vergessen haben sollte: Ich finde Dein Projekt super!

knickohr, 2023-07-03:
Vooorsicht !!!  Langsam mit den jungen Pferden 😉

lumapu, 2023-07-03:
verstehe deinen Punkt. Ich würde vorschlagen wir warten erst mal und hören uns um was die Allgeminheit hierzu denkt. @knickohr denkt hier so wie ich 😀

kiu77, 2023-07-03:
Hi @knickohr 
Danke für die Auflistung.

Es geht leider nicht mit dem 2,42" Display mit SSD1309 und 4-poligem I2-C-Anschluss, weil das ne andere Adresse hat. Zumindest bei der Version, die ich habe. Mag sein, dass es mit diesen Versionen mit 7-poligem Anschluss geht, die man von SPI auf I2C umjumpern kann. Ich werde mal eins kaufen und ausprobieren und dann mitteilen, ob das geht.

Sehr gut und absolut fehlerfrei geht bei mir ein 1,54" OLED Display mit SSD1309 - ich hab das ausprobiert und im Einsatz.
Es gibt wirklich keinerlei Darstellungsfehler. Insofern dürften OLED-Displays mit SSD1309 in der Regel doch eher funktionieren, da es ja ähnlich zum 1306 ist. Voraussetzung natürlich: Die I2C-Adresse stimmt.

Echt, das ePaper-Display ist das am meisten verwendete? Erstaunlich.
ich war der Ansicht, das fast nur die OLEDE-Displays verwendet werden - zumindest wenn man die Resonanz auf Seiten im Internet so sieht. Aber so kann man sich täuschen...

BastyESP, 2023-07-03:
Wenn ich den Text von Knickohr so lese, frage ich mich kommt denn das 2,42" nun Endlich "fest" auch via Dropdown? Habe schon öfter im Discord gelesen...das es eine eigene Routine gibt, aber extra umgesetzt wird es ja bisher nicht? War aber jetzt auch ne Frickelei mit euren E-papern, evtl kommt da ja noch was, auch wenn es als 0,96" ja schon "GUT" Funktioniert. 

Die Funktion die der Ersteller hier Fordert nützt mir als normaler Anwender ohne Ahnung, ohne das jemand sagt, z.b. Display 1.54" OLED geht, so nicht wirklich etwas. Auch wenn mich das 1.54" Oled noch Deutlich mehr Anspricht wie das 1.54 Epaper.

kiu77, 2023-07-03:
@BastyESP 
Das 1,54" OLED Display geht tatsächlich ohne Probleme direkt, trotz SSD1309.

Von der 2,42" Version mit SSD1309 gibt es aber mehrere Varianten. Ich hab das mit dem 4-poligen I2C-Anschluss.
Das kann wohl nicht gehen, da die I2C-Adresse anders (0x78) ist.
D2 ist auf meinem Display die Diode rechts unten (Pfeil), deren Anode mit SDA verbunden ist. Die Diode dient wohl zum Klemmen der Spannung an SDA auf 3,3V, oder? Und Du meinst, das Brücken der Diode würde helfen?
Obwohl die I"C-Adresse nicht stimmt?

[image]

BastyESP, 2023-07-03:
Definitiv hab ich das selbe Display und da D2 Gebrückt ohne solche Spielchen wie du sie da hast, Läuft....hab nach dem Löten aber keine Bilder gemacht...hier mal das erste Bild nach dem Anschließen vor ca. 4 wochen. Es lief sofort....ich finde es nur für meinen zum Spaß auf OpenDTU im Zimmer Laufenden ESP32 etwas zu Teuer um sich noch ein zweites zu Holen, daher finde ich die 1.54" OLED Lösung interessant, da werde ich bei Ali mal schauen...schöne Info!

Hier das Bild, wenn es sein muss steck ich das Display auch ab und mach dir nen Bild von hinten.

[image]

kiu77, 2023-07-03:
@BastyESP 
Interessant!
Ich werde es testen. 
Mein 1,54" OLED-Display war das hier [https://www.amazon.de/dp/B07ZZ8LLTH?psc=1&ref=ppx_yo2ov_dt_b_product_details](url) (ich hab das Letzte gekauft.
Aber bei Aliexpress gibt es sie auch, wenn auch wegen Porto nicht preiswerter.
[https://de.aliexpress.com/item/1005002767134354.html?spm=a2g0o.productlist.main.1.67143f30ETBZzG&algo_pvid=f928647c-858b-4e56-b695-b048a8a43aaa&algo_exp_id=f928647c-858b-4e56-b695-b048a8a43aaa-0&pdp_npi=3%40dis%21EUR%215.0%214.4%21%21%21%21%21%40211bf4c516884233175393190d0767%2112000033284992077%21sea%21DE%214212928404&curPageLogUid=3WvrYjWrHKRT](url)

Wenn das bei meinem Display klappen sollte, wäre der Schluss, dass die korrekte I2C-Adresse gar nicht nötig ist, sondern dass wohl die Lib abfragt, und sich die Adresse merkt, die sich meldet. Das wäre super und meine Anfrage dann redundant ;-)

kiu77, 2023-07-03:
Es hat mir keine Ruhe gelassen. 
Also bin ich ab in den Keller, Lötkolben angeworfen, Diode raus, ein Zinnklecks drauf, hoch, eingesteckt eingeschaltet
und
**ES LÄUFT!**
Hammer, echt.

Danke vielmals @BastyESP !!!

Ich werde morgen mal die I2C-Adresse eines 1,3" OLED Displays umjumpern und schauen, ob es dann noch geht.
Wenn ja, hat sich mein Anliegen erschöpft und dieser Thread hat es geklärt.
Gewissermaßen über Umwege was herausgefunden. Sachen gibts!

BastyESP, 2023-07-04:
[image]
Läuft!
Das ist doch schonmal super, das dir der Tipp half.

Amazon scheint mittlerweile keine Alternative mehr zu sein für das 1,54" und 5 vs..12€ für das 2,42" ist halt doch ein Unterschied.

Bin gespannt was rauskommt bei deinen Experiment.

knickohr, 2023-07-04:
Bitte nicht in Euphorie verfallen wegen der SSD1309-Displays. Wie ihr selbst seht, gibt es viele unterschiedliche Varianten. Nicht allein wegen der Größe, Adresse und Pinbelegung, mit 7 oder mit 4 Pins, umkonfigurierbar und auch nicht. Schon allen das macht es schwierig da eine Linie zu finden.

Es wird bei diesem Display wohl immer ein Gebastel bleiben und je nachdem wo man kauft, auch Glückssache. Ja, das SSD1309 kommt rein, in OpenDTU ist es schon als Pinmapping in der DEV vorhanden.

BastyESP, 2023-07-04:
Ist bei OpenDTU das Problem der sekündlichen Aktualisierung (Pixel Shift) dann auch weg? Da hatte doch teletapi sorgen mit? Aber das hier eh das falsche Forum für ;)

Aber schön zu lesen das es bald Aktiv ist in Ahoy

knickohr, 2023-07-04:
Jepp, so wie ich weiß, jetzt nur noch alle 10 Sekunden.

kiu77, 2023-07-04:
@knickohr 
Keine Euphorie, aber Erfolg. Das 2,42" läuft ebenfalls super (siehe Bild). 

@all
So, ich habe jetzt rausgefunden, wie sich die Sache verhält:
Die Firmware bzw. die u8g2lib erwartet strikt die I2C-Adresse 0x3C.
1. Ich habe ein umjumberbares 0,96"-Display auf 0x3D umgelötet und nix, nada. Nach zurücklegen auf 0x3C tat es wieder.
2. Beim 2,42"-Display war HTML und der Aufdruck auf der Rückseite geduldig: Statt der angegebenen Adressen 0x78 und 0x7A war das faktisch 0x3C. Ich habe das mit einem I2C-Scanner überprüft. Kein Wunder also, dass es direkt tut (wenn man ACK aktiviert).

Mein Problem ist also erst einmal gelöst: Alle meine Displays haben die I2C-Adresse 0x3C. Auch wenn 0x78 hinten drauf steht (ist bei mehreren Displays so). Vermutlich werden die zwischen SPI und I2C umschaltbaren Displays im I2C-Betrieb dann einfach auf die "Standard-Adresse" 0x3C eingestellt.
Da ich faktisch kein von 0x3C abweichendes Display habe, habe ich auch kein Problem, wie ich durch diesen Thread herausgefunden habe. Danke an alle Beitragenden.

Man kann also sagen, dass alle diese OLED-Displays mit I2C gehen: 0,96", 1,3", 1,54" und 2,42". Es gibt auch noch ein 2,7" mit dem Controller SSD13012, wenn ich mal Lust habe, ordere ich das und probiere es aus.

Nochmals vielen Dank an alle!

[image]

BastyESP, 2023-07-04:
@kiu77 
Na siehst du, so einfach kann es manchmal sein. 
Nach deinem Controller zu dem 2,7" gesucht, finde ich aber nichts? Sicher das der von dem 2,7" so heißt?
Finde bei Aliexpress ein Angebot von 2,7er OLED (leider gibts die nur in Grün) mit einem SSD1327, zu normalen Preisen...ca 2€ Teurer wie das 2,42".

Aber brauch lieber erstmal ein Gehäuse...für mein 2,42" das reicht ja von der Größe und Funktionalität erstmal aus....wenn du aber mal eine Positive Rückmeldung gibst, falls du dir soeines bestellst, gerne her damit. Nach dem Motto ;) Bigger is Better

@knickohr 
bei Ahoy hab ich die Dev versionen noch gut gefunden, über die Startseite, wie bekommt man die denn bei Open?
Wenn es irgendwo bei Github versteckt ist, bin ich eh raus, die Seite finde ich von der Übersichtlichkeit unterdruchschnittlich...

Wird die Geschichte "in Watt" anzeigen und Kilowatt bis 9999w weglassen auch angegangen, denn das Nervt mich bei meiner OpenDTU hier im Zimmer am meisten, bis 999w genau, und danach 1.0 kW, 1.1kW, Ahoy macht ja wenigstens noch z.b. 1.07kW also auf 10w genau, statt 100w, aber das nur am Rande, da es hier ja eigentlich um Ahoy geht...

kiu77, 2023-07-04:
@BastyESP 
sorry, Du hast natürlich recht, das 2,7er hat einen SSD1327.
Ich werde es aber trotzdem ausprobieren. Wie Du siehst, habe ich ein simples Plastikgehäuse (auch von Ali) mit 82x82mm. Da würde das 2,7er mit Ach und Krach gerade so reinpassen. Ich hab auch noch ein Gehäuse über...
Die Befestigung erledige ich schlicht mit Heißkleber. Dazu ist eine kleine Pistole mit 7mm Material besser geeignet als die dicken Dinger für 11mm Stäbe.

knickohr, 2023-07-04:
Das SSD1327 hat 128x128 Pixel. Selbst wenn es funktioniert wird die die Hälfte vom Bildschirm was angezeigt.

Aber die u8g2 hat es implementiert, man müßte den richtigen Aufruf in der Lib machen.

kiu77, 2023-07-04:
Tatsächlich gibt es gerade nur grüne 2.7er.
Dafür gibt es ein quadratisches 1,5er mit SSD1322 und 128x128 Pixel.
Dass beim 2,7er mit SSD1327 nur oben was angezeigt wird, das glaube ich nicht.
Das Steuer-IC kann 128 Pixel vertikal - das Display selbst ist aber 128x64.
Solange nur in die oberen 64 Reihen geschrieben wird (wie bei den anderen Displays)
Sollte einfach eine n normale Anzeige erscheinen.

BastyESP, 2023-07-04:
@knickohr hab nur die Daten vom Ali genommen, es sah aus wie das richtige Display stand 128*64 und 2,7". War auch nur weil ich auf das Markieren und Suchen seines SSD13012 nichts gefunden habe...

Will da auch gar nichts neu erfinden, aber beim 1.54" hatten wir in Discord auch schon, die SSD1309 diskussion...da sagtest du mir auch das es ein anderer Controller ist...obwohl Berrybase oder sowas, das als 1309 führte, und im Datenblatt stand wieder was anders...das scheint sehr "Flexibel" zu sein...aber wohl auch Teilweise zu Funktionieren, das 2,42er gibts ja auch noch in Blauer Platine, ob das überhaupt so einfach zu überzeugen ist...daher Schwarzes und D2 Brücken läuft...wissen muss man es nur...

Am Ende scheint die U8g2, ja einiges herzugeben ;) wenn wirklich ein Wille da ist es zu Implimenteren, momentan reicht ja die Auswahl an Display von winzig 0,91" bis 2,42", wenn jetzt noch das 1,54" als Oled geht, wie kiu77 schreibt, dürfte ja so langsam kein Auge mehr Trocken bleiben...außer die Bigger is Better fraktion....

knickohr, 2023-07-05:
Die Hersteller, vor allem die in China, machen so wie sie wollen. Und das ist auch unser größtes Problem 😢 Es gibt unzählige Varianten von Displays und der Babbelfisch übersetzt das auch immer spannend. Wir haben uns bei der Implementierung daran gehalten was so in D gängig verfügbar ist, vor allem bei AZ. Ganz einfach weil die Bauteile von denen bis jetzt immer funktioniert haben.

Die u8g2 unterstützt gefühlt ca. 1000 Displays. Kannst Dich ja mal dran versuchen. Im Prinzip wäre es ein Leichtes das zu implementieren, vorausgesetzt man schreibt die Routinen für die entsprechenden Auflösungen. Mir nur einfach (korrekt) aufrufen ist es leider ja nicht getan. Das muß ja auch auf die Anzeige passen und nicht zu viel Schwarz drum herum haben 😉

BastyESP, 2023-07-05:
Also ich Arbeite mich da sicherlich nicht rein, dazu hab ich mal ganz viel zu wenig Ahnung von sowas...

Mir reicht die Auswahl ja auch, Hauptsache das 2.42" Läuft schön...das 1.54" werde ich die tage wohl mal eins bestellen, weil Kiu77 ja damit Positive Erfahrungen gemacht hat. Für mich wäre noch Toll wenn das "kW" Problem auf die >9999w verlagert wird...da es ja scheinbar nur eine Software Einstellung ist, wenn es beim E-Paper funktioniert.
Werden aber die meisten eh nicht bemerken mit 600/800w Wechselrichter...wenn Kiu77 sich so ein 2,7er Bestellt, bin ich auf die Rückmeldung gespannt, an was es dort evtl hakt....D2 war ja hier auch nur eine Kleinigkeit...

knickohr, 2023-07-05:
Das kW-Problem tritt aber nur bei OpenDTU auf 😉

BastyESP, 2023-07-05:
Jein, das ganz fiese 1.0kW 1.1kW 1.2kW (mehr Leistung hab ich nicht) Problem betrifft nur die OpenDTU, das stimmt, die AhoyDTU kennt wenigstens noch zwischenschritte 1.0kW 1.01kW etc. aber es hatte ja schonmal jemand in Support-Display in Discord angefragt vor ca 1. Monat

"Vielleicht hatten wir das Thema schon und es gehört hier ggfs. auch nicht hin , IST es möglich die Anzeige im Display in Watt auszugeben auch wenn mehr als 1 KW produziert werden ?"

Das meinte ich...und ob nun 1000w oder 1.00kW geschrieben werden, dürfte "Platztechnisch" nicht relevant sein...sind beides 5 Stellen (ja auch ein Punkt nimmt Platz weg ;) ) und in der Obesten Zeile welche die Leistung anzeigt, ist ja noch "Reichlich" Platz...

lumapu, 2023-07-05:
oh je, da war noch was - wie hieß der issue doch gleich? #914?

knickohr, 2023-07-05:
Ich glaube das betrifft nur die OLEDs und das Nokia, nicht das ePaper. Zumindest steht bei mir die Leistung korrekt im Tausend Watt. Erst ab 9999 wird auf 10.00kW umgeschaltet.

lumapu, 2023-07-05:
da gings eher allgemein um die Anzahl der Stellen aller im System befindlicher Zahlen

knickohr, 2023-07-05:
Hier in diesem Issue ist das wohl aber auf die Displays bezogen ?  😉

Egal, die vielen Stellen die da bei MQTT oder in der API raus purzeln stören mich herzlich wenig, weil am Ende bei der Darstellung auf einem Dashboard formatiere ich das eh nochmal nach meinem Gusto.

Aber stimmt schon, man sollte keine höhere Genauigkeit vorgaukeln als vorhanden ist.

BastyESP, 2023-07-15:
Hatte ja angedroht mir mal ein 1.54" OLED vom Ali kommen zu lassen, das wurde eben geliefert, also mal Angeschlossen an meine OpenDTU* im Zimmer, geht Sofort an, steht auf dem Kopf und hat den Pixelfehler an der Seite, also das 1.3er Eingestellt, Läuft als müsste es so sein, also ohne Jeglichen Lötaufwand und deutlich Günstiger wie das 2.42...ja auch nochmal kleiner wie das 2.42, nur für Leute die nicht Löten wollen/können, und ein möglichst großes Display wollen, sicher eine Gute Alternative. 

 *Gehe davon aus in Ahoy gehts genau so, aber ich hab es nicht getestet.

kiu77, 2023-07-15:
@BastyESP 
Ja, das habe ich auch schon eingesetzt und funktionierte super.
Nächste Woche kommt das 2,7" - mal schauen ob es tut oder ob es vorzieht, als Briefbeschwerer zu fungieren.

BastyESP, 2023-07-15:
Bin ich je gespannt ob es einfach will, oder auch gar nicht, mittlerweile ja ne große Palette an Funktionierenden Displays von 0.91 (Ahoy Only) über 0.96 1.3 1.54 und 2.42, wenn nun noch das 2.7er will, bleibt Oled Mäßig ja kaum noch ein Auge Trocken.

BastyESP, 2023-07-15:
Wir implimentieren doch nichts, wir schließen auf gut Glück was Bestelltes aus China an und wenn ich bei dem selben Händler das selbe Display mit anderer Farbe bestelle, und es unterschiedlich Ansprechen muss, bringt doch das doch alles eh nichts...hätte ich das Blaue nicht mitbestellt hätte ich geschrieben bei dem 1.54" musst du das 1.3er einstellen, dann läufts, jemand bestellt sich das in Blau, und hat die Pixelfehler an der Seite und muss das 0.96" auswählen...

Zudem ich eh keine Ahnung von dem Code habe, und dann alles rauskommen würde, nur kein Lauffähiger Datenbank aufruf.

--- Offtopic ---

Hab auch noch ein 0.91" mitbestellt, da ich mal sehen wollte wie die Darstellung dort so ist...weil es ja nur als Pull Request rein kam und nie jemand Bilder zeigte.

Das sieht so aus, ist also ECHT winzig...sitzt man nicht direkt davor, kann man sich das Sparen...dachte es wären dort Infos weggelassen worden, aber es wird alles Angezeigt in mini Schrift...

[image]

Dann mal meine "Displaysammlung" auch mal zum Größenvergleich, fehlt eigentlich noch ein 1.3"

[image]

Und wenn man mal eine Andere Darstellung möchte stellt man das 0.91" einfach ein ;)

[image]

Ja, so im Nachhinein hätte ich mir das 0.91" Sparen können, wenn ich gewusst hätte das man sich die Darstellung auch so anschauen hätte können, aber die kosten ja eh fast nichts...

kiu77, 2023-07-15:
Das 0,91" hat sicher nur 128x32 Pixel.
Hammer dass die Bibliothek das erkennt und die Darstellung in der Firmware gut angepasst wird.

Btw.. vorhin kam das 2,7" Teil. Ich komme aber wohl erst Montag zum Ausprobieren.

BastyESP, 2023-07-15:
Hat es auch nur, das kann man in Ahoy auch direkt Auswählen unter den Displays daher kennt er das, und stellt das Korrekt dar, klingt ja Toll das es schon da ist, dann lesen wir wohl Montag von Erfolg oder Misserfolg.

[image]

kiu77, 2023-07-15:
Ah, super, war mir wieder mal entgangen...

BastyESP, 2023-07-15:
Ist Aber schon in 0.6.11? oder sowas reingekommen, da ich nie in den Display Einstellungen rumgemacht habe (hatte ja nur das 0.96" am laufen, weil ich es da hatte) brauchte man dort ja nicht mehr rein....hatte halt erwartet Zeigt nur die Leistung an, und als kleine Ergänzung evtl Yield Day, aber das alles angezeigt wird, macht es dann natürlich sehr "Winzig".

War eben der Spaß mal was wert, hab die Bilder gemacht, das es auch andere sehen können, und dann fliegt es wieder in die Ecke zu den anderen Displays die man nicht benutzt...

knickohr, 2023-07-15:
> Wir implimentieren doch nichts, wir schließen auf gut Glück was Bestelltes aus China an und wenn ich bei dem selben Händler das selbe Display mit anderer Farbe bestelle, und es unterschiedlich [...]

Ja, das ist doch genau das !  Ich schließt da einfach was an und hofft das es funktioniert. Egal ob es korrekt angesprochen wird oder nicht. Klar, die Controller sind recht tolerant und fressen so einiges. Heißt aber nicht das es dann auch supportet wird. Offiziell hat halt Ahoy (und OpenDTU) nur eben die 5 aufgelisteten Displays im Paket.

kiu77, 2023-07-15:
@knickohr 
Ich würde ja gerne mitmachen bei der Code-Entwicklung, aber bin schon mit Github und dessen Code-Organisation überfordert...

@BastyESP 
Richtig, was ist ein Elektroniker ohne ausgiebige Sammlung nicht genutzter Displays. Du solltest meinen Bastelkeller sehen...

BastyESP, 2023-07-15:
@knickohr Können sie doch auch haben, es kauft/verkauft weder kiu77 noch ich ein Produkt was Ahoy/OpenDTU kompatibel war, es wird einfach versucht, wenn es nicht läuft hat man entweder Pech, oder man findet jemanden der es einem Anpassen kann...es sind keine Werte, die ganze bestellung aus 3 Display und ein paar Kabeln (damit ich noch mehr zeug Zusammen stecken kann) kostete weniger wie das 2,42", das muss man auch mal sehen...Preis/Leistung seh ich bei den 1.54 Weit vorne, 0.91/0,96" kann man sich sparen, eines hatte bzw habe ich noch genug da, und das eine hab ich nun dazu gekauft...

Werde mich jedenfalls in das Coding nicht reinlesen, weil ich für mich keinen Vorteil sehe, es Läuft mit den Displays die ich hier habe, komplett..

kiu77, 2023-07-15:
Ich würde mich schon reinfuchsen. Aber nicht im Moment. Und ich brächte ne Anleitung "in einfacher Sprache"...

knickohr, 2023-07-15:
Korrekt, sehe ich auch so, habe selbst genügend Displays hier rum liegen. Andere sehen es aber vielleicht anders und fordern dann auch noch Support, obwohl die Dinger nicht offiziell drin sind.

Kennen wir doch irgendwoher von diesen Kleinanzeigen-Bastelbausätzen die für teuer Geld verhökert werden und wir in der Community dann den Support leisten sollen.

Ja klar, ich würde auch liebend gerne Codebeiträge leisten wollen, ich komme aber mit dem PlattformIO-Gedöns auf meiner Windows 7 Büchse nicht zurecht 😢

kiu77, 2023-07-15:
Ich hab zwar auch MacOS und Win11, aber das hilft mir nicht viel ;-)

BastyESP, 2023-07-15:
Also ich verkaufe keine DTU auf Kleinanzeigen, keine Angst.

Glaube jetzt auch nicht das soviele sich nun 1.54" oder 2,42" oder wenn es Funktioniert 2,7" bestellen, wenns dazu keinerlei Gehäuse gibt, evtl sieht man es auf einem Bild, mein 2.42" hängt an 2 Gummibändern vor einer Shelly 1PM kiste in der Küche und Staubt ein ;) Schön ist das nicht...aber es Funktioniert ganz gut. 

Was die Sache halt auch verrückt macht das sich die hersteller bei den Gleichen Display und auch Gleichen Rückseiten, nicht mal auf die selbe "Ansprache" einigen können, da wirds dann auch mit Support schwer, außer man sagt, stellt testweise mal was anderes ein, es wird schon Funktionieren....das Weiße hängt außer die 10min wo ich das Blaue nochmal angesteckt habe, an der OpenDTU mit ihrer Sekunden Hoppelei und macht keine fehler...glaube nicht das sich da viel Aufwand lohnt...wenn kiu77 einen Aufruf für das 1.54" schreibt, werde ich ihn testen, wenn er bei dem Weißen oder Blauen nicht geht, geh ich auf die Einstellung die Geht.

knickohr, 2023-07-15:
Nun ja, für das 2,42“ gibt es Gehäuse und es gibt schon eine Beta-Firmware (zumindest für OpenDTU). Also sooo weit hergeholt ist das jetzt nicht.

Wenn ich @dAjaY85 richtig verstanden habe, sind es wenige Zeilen Code um den Lib-Aufruf passend zum Controller zu machen. OK, bei Ahoy muß man noch das WebGUI anpassen, ist aber auch Pillepalle. Schaut mal unter dem PR #902 nach, da müßte eigentlich alles drin sein was gemacht werden muß.

BastyESP, 2023-07-15:
@knickohr 
[image]

Links das weiße und rechts das blaue.

Selber Händler selber Angebot... unterschiedliche Ansprache...auch das Angebot noch raus und Stelle es mit ein..

knickohr, 2023-07-15:
Ja, das ist die PCB, der Conrtoller sitzt mit auf dem Glas.

BastyESP, 2023-07-15:
https://de.aliexpress.com/item/1005005489779639 war das Angebot, wo ich gekauft habe.

Man kann jeweils beide Farben als 7 oder 4 Pin auswählen...mehr sorgen hatte ich mir auch nicht gemacht...es hieß das es laufen sollte, das schrieb Kiu77 ja hier, und dann hab ich 2 genommen um in die 10€ nähe zu kommen, und noch etwas dazu.

BastyESP, 2023-07-15:
Öhm, hatte für die Fotos ja das Display abgesteckt, das 0,96" aufgesteckt (das läuft aber nicht automatisch ohne reboot an, wie die 2,42" oder 1,54") dann das weiße wieder angsteckt, darstellung Falschrum mit Pixelfehler.. die DTU neugestartet, richtig rum mit !Pixelfehler!...nun auf SD1306 gestellt, zeigt ganz normal an...1306 war ja das pedant zum 0.96er? Hä? Warum das...die ganze DTU ne Minute vom Strom genommen...und zeigt weiter unter 1306 normal an...und sag mir nicht das es am Aufruf liegt...es lief ja dann heute mit Beiden aufrufen mal Richtig... irgendwie strange! Komische Displays...aber wenn sie laufen machen sie auf beiden keine Zicken, einzig Anstecken im Laufenden Betrieb bekommt ihnen scheinbar nicht...sollte ich mir abgewöhnen...

knickohr, 2023-07-15:
Es ist eben kein SSD1306, auch wenn es zufälligerweise funktioniert. Das ist das was ich versuche die ganze Zeit Dir/Euch zu erklären.

BastyESP, 2023-07-15:
Ach es ist mir eigentlich auch egal, was und wo es ist, auf der Platine vom Display steht ja hinten ein CH1116, was ja nahe an dem SH1106 liegt wie es in der OpenDTU/AhoyDTU heißt, also wäre der Aufruf womit es am Sinnvollsten laufen sollte wohl das 1,3", nun läuft es aber via 0.96" und zeigt richtig an.

Hauptsache etwas größer wie das bisherige 0.96e, was ich angesteckt hatte, das reicht mir ja eigentlich schon...das mit dem vorher Ausmachen (muss ja eh zu 99% Neugestartet werden danach, werde ich nun Beachten...) ansonsten bin ich für das Geld ganz zufrieden, auch wenn die Unterschiedlichen Einstellungen nach dem Foto sehr verwirrend waren...nachdem ich ja die Unterschiede zwischen den Farben hatte, nun wären ja beide wieder Gleich...0.96" Aufruf...aber ich steck es nun eh nicht wieder um..

knickohr, 2023-07-15:
Und nochwas, Du wirst sehen das das weiße schneller altert als das Blaue 😲

BastyESP, 2023-07-15:
Spannend das es da in der Haltbarkeit solche unterschiede gibt, hatte bisher nur die blauen 0.96er und noch alte 0.96er blau gelb, das 2,42" hab ich eigentlich bewusst hell gekauft das es auch meine Eltern noch gut erkennen können, bei den 1,54er ging je nur eines mit günstigem Versand, und beim 0.91 war es eh egal, was hält denn am längsten, die 2,42" gibt es ja in Gelb und Grün noch...die Display kann man bei dem großen ja einzeln tauschen, von daher bin ich da beruhigt, bei 5€ für das 1,54" auch, egal welches man nutzt... Aber falls man sich ein 2,7er kauft kann man ja drauf achten..

knickohr, 2023-07-15:
Ich würde in dieser Reihenfolge antworten :   Weiß, Gelb, Blau

Zumindest habe ich mit denen Erfahrung. Rot (eigentlich mehr orange) und grün habe ich bei meinen 3D-Druckern verbaut. Mal sehen wie lange sie es machen. Weiß ist jedenfalls recht schnell verblaßt 😢

BastyESP, 2023-07-15:
Naja es ist nicht Kriegsentscheidend solange es nur Verblasst...schlimmer wäre dann das einzelne Pixel total ausfallen bzw deutlich dunkler sind wie der Rest. Wollte heute Abend eh mal in Thingiverse schauen obs für das 1,54 Oled irgend ein Case gibt, was man zweckendfremden könnte für meine DTU...evtl verbau ich dann das Blaue da rein, und lege mir das Weiße auf Halde, aber im endeffekt wenn das weiße verschlissen ist, baut man das Blaue ein, nimmt sich nicht viel. Aber gut zu wissen, das es da unterschiede gibt.

BastyESP, 2023-07-17:
@kiu77 Muss erstmal nicht zwingend das Todesurteil sein, meist haben die via Löten die Option auf i²C zu kommen, schau doch mal auf der rückseite was diese Sagt?

[20 more replies]