lumapu/ahoy #1013: [Bug] MQTT sendet 0-Werte bei /total beim setzen eines Power-Limit

knickohr, 2023-07-04:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.7.7

### Github Hash

982c201

### Build & Flash Method

ESP Tools (flash)

### Setup

13 Inverter, MQTT aktiv, Power Limit

### Error description

Seit der 0.7.7 wird über MQTT Null-Werte nach dem setzen eines Power-Limits in den Total-Werten ausgegeben. 

[image]

[image]

Ich schätze das hat was mit den beiden gefixten Issues #927 und/oder #929 zu tun. Mit der 0.7.6 ging es noch (bin mir da bei nicht zu 100% sicher da ich erst seit gestern Power-Limits setze).

lumapu, 2023-07-05:
ja ich verstehe das Problem, denn das Powerlimit ist das gleiche Kommando wie die Abfrage der Live-Daten. Einziger Unterschied ist die Länge der Payload. Ich weiß was zu tun ist ...

lumapu, 2023-07-05:
verstehe noch nicht ganz warum das bis zu MqTT durchschlägt, das muss ich live sehen um daran arbeiten zu können. Dem Code nach dürfte das nicht passieren.
Welche Wechselrichtergeneration verwendest du? HM-Serie richtig?

knickohr, 2023-07-06:
Es passiert direkt nach dem Bestätigen des PowerLimits vom WR. Dann folgen die 4 /total Werte mit Null. Danach geht’s normal weiter.

Alles HM-300 mit der Seriennummer 11218xxxxxxx

knickohr, 2023-07-06:
Ich habe gerade bemerkt, es passiert nicht nur beim Setzen des Limits, sondern auch wenn ein Power-Limit aus irgendeinem Grund (beim Start der Inverter ?) reportet wird. Das meintest Du wohl mit „gleichem Kommando wie Abfrage der Live-Daten“.

lumapu, 2023-07-07:
@artcrime welcher ESP? 8266?

lumapu, 2023-07-07:
@knickohr korrekt, ich hoffe die nächsten Tage, evtl. sogar am WE das fixen zu können. 😊
Siehst du noch was, was einem Release in Wege stehen könnte?

knickohr, 2023-07-07:
Sieht gut aus. Wenn das mit dem Power Limit weg ist dann haben wir einen sehr stabilen Build. Noch das Problem mit den Nullen beim HMS (Issue #1016) dann ist es eine Release.

Leider funktioniert halt das Fusion-Board nicht mehr seit der 0.7.x, aber das ist eine größere Baustelle.

artcrime, 2023-07-08:
> @artcrime welcher ESP? 8266?

ja genau, ESP8266. bekomme heute noch nen ESP32 und teste mal ob es damit besser wird. Es scheint auch vom Topic abzuhängen mit welchem ich steuere.

Mit inverter/ctrl/limit/0 gibts weniger Probleme als mit folgendem inverter/ctrl/limit_nonpersistent_absolute/0

Auch wenn man den Wechselrichter auf komplett 0% steuern möchte hat es den selben Effekt. Wenn man dann über die Weboberfläche einmal von Hand auf 100% regelt ist es mal wieder für ne Zeit alles ok.

knickohr, 2023-07-08:
@lumapu 
Lukas, mit der 0.7.9 stimmt was nicht 😲 Er donnert nur noch im Sekundentakt die Hardwareinfo der WRs raus und nach einem weiteren Reboot ist Comm_disabled True, was natürlich jetzt nicht stimmt. Ja, Koordinaten sind eingetragen und NTP ist auch synchronisiert.

lumapu, 2023-07-08:
habs Grad auch gesehen, danke für den Hinweis

knickohr, 2023-07-08:
Offenbar nicht immer. Hab’s gerade nochmal geflashed, scheint jetzt zu funktionieren. Allerdings sind jetzt die WRs aus.

lumapu, 2023-07-08:
es fehlt eine Bedingung, daher bleibt die MqTT-Schleife hängen, gleich kommt `0.7.10`

knickohr, 2023-07-09:
Nein, tütet immer noch nicht. Fast gleiches Verhalten, immer nur HW-Infos von den WRs, nur ab und zu mal was „richtiges“ dazwischen.

[image]

knickohr, 2023-07-09:
Jahhh, jetzt schnurrts wieder richtig 👍
Dann werde ich mal den eigentliche. Bug überprüfen 😉

lumapu, 2023-07-09:
super, denke heute kommt noch eine Version bzgl. @MiniOh Issue #1016 😎

knickohr, 2023-07-09:
Zu das Ding, funktioniert 👍

closed 2023-07-09 by knickohr