lumapu/ahoy #1014: Einfacheres Setup mit WLAN

rmayergfx, 2023-07-05:
@lumapu 
Mit den ESP Web Tools https://esphome.github.io/esp-web-tools/ und passendem Browser (Chrome, Edge) kann man ganz einfach ESP8266 und ESP32 Hardware installieren und auch gleich das WLAN konfigurieren. Das ist sehr komfortabel und relativ schnell und einfach erledigt, man wird nach erfolgreicher Installation und Verbindung zum WLAN auf Wunsch auch noch gleich mit dem Device verbunden. Damit kann ich jederzeit testen ob ein ESP sauber funktioniert, bzw ein Device auch zurücksetzen.

Setzte ich das Device mit diesen Tools auf, kann dann auch über den Button OTA-Update die benötigte *.bin für AhoyDTU einspielen. Mehrfach erfolgreich getestet. Leider wird nun mit der neuen Software auch gleich wieder das WLAN auf default zurückgesetzt und man muss sich wieder mit dem HotSpot AhoyDTU und Kennwort verbinden. Ein Fallback auf den HotSpot ist super, aber für das erste Setup wäre es doch wirklich hilfreich das man wie bei den ESP Web Tools ein vorhandenes WLAN am PC schon auswählen könnte und auch direkt an der Tastatur gleich das Kennwort mitgibt.

Einfach mal darüber nachdenken, ob so etwas vielleicht möglich ist oder man zumindest eine config mitgibt, z.B. wie beim Raspberry Pi die wpa_supplicant.conf.

lumapu, 2023-07-05:
hi, ich arbeite schon seit längerem daran, nur leider funktioniert es bei mir nur in den seltensten Fällen. Soll ich es mal in die dev Version bauen, dann könntest du es auch testen?

rmayergfx, 2023-07-05:
Ja gerne

lumapu, 2024-02-29:
hast du inzwischen den "Wizard" mal gesehen? Wie gefällt er dir?

rmayergfx, 2024-03-01:
@lumapu 
Wizard? Seit welcher Release gibt es den und wie funktioniert der? Wo taucht der auf? Gibts dazu ein Howto. Lese das gerade das erste mal.

lumapu, 2024-03-01:
seit dem 0.8.83 Release, wird automatisch aufgerufen, wenn du im AP Modus bist. Alternativ kannst du es auch direkt mit `/wizard` aufrufen.

[image]

rmayergfx, 2024-03-02:
Kann man das irgendwo ganz oben anpinnen oder auch auf der Homepage schreiben. Das find ich super.