lumapu/ahoy #1015: Build failed

gent99, 2023-07-05:
## Hardware
  * [ ] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [ ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [ ] external antenna (SMA)

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: _._.__
Github Hash: _______

### Build & Flash Method:
* [ ] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

gent99, 2023-07-05:
I use PIO in VSCode to clone https://github.com/lumapu/ahoy.git and build failed with

In file included from .pio/libdeps/esp8266-release/RF24/RF24.h:18,
                 from app.h:11,
                 from app.cpp:6:
.pio/libdeps/esp8266-release/RF24/RF24_config.h:186:26: error: expected ';' at end of member declaration
  186 |         #define printf_P Serial.printf
      |                          ^~~~~~
.pio/libdeps/esp8266-release/ESP Async WebServer/src/AsyncWebSocket.h:170:12: note: in expansion of macro 'printf_P'
  170 |     size_t printf_P(PGM_P formatP, ...)  __attribute__ ((format (printf, 2, 3)));
      |            ^~~~~~~~
.pio/libdeps/esp8266-release/RF24/RF24_config.h:186:32: error: expected unqualified-id before '.' token
  186 |         #define printf_P Serial.printf
      |                                ^
.pio/libdeps/esp8266-release/ESP Async WebServer/src/AsyncWebSocket.h:170:12: note: in expansion of macro 'printf_P'
  170 |     size_t printf_P(PGM_P formatP, ...)  __attribute__ ((format (printf, 2, 3)));
      |            ^~~~~~~~
.pio/libdeps/esp8266-release/RF24/RF24_config.h:186:26: error: expected ';' at end of member declaration
  186 |         #define printf_P Serial.printf
      |                          ^~~~~~
.pio/libdeps/esp8266-release/ESP Async WebServer/src/AsyncWebSocket.h:261:12: note: in expansion of macro 'printf_P'
  261 |     size_t printf_P(uint32_t id, PGM_P formatP, ...)  __attribute__ ((format (printf, 3, 4)));
      |            ^~~~~~~~
.pio/libdeps/esp8266-release/RF24/RF24_config.h:186:32: error: expected unqualified-id before '.' token
  186 |         #define printf_P Serial.printf
      |                                ^
.pio/libdeps/esp8266-release/ESP Async WebServer/src/AsyncWebSocket.h:261:12: note: in expansion of macro 'printf_P'
  261 |     size_t printf_P(uint32_t id, PGM_P formatP, ...)  __attribute__ ((format (printf, 3, 4)));
      |            ^~~~~~~~
In file included from .pio/libdeps/esp8266-release/RF24/RF24.h:18,
                 from app.h:11,
                 from main.cpp:7:
.pio/libdeps/esp8266-release/RF24/RF24_config.h:186:26: error: expected ';' at end of member declaration
  186 |         #define printf_P Serial.printf
      |                          ^~~~~~
.pio/libdeps/esp8266-release/ESP Async WebServer/src/AsyncWebSocket.h:170:12: note: in expansion of macro 'printf_P'
  170 |     size_t printf_P(PGM_P formatP, ...)  __attribute__ ((format (printf, 2, 3)));
      |            ^~~~~~~~
.pio/libdeps/esp8266-release/RF24/RF24_config.h:186:32: error: expected unqualified-id before '.' token
  186 |         #define printf_P Serial.printf
      |                                ^
.pio/libdeps/esp8266-release/ESP Async WebServer/src/AsyncWebSocket.h:170:12: note: in expansion of macro 'printf_P'
  170 |     size_t printf_P(PGM_P formatP, ...)  __attribute__ ((format (printf, 2, 3)));
      |            ^~~~~~~~
.pio/libdeps/esp8266-release/RF24/RF24_config.h:186:26: error: expected ';' at end of member declaration
  186 |         #define printf_P Serial.printf
      |                          ^~~~~~
.pio/libdeps/esp8266-release/ESP Async WebServer/src/AsyncWebSocket.h:261:12: note: in expansion of macro 'printf_P'
  261 |     size_t printf_P(uint32_t id, PGM_P formatP, ...)  __attribute__ ((format (printf, 3, 4)));
      |            ^~~~~~~~
.pio/libdeps/esp8266-release/RF24/RF24_config.h:186:32: error: expected unqualified-id before '.' token
  186 |         #define printf_P Serial.printf
      |                                ^
.pio/libdeps/esp8266-release/ESP Async WebServer/src/AsyncWebSocket.h:261:12: note: in expansion of macro 'printf_P'
  261 |     size_t printf_P(uint32_t id, PGM_P formatP, ...)  __attribute__ ((format (printf, 3, 4)));
      |            ^~~~~~~~
Compiling .pio/build/esp8266-release/lib0b9/ESP Async WebServer/WebServer.cpp.o
Compiling .pio/build/esp8266-release/libe59/SPI/SPI.cpp.o
Compiling .pio/build/esp8266-release/libef7/RF24/RF24.cpp.o
Compiling .pio/build/esp8266-release/lib015/Time/DateStrings.cpp.o
Compiling .pio/build/esp8266-release/lib015/Time/Time.cpp.o
Archiving .pio/build/esp8266-release/libe59/libSPI.a
Indexing .pio/build/esp8266-release/libe59/libSPI.a
Compiling .pio/build/esp8266-release/lib843/espMqttClient/MqttClient.cpp.o
Compiling .pio/build/esp8266-release/lib843/espMqttClient/Packets/Packet.cpp.o
Archiving .pio/build/esp8266-release/libef7/libRF24.a
Indexing .pio/build/esp8266-release/libef7/libRF24.a
Compiling .pio/build/esp8266-release/lib843/espMqttClient/Packets/Parser.cpp.o
Archiving .pio/build/esp8266-release/lib0b9/libESP Async WebServer.a
Compiling .pio/build/esp8266-release/lib843/espMqttClient/Packets/RemainingLength.cpp.o
Indexing .pio/build/esp8266-release/lib0b9/libESP Async WebServer.a
Compiling .pio/build/esp8266-release/lib843/espMqttClient/Packets/String.cpp.o
Compiling .pio/build/esp8266-release/lib843/espMqttClient/Transport/ClientAsync.cpp.o
Archiving .pio/build/esp8266-release/lib015/libTime.a
Compiling .pio/build/esp8266-release/lib843/espMqttClient/Transport/ClientPosix.cpp.o
Indexing .pio/build/esp8266-release/lib015/libTime.a
Compiling .pio/build/esp8266-release/lib843/espMqttClient/Transport/ClientSecureSync.cpp.o
Compiling .pio/build/esp8266-release/lib843/espMqttClient/Transport/ClientSync.cpp.o
Compiling .pio/build/esp8266-release/lib843/espMqttClient/Transport/IPAddress.cpp.o
[... cut ...]
*** [.pio/build/esp8266-release/src/main.cpp.o] Error 1
*** [.pio/build/esp8266-release/src/app.cpp.o] Error 1
=============================================================== [FAILED] Took 8.97 seconds ===============================================================

Environment      Status    Duration
---------------  --------  ------------
esp8266-release  FAILED    00:00:08.973
========================================================= 1 failed, 0 succeeded in 00:00:08.973 =========================================================

gent99, 2023-07-05:
duplicate of #994 but that's not resolved yet

knickohr, 2023-07-05:
Ja und warum dann nochmal einen aufmachen ?  Es geht deshalb nicht schneller 😉

closed 2023-07-05 by lumapu

lumapu, 2023-07-07:
ist doch egal, einfach frohes öffnen, wir stellen dann klar, das es ein duplicate ist und markieren / schließen entsprechend 😃