lumapu/ahoy #1018: Strange Serialnumber

daved3luxe, 2023-07-07:
## Hardware
  * [x] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: TSUN TSOL-MS600
Retailer URL: [TSUN](https://de.tsun-ess.com/Micro-Inverter/M800)

### nRF24L01+ Module
* [x] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [x] circuit board
* [ ] external antenna (SMA)

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: 0.7.8
Github Hash: SHA: 70b4f6b

### Build & Flash Method:
* [x] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

Hi,
I received my SUN TSOL-MS600 today and wanted to integrate it into AhoyDTU.
Unfortunately, the serial number on the back looks completely different from the given examples.
The serial number contains letters and numbers and is 16 characters long.
Can I use AhoyDTU at all or have I overlooked something?
Regards,
Dave

lumapu, 2023-07-07:
it's not supported from my view, maybe search the issues for that topic, I remember someone else faced the same kind of SN on his Inverter

lumapu, 2023-07-07:
check issue #909

rmayergfx, 2023-07-08:
@lumapu 
Pin doch das https://github.com/lumapu/ahoy/issues/939 bitte mal ganz oben an, das wird nun noch öfter kommen.
[image]

daved3luxe, 2023-10-21:
Gibt's hierzu Neuigkeiten?

stefan123t, 2024-10-27:
@daved3luxe wenn Du uns Deine Seriennummer mitteilen willst, dann können wir ggf. auch prüfen ob sich dafür eine sinnvolle Adresse zur RF-Kommunikation ermitteln läßt.
Bitte auch mal das Issue #909 durchlesen und ggf. dieses hier schließen.

closed 2024-10-29 by daved3luxe