lumapu/ahoy #1019: Keine Verbindung zu Wechselrichter ohne Internet

querys1, 2023-07-07:
## Hardware
  * [ X] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [X ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [X ] external antenna (SMA)

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: 0.5.66
Github Hash: _______

### Build & Flash Method:
* [ X] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

Hallo zusammen,
bei mir war jetzt mal wieder einige Tage kein Internet, da die DSL Leitung durch einen Bagger vom Glasfaserausbau abgerissen wurde.
Hierbei fiel mir mal wieder auf, dass sich die Ahoy DTU nicht mit den Wechselrichtern verbinden kann, auch mit der Verbindung zum MQTT gibt es Probleme. Wenn man 1x am Tag eine Internetverbindung hatte, dann läuft das ganze bis zum Abend durch.
Hat jemand eine Idee woran das liegt? Die Wechselrichter produzieren jedenfalls normal weiter.

knickohr, 2023-07-07:
[image]

querys1, 2023-07-07:
Danke dir, der Time Sync war durchaus vorhanden, den kann ich ja vom Handy aufspielen. Aber ich probier die Version mal aus!

knickohr, 2023-07-07:
Sie ist verfügbar !

querys1, 2023-07-07:
Danke direkt ausprobiert, das mit der dev-Version kannte ich nicht. Läuft so unkompliziert wie vorher. 
Mal sehen wann das nächste mal die Internetverbindung gekappt wird. Zwei Straßen weiter haben die 4x die Leitung getrennt.

rmayergfx, 2023-07-08:
Kleiner TIP zu NTP:
[image]
Hier habe ich den DE Pool eingetragen. Wenn man allerdings öfters ein Problem mit dem Internet hat, so stellt man hier bei NTP die IP z.B. des Routers ein, der sollte immer online sein und liefert die Zeit auch wenn kein Internet vorhanden ist. Es geht natürlich auch jedes andere System das NTP im Hausnetz zur Verfügung stellen kann, z.B. NAS etc, sofern die Dienste aktiv sind und das Gerät immer online ist.

@lumapu 
Kannst du bitte noch ein Feld bei NTP einfügen, um den Zeitsync zu definieren, d.h. wie oft er gemacht wird?

querys1, 2023-07-08:
Sehr guter Tipp! Danke! Auf die Idee bin ich noch nicht gekommen.

closed 2023-09-16 by lumapu