lumapu/ahoy #1020: Powerlimit mit Home Assistant steuern - Kurzanleitung

rmayergfx, 2023-07-08:
Getestet mit HA 2023.7.0
AhoyDTU dev 0.7.8

Man kann recht einfach eine Automatisierung mit HA erstellen:
[image]
[image]
[image]
[image]
z.B. automatisch um eine bestimmte Uhrzeit ein Powerlimit, kann dazu genutz werden wenn mit dem WR von einer Batterie Nachts eingespeist wird.
[image]
[image]
[image]
[image]
Beispiel um das Powerlimit auf 100% zu setzen:
[image]

<TOPIC>/ctrl/limit/0

Topic ist identisch mit den MQTT Settings in AhoyDTU
[image]

Nun noch speichern mit einem passenden Namen, z.b. Leistung_WRNAME_100 und dann kann sofort getestet werden.
Wenn man die Automatisierung nicht benötigt kann man sie auch deaktivieren und trotzdem jederzeit über das Menü ausführen lassen:
[image]

Damit man sich nicht für jeden WR das nochmals einzeln programmieren muss, einfach den Button duplizieren benutzen und dann die geänderten Werte einfach anpassen.

Viel Spaß damit.

reinhard-brandstaedter, 2023-07-20:
Danke @rmayergfx ! Ich hab das bisher immer über die REST API gemacht weil ich keine Doku dazu gefunden habe.
Geht das ganze auch mit absoluten limits, also non-persistent absolute in W?

rmayergfx, 2023-07-21:
@lumapu 
Wenn ich die letzte Doku von AhoyDTU richtig gelesen habe werden die Werte immer non-persistent geschrieben um den Flash Speicher zu schützen, da dieser nicht x-beliebig beschrieben werden kann, verbessere mich bitte wenn das falsch sein sollte.

@reinhard-brandstaedter 
Probiere es doch einfach mal aus. Laut Doku müsste dann dort im Feld Payload statt 100 (%) der gewünschte Wert in Watt eingetragen werden, z.B. 300W. https://github.com/lumapu/ahoy/blob/main/User_Manual.md
Werden beim Zahlenwert keine Einheiten mitgegeben, so nimmt das System automatisch %.

lumapu, 2023-07-22:
@rmayergfx stimmt, werden genau aus diesem Grund non-persistent geschrieben

TechSaiyajin, 2024-01-03:
Hallo Leute,

könnte man über eine Automation ne vernünftige dynamische Regelung abhängig vom gemessenen Einspeisewert realisieren?

rmayergfx, 2024-01-04:
Vernünftige dynamische Regelung scheidet mit diesen Systemen aus meiner Sicht komplett aus, da nicht in Echtzeit auf die Verbräuche reagiert werden kann. Dafür würde ein ganz anderes System benötigt, solch schnelle Systeme haben aber dann auch wieder einen entsprechend hohen Stromverbrauch. 
Wenn es dir aber genügen würde, das das ganze System etwas träger ist und nicht immer sofort reagiert kannst du das mit HA durchaus machen, dann aber bitte mal im Forum von HA nachschauen, da gibt es genügen Beispiele wie man etwas regeln kann.
Ich würde noch etwas warten bis die Systeme mit Batterien und Regelung günstiger werden und dann u.U. damit arbeiten, d.h. den Überschuß in die Batterie und nicht zum Netzbetreiber. Da gibt es jetzt schon einige gute Ansätze dazu.