lumapu/ahoy #1021: [Bug]

slintes, 2023-07-08:
### Platform

ESP32

### Assembly

the DTU was already assembled

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.6.9

### Github Hash

15ec6a0

### Build & Flash Method

was already installed

### Setup

Inverter Interval 10s
MQTT server and topic added, interval left at 0

### Debug Serial Log output

```shell
a few times this:
14:55:07 I: MQTT disconnected, reason: TCP disconnect
then no MQTT logs anymore
```

### Error description

1. Added MQTT settings, all working fine
2. MQTT Server was restarted (happened the first time when restarting the Synology NAS, where Mosquitto is running on in a docker container. But a quick container restart is already enough)
3. DTU does not reconnect when server is back: "MQTT is not connected"

rmayergfx, 2023-07-08:
Bitte einen Aussagekräftigen Title eintragen! BUG kann alles sein. Danke

slintes, 2023-07-08:
> Bitte einen Aussagekräftigen Title eintragen! BUG kann alles sein. Danke Dazu einfach den Thread aufrufen und oben recths auf Edit gehen!
> 

Sorry, ist im Eifer des Gefechts untergegangen, `[Bug]` war vorausgefüllt.

> Probiere einfach mal die aktuelle dev 0.7.8 die unter Actions zu finden ist. Habe zum test 2 Browserfenster offen, Synology Docker und AhoyDTU Statusseite. Ich kann beliebig Eclipse-Mosquitto stoppen und starten, Anzeige wechselt entsprechend zwischen [image]
> 
> und
> 
> [image]
> 

Danke fürs testen! Habe 0.7.8 installiert, hilft aber leider nicht.

> Wenn das nicht funktioniert bitte mal auf dem NAS die installierten Versionen kontrollieren. 
>

Da der MQTT Explorer sich sofort wieder reconnected, schliesse ich ein Server Problem eigentlich aus

> Und lösche bitte den ganzen Vorspann der nicht ausgefüllt ist und zudem nicht relevant für das Problem

Woher soll ich wissen, was hiervon relevant ist? So ein Template existiert ja nicht zum Spass 🤷🏼‍♂️ 🙂 

> und stelle die techn. Daten der Synology lieber mit ein, z.B. DSM und Docker Version, installiertes MQTT Image mit Versionsnummer.

Siehe oben, ein Server Problem schliesse ich aus. Es ist ja auch nicht Aufgabe eines Servers, sich zu allen Clients zu reconnecten.

slintes, 2023-07-08:
> Habe 0.7.8 installiert, hilft aber leider nicht.

Kommando zurück... jetzt funktiniert es 🤔 🤷🏼‍♂️ 
Danke für den Tipp mit dem Update!

closed 2023-07-08 by slintes

rmayergfx, 2023-07-08:
Hauptsache mal schnell aus der Hüfte einen Bug der gar keiner ist gepostet.
Auch muss man nicht unnötig einen Fullquote mit Ergänzungen schreiben wenn man direkt antwortet.
Überleg mal, wüfur das Template gedacht ist? Ja genau, damit es ausgefüllt wird, falls es Probleme geben sollte das die Hardware nicht funktioniert, viele bekommen das nRF24L01+ nicht ans laufen und dann ist das sinnvoll um nicht alles einzeln abzufragen.

Da dein System funktioniert und es nur ein MQTT Problem war, hätten da die 3 Zeilen am Schluss mit Info zur DSM und Docker vollkommen genügt.

Man sollte halt vorher mal alles prüfen und ggfs auch einen Neustart der Geräte machen, bevor man einen Bug postet der gar nicht so vorhanden ist.

Und wenn man ein Problem als gelöst markiert wäre es für andere auch hilfreich zu wissen woran es im Endefekkt gelegen hat.
So ist das wieder ein Thread für die Tonne. Viel Text aber im Endefekt keinerlei wirkliche Hiflestellung oder Problemlösung.

slintes, 2023-07-09:
Gestern hat die DTU doch noch 2 mal die Verbindung zum MQTT nicht wiederhergestellt, aber bei dem Ton hier vergeht mir die Lust mich hier weiter zu beteiligen. Diskussion über Template Nutzung und Quote Style scheint ja sehr wichtig zu sein, gemixt mit haltlosen Unterstellungen, da fühlt man sich so richtig willkommen in der Community... ne, danke.