lumapu/ahoy #1023: [Bug] after update from 7.3 to 7.11 the ESP seems to be partially unresponsive

mmattel, 2023-07-09:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.7.11

### Github Hash

0be3a23

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Worked in 0.7.3

### Debug Serial Log output

```shell
not attached
```

### Error description

Successfully using 0.7.3 and to my shame, I have overwritten the zip file 🤦‍♂️ with 0.7.11

Post updating, which worked well, the ESP became more or less unresposive via the webui and also HA shows that eg sensor values are all 0 (zero). Every now and then, I see a change in HA with normal values but falling back to 0 again.

Something (tm) made the controller blocking to responses and I have no clue how I can support other than reporting this.

On the other hand side, how/where can I download a former dev release?

knickohr, 2023-07-09:
Try rebooting a second time.

mmattel, 2023-07-09:
> Try rebooting a second time.

Just did, no improving responsivness, same bad as before.

knickohr, 2023-07-09:
https://github.com/lumapu/ahoy/suites/13507701460/artifacts/742336627

mmattel, 2023-07-09:
THX
Info, during the downgrade, the installer reported: `The device has been lost`, but worked with a second attempt 👍 

I can definitely prove that the unresponsiveness is gone with downgrading to 0.7.3. Seems that something in beween this and 0.7.11 is causing that.

knickohr, 2023-07-09:
Schreib mal ein bißchen mehr zu Deinem Setup.

- wieviele Inverter und welche ?
- MQTT ?
- Display ?
- Power Limit ?
- …

mmattel, 2023-07-10:
* 1 Inverter, HM-800, 2 Paneele
* MQTT ist eingerichtet und funktioniert bestens
* Kein Display, dient als reiner Umsetzer- read only um Daten über MQTT in HA anzuzeigen.
* Kein Powerlimit

MartinBachmannHD, 2023-07-10:
Dasselbe Verhalten (ui antwortet nicht ab 0.7.3) und ähnliches Setup bei mir:
- HM-600 mit 2 Zellen
- 8266
- MQTT klappt prima
- kein Display, kein Powerlimit, reine Versorgung von OpenHab mit Daten

knickohr, 2023-07-10:
Hmm, hätte jetzt behaupten können das kommt vom vielen Werkeln des 8266. aber kein Display und kein PowerLimit, macht so gut wie nix. Ihr könn5 mal versuchen MQTT zu deaktivieren ob es dann besser ist. Nur mal zum Test,

kiu77, 2023-07-10:
Mit meinem ESP32 und eByte NRF24L01+ (externe Antenne) klappt es auch mit 0.7.11 gut.

MartinBachmannHD, 2023-07-10:
Hab eben unter 0.7.3 MQTT deaktiviert, dann auf 0.7.11 aktualisiert und zwei mal neu gestartet - UI war top (um 23:00 kamen aber auch keine Signale vom Inverter...). Dann unter 0.7.11 wieder die IP meines Mosquitto eingetragen - und nichts ging mehr (selbst ohne Inverter-Signale)... Wie gesagt, unter 0.7.3 geht alles prima, da muss sich irgend etwas resourcen-schädliches im MQTT-Umfeld eingeschlichen haben...

reserve85, 2023-07-11:
Bei mir auch, sobald MQTT aktiviert wird nichts geht mehr mit 0.7.11 - auffällig fand ich auf den ersten Blick die Anzahl der gesendeten MQTT Pakete, die schnellte in die Höhe (über 10000 in wenigen Minuten). Habe dann auf 0.7.8 downgraded und läuft seitdem wieder problemlos.

knickohr, 2023-07-11:
Ihr seid sicher die 0.7.11 erwischt zu haben. Ja, 0.7.9 und 0.7.10 machen das.

reserve85, 2023-07-11:
@knickohr hab es soeben nochmal aufgespielt und jetzt scheint es zu funktionieren. Ich hätte schwören können, dass es die 0.7.11 war.

MartinBachmannHD, 2023-07-11:
...zur 100% Sicherheit habe ich eben auch nochmal auf meinen ESP8266 die 0.7.11 installiert - und hier ist es wirklich so, dass praktisch jeder Klick ca 30 sec bis ein paar Minuten dauert. Zur Sicherheit habe ich ein paar mal per UI rebootet (aber kein Factory Reset). Evtl könntet ihr ja hier noch unter 'System' die CPU-rate oder so hinzufügen (mein Verdacht ist, dass die CPU konstant auf 100% ist...). Im MQTT-Exporer habe ich gesehen, dass jeder Klick im UI mit einem heap_frag von ca 40 belohnt wird, der dann aber nach dem nächsten refresh (30 sec) wieder zurück auf unter 10 geht. 
Nach so 15 Min experimentieren scheint es auch langsam besser zu gehen, die 30-Sekunden 'aussetzer' gibt es jetzt nicht mehr und ich kann es relativ normal bedienen...
[image]

nexulm, 2023-07-18:
Kann ich absolut NICHT bestätigen.
Gestern auf v0.7.11 -von v0.7.3 kommend- aktualisiert.
Mich interessierte v.a. das bis dato mir unbekannte neue Feature/Setting "Reboot Ahoy at nidnight"!
Das ESP8266 basierte Ahoy läuft bei mir mit einem HM1500 Wechselrichter und aktiv genutzten MQTT unauffällig und stabil.

lumapu, 2023-07-19:
kann ich auch nicht bestätigen, bei läuft die 0.7.11 auf dem ESP8266 seit mehr als 10 Tagen. Habe auch nur einen HM1200 Wechselrichter dran, MqTT ist aktiv.

MartinBachmannHD, 2023-07-20:
...aber irgend scheint doch zu sein, ich habe jetzt ein paar Tage 0.7.11 laufen gehabt, ging schon so mit etwas Geduld auf ESP8266, Aber nach den ca 5 Tagen hat das UI praktisch gar nicht mehr geantwortet. Mit viel Geduld konnte ich online zurück auf 0.7.3 (das läuft komplett stabil und performant). Ob da irgendwelche 'Reste' von meinen Versuchen mit den verschiedenen Versionen den Speicher belegen? Factory Reset mal drücken? Ich vermute aber, dass das nur die Settings beinflusst, nicht die 'Reste'. Ob ihr unter System noch mehr Werte auflisten könnt? CPU Auslastung und Temperatur?

kiu77, 2023-07-20:
Ich weiß, es klingt nach Voodoo, aber ich würde da sogar 0.6.9 mit Löschen per Webinstaller draufmachen, dann auf 0.7.11 updaten und die Settings neu machen.

reserve85, 2023-07-20:
Von mir auch noch die Rückmeldung: bei mir läuft die 0.7.11 seit 11.7. stabil. (ESP32)

Strohhutpat, 2023-07-24:
Bin jetzt auf ESP32 gewechselt, jetzt ist alles schnell und stabil

closed 2023-08-05 by lumapu