lumapu/ahoy #1024: Update von 5.66 auf höher geht nicht

kuttek, 2023-07-13:
## Hardware
  * [ x] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [x ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [x ] circuit board
* [ ] external antenna (SMA)

### Power Stabilization:
* [ x]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: _._.__
Github Hash: _______

### Build & Flash Method:
* [ ] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

Moin,

wie schon in der Überschrift steht geht ein Update auf 0.6.0 oder 0.6.9 leider nicht update wird angestoßen mit 
Passenden zip läuft durch und nach Neustart ist immer noch die 0.5.66 drauf, es kommt keine Fehlermeldung oder was
anderes die Übersicht geht einfach nach ca. 30 sec  in nicht erreichbar beim neuen einloggen dann alles wieder wie davor
keine Änderung auf 0.6.0 oder 0.6.9.

Ollipop030, 2023-07-13:
Moin,

hast du die *.zip auch vorher entpackt?

kuttek, 2023-07-13:
Klar doch ist zwar Warm hier aber sooo Warm ist es doch nicht ...hehe
Habe jetzt mal die Prometheus genommen und parallel dazu im Browser noch ein Paar Einstellungen zugelassen und Schwups ging es kann jetzt nicht sagen welches der beiden Sachen gewirkt hat.
Die Prometheus kann auch genutzt werden wenn man das nicht nutzt?

knickohr, 2023-07-13:
Wer solche Versionssprünge macht sollte komplett neu flashen und vorher erasen.

rmayergfx, 2023-07-13:
@knickohr 
Stimme ich soweit zu, aber man sollte vorher auf jeden Fall ein Backup seiner Settings machen. Sinnvollerweise auch noch ein paar Screenshots von den eigenen Einstellungen. Dann einfach über Web die aktuelle Stable einspielen.

knickohr, 2023-07-13:
Das Backup wird Dir unter Umständen auch nix helfen. Auch da wurde einiges geändert.

rmayergfx, 2023-07-13:
Die wichtigsten Eckdaten kann man  aus dem ahoy_setup.json mit jedem Texteditor auslesen, egal ob sich die Version geändert hat oder nichts.  Besser als kein Backup und man muss sich alles erst wieder zusammensuchen. Ich lege mir für jede Version ein Backup mit Datum und Versionsnummer ab, sicher ist sicher.

@lumapu 
Könntest du bitte beim Export der Settings per default Datum_Uhrzeit und Versionsnummer mit einpflegen?
z.b.
2023.07.13_v0.7.11_ahoy_setup.json

closed 2023-08-05 by lumapu