lumapu/ahoy #1026: Hoymiles HM300 speist nicht zuverlässig ein - Nachteinspeisung

mallkuss2, 2023-07-15:
Hallo zusammen,

habe eine AhoyDTU zusammen mit einem HM300 an einem 24V Akku (LiFePo4) hängen und will über Nacht beispielsweise 100W einspeisen um den Eigenverbrauch zu drücken.

Der Ahoy funktioniert auch wirklich gut, kann auch das Power LImit super setzen etc.

Allerdings sehe ich oft die Meldung dass der Hoymiles da ist ("available") aber "not producing".
Ich sehe eine DC Eingangsspannung von gut 26V, Netzspannung und  - frequenz scheinen normal. Er zieht einfach keinen Strom aus dem Akku. Das passiert beim Testen auch tagsüber, ist völlige Lotterie ob er nun was tut oder eben nicht.

Ich habe gesehen dass man einen "Serial Log" aktivieren kann, sollten da dann die Errorcodes des Hoymiles zu finden sein?
Also sowas wie DC Spannung zu niedrig, oder Netzfrequenz zu hoch? Wo schreibt er den log hin, wie kann ich den sehen?

Aktuell habe ich ihn testweise an einem echten Solarpanel, da rennt er mit gut 200W bei 29.8V gerade unauffällig.

Danke für  eure Hinweise, super Projekt! Ihr wollt nicht zufällig sowas auch für Envertech machen? :)

Viele Grüße
Markus

## Hardware
  * [ ] ESP8266
  * [X] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [ ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [X] external antenna (SMA)

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: 0.6.9_
Github Hash: _______

### Build & Flash Method:
* [ ] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

knickohr, 2023-07-15:
Du hast Dir die Antwort schon selbst gegeben 😉

Wie Du bemerkt hast, läuft er mit 29V. das ist auch innerhalb seines MPP, wie man aus den Datenblättern lesen kann.

Fehlercodes kann Ahoy leider nicht so schön ausgeben wie OpenDTU. Bin mir aber sicher das ein Fehler in Bezug auf Unterspannung kommen wird.

Ach, ja, die HM sind nicht für eine Einspeisung per Batterie konzipiert !  Also bitte keine Issues in Bezug auf Ahoy oder OpenDTU aufmachen wenn beide Firmware korrekt funktionieren.

mallkuss2, 2023-07-15:
Hallo Knickohr,

danke dir für deine Antwort!
Laut Datenblatt hat der HM300 eine Anlaufspannung von 22V und einen MPPT Bereich von 16-60V. Sehe da keinen Denkfehler....?
ok dann würde ich hier mit OpenDTU weiter kommen, danke für den Hinweis.

Ui, die HM sind nicht dafür geeignet? Dann habe ich das falsch verstanden, welcher WR wäre denn dann dafür geeignet?

knickohr, 2023-07-15:
Laut meinem Datenblatt nicht 😲

Ich denke Du hast MPPT-Spannungsbereich und Betriebsspannungsbereich verwechselt. Klar, er funktioniert von 16 bis 60V aber eben halt nicht richtig. Korrekt einspeisen tut er im Bereich von 29 bis 48V und in diesem Bereich sollte man ihn dann auch „füttern“.

Wie gesagt, Ahoy bringt nicht sehr viele Fehlercodes, da ist OpenDTU besser, zumal es die Fehler auch speichert. Aber warum Fehler suchen wenn wir ihn gefunden haben ?

Ein Solarwechselrichter ist eben kein Inverter für Batterieeindpeisung. Auch wenn die Hardware vermutlich die gleiche ist. Im Garantiefall will davon dann niemand was wissen 😱

mallkuss2, 2023-07-15:
tja es kommt darauf an in welches Datenblatt man schaut:
reichelt: https://cdn-reichelt.de/documents/datenblatt/D400/HO_HM-300_DB-DE.pdf
MPPT Bereich ab 29V
Hoymiles: https://www.hoymiles.com/wp-content/uploads/downloadupload/Datasheet_HM-300-350-400_LA_EN_V202206.pdf
MPPT Bereich ab 16V
Haben die ihre eigenen Datenblätter nicht im Griff? 
Der neue HMS 300 scheint es zu können:
https://www.hoymiles.com/wp-content/uploads/downloadupload/Datasheet_HMS-300-350-400-450-500_EU_EN_V202201.pdf

knickohr, 2023-07-15:
😂

Jajaaa, die Chinesen 🤪

ich kann nur bestätigen was meine 13 HM-300 machen. Und unter 29V gehen sie in den Status 1 (available but not producing). Es kommen dann nur noch ein paar Watt raus und der Wirkungsgrad geht in den Keller.

(Ich traue dem Datenblatt des HMS nicht, da werden alle in einen Topf geworfen 😲)

mallkuss2, 2023-07-15:
hehe, ja die Chinesen! Dachte das passiert nur bei Lumen und Watt....
danke dir für deine Infos, das ist doch mal ne Ansage!
Ich suche mir einen DCDC Up-Converter 24 => 48V und melde mich wieder :-)

knickohr, 2023-07-15:
Aufpassen !  Dein Step-Up muß eine Strombegrenzung haben !  Ansonsten fliegt er Dir 1 Millisekunde nach dem Anschließen um die Ohren. Die Kondensatoren im WR sorgen sicherlich dafür 😱

Auf 48 würde ich nicht gerade gehen, beim ein paar Volt darunter.

mallkuss2, 2023-07-15:
danke dir für den Hinweis.
Habe jetzt testweise mal ein Netzteil mit 40V als Quelle dran und er speist sauber ein.
Wenn ich per AHOY das Power Limit hochsetze hört man sofort den Lüfter vom Netzteil hochdrehen :-)
MIr scheint also das Problem ist gelöst! Danke nochmal :-)

kiu77, 2023-07-15:
Falls Du die 24V aus zwei 12V-Akkus in Serie hast, besorge Dir einen dritten 12V-Akku. Ein Stepup-Regler bringt zusätzliche Verluste. Der Vorteil einen solchen Reglers aber ist, dass in der Regel der Ausgangsstrom begrenzbar ist, was den inverser schützt, denn die mögen keine hohen Kurzschlussspannungen. 
Nachteil der Lösung mit Stepp-Regler: Wenn nicht einstellbar ist, dass er z.B. bei <22V Eingangsspannung abschaltet, saugt er unter Umständen Deinen Akku so leer, dass er tiefentladen wird. Das wird nicht lange gut gehen und er ist defekt.
Ganz so trivial ist das also nicht (aber machbar).

knickohr, 2023-07-16:
Aber bitte :  Wenn schon 2 oder 3 Akkus, dann mit Balancer. Ansonsten ist einer ganz schnell hin und die anderen folgen schnell.

kiu77, 2023-07-16:
Bei Li-Akkus ja, bei Blei müssen sie nur gleich sein.

knickohr, 2023-07-16:
Da widerspreche ich dir aufs schlimmste !  Sollten die Zellen nicht absolut gleich sein, sowohl von der Kapazität, als auch von der Spannung, Temperatur und dem Alter, dann ist eine Zelle ziemlich schnell hin und über kurz oder lang sterben die anderen Zellen immer schneller hintereinander.

Bei Lithium-Akkus wird das unter anderem gemacht weil sie Dir sonst ziemlich schnell um die Ohren fliegen 😲

kiu77, 2023-07-16:
Da widerspreche ist Dir noch wesentlich schlimmer!
:D
Da habe ich einfach etwas mehr Ahnung von der Materie, als alter Vollblutelektrioniker. Bleiakkus balancieren sich selbst. Man muss lediglich gleiche Akkus (auch gleich alte) verwenden, dann ist das Null Problemo.
Ansonsten bräuchten auch alle Bleiakkus interne Balancer, denn da sind bei 12-V-Typen immerhin 6 Zellen einfach so in Serie geschaltet

mallkuss2, 2023-07-17:
Danke für die vielen Tipps :-)
Ist ein LiFeapo4 Akku 8S2P mit balancer und BMS. Im Normalfall speist.er über einen 1200W Victron Phoenix eine kleine Insel. Jetzt soll er auch bisschen was ins Haus schieben.

kiu77, 2023-07-17:
Der Akku ist sicher gut, hat aber, wie knickohr ja schon dargelegt hat, hat er klar zu wenig Spannung.
Immerhin kannst Du sicher sein, dass ein Step-up-Wandler den Akku nicht leersaugt und tiefentlädt, weil er ein BMS hat, dass das verhindern sollte. 
Sowas in etwa müsste es dann tun:
https://www.ebay.de/itm/235088830470?hash=item36bc62e006:g:H74AAOSwieRkr6XX&amdata=enc%3AAQAIAAAA4GavURy7DpCScfg0R%2Fs785rmMluLYqgZQ7fSX1r1WR9gwMqNBzyKRh1AIgeumHOXN7GOiVJsxOGItMjBswPkRv3XoZy0KmHS5yNFKLnq30eGHP46SfG3Fqfgx5Fc5hvuwkk7S8hcKJ2aN%2FolkVDSlg06QumvV3ST5E26mJDFOFKq3wQ0gkkMNJB6vsDE%2BuhhyiGFZvmg6rNS20hOi8GXCyDYFJhhJdaqbu7CwDjHcXrw1%2BlcUN5IMV1Msx7mQ8lPMcLhC6AGmUjsIUqZbu2%2Bb7SJwym1O75XkqwasNHIOXBU%7Ctkp%3ABk9SR77Q_MCsYg

Mehr als 144W brauchst Du ja nicht.

closed 2023-08-05 by lumapu