lumapu/ahoy #1028: [Bug] Falsche Berechnung vom Wechselstrom

GHolli, 2023-07-16:
### Hardware
ESP8266

### Version / Git SHA: 
Version: 0.6.9
Github Hash: 15ec6a0

## Error description

Soweit ich es verstanden habe, werden alle "Live"-Werte vom Wechselrichter zur Verfügung gestellt, ausgenommen die "Irradiation" und der AC-Strom.
Der Strom wird folgendermaßen berechnet:
I AC = (AC Power)/(AC Voltage)
Dies ist aber nicht ganz richtig, weil eine potentiell vorhandene Blindleistung den Strom erhöht. Die korrekte Berechnung lautet daher:
I AC = (AC Power)/(Power Factor)/(AC Voltage)

Schwer zu sagen, wie genau diese Werte sind und ob man diese auch messtechnisch bestätigen kann. Was ich nämlich sehr verdächtig finde ist, dass sich der Wirkungsgrad meistens um die 95 % bewegt, unabhängig davon ob 5 oder 500 W eingespeist werden. Irgendwie habe ich den Eindruck, dass der Eigenbedarf des Wechselrichters in der AC-Leistung inkludiert ist und den ausgegebenen Wirkungsgrad damit nicht beeinflusst.

rmayergfx, 2023-07-19:
Bitte mal den Titel ändern, der Wechselstrom wird nicht berechnet! 
Also ich sehe in den TechDocs zum Hoymiles folgendes:
[image]
Somit wird die Frequenz (Hz) und die elektrische Spannung (V) direkt vom WR geliefert.

Nun erklärst du uns mal bitte mit den richtigen Fachbegriffen, was aus deiner Sicht falsch berechnet wird.

GHolli, 2023-07-20:
Ich habe die Bezeichnungen vom Webinterface verwendet, um Verwechslungen zu vermeiden:
[image]
Die Stromstärke (AC) wird falsch berechnet, Leistung (AC) dividiert durch Spannung (AC): 5,3/238,2 = 0,02 A
Korrekt wäre: 5,3/0,23/238,2 = sqrt(5,3²+22,1²)/238,2 = 0,10 A

lumapu, 2023-07-22:
Der Wechselrichter liefert AC Spannung, Strom und Leistung.
Er liefert nicht die YieldDay und YieldTotal Werte. Dieser werden fälschlicherweise aus den DC Yields summiert.
Geplant ist eine Option einen Wirkungsgrad abgeben zu können, damit könnte man die AC Yields korrigieren.

knickohr, 2023-07-25:
Was heißt korrigieren ?  Wenn ich nur stumpf 95% eingebe, dann ist das genauso Glaskugel wie es jetzt schon ist 🤔

lumapu, 2023-07-25:
naja, aber wahrscheinlich 4% näher an der Realität, es bleibt jedem selbst überlassen hier einen Wert kleiner als 1.0 einzutragen

GHolli, 2023-07-27:
Verstehe, der Wert kommt auch direkt vom Wechselrichter - hatte das falsch interpretiert, dachte er wird berechnet:
`float calcIpv    (float *measure) { return (measure[10] != 0 ? measure[12]/measure[10] : 0); } // IDX_PAC/IDX_UAC`
Damit hat sich der PR erledigt...

Aber es wäre vielleicht interessant ein "known Hoymiles issues" Kapitel in die Dokumentation aufzunehmen:
- Wechselrichter berücksichtigt beim Wert "Total Production" den Wirkungsgrad nicht, sondern summiert die DC Energie
- Der Wechselstrom wird vom Wechselrichter mittels Division der Wirkleistung durch die Spannung errechnet, die Blindleistung/der Leistungsfaktor bleibt dabei unberücksichtigt
- Der vom Wechselrichter angegebene Wirkungsgrad ist unglaubwürdig
- ...

lumapu, 2023-07-29:
mache hier nochmal auf, finde die Idee der known issues gut.
Der Wirkungsgrad wird von Ahoy bestimmt, wenn ich nicht irre - das muss ich nochmal prüfen.

knickohr, 2023-07-29:
Also der Wirkungsgrad ändert sich ein wenig. Von Morgens mit wenigen Watt Produktion, da liegt er schon mal bei 93% bis Mittags zum Maximum, da hat er dann knapp 96 und abends geht’s dann wieder runter. Wenn die Sonne recht draufknallt auf den WR, dann meine ich auch das er kaum 95% hat, muß das aber nochmal verifizieren.

Jetzt ist die Frage, ob das der Wirkungsgrad DC-seitig ist, oder ob es da die Umwandlung auf AC auch schon mit rein gerechnet wurde ?

Ich persönlich würde tippen, es ist alles DC. Man bräuchte ein gutes Wirkleistungsmeßgerät das auch wirklich Wirkleistung sauber mißt.

lumapu, 2023-07-30:
der Wirkungsgrad bezieht sich nur auf die Wirkleistung, es wird einfach die AC Leistung durch die DC Leistung geteilt, daher gibt es keinen AC oder DC Bezug

GHolli, 2023-07-31:
Danke, verstehe, der Wirkungsgrad wird also berechnet. Die Berechnung macht absolut Sinn, aber ich habe etwas Zweifel an den Ausgangsdaten.
Also, ich habe folgende Werte beobachtet:
- 92,11 % @ 3,5 W (AC)
- 93,94 % @ 9,3 W
- 95,02 % @ 700,4 W
- 96,97 % @ 6,4 W

Ich würde einmal vermuten der Wirkungsgrad sollte bei sehr kleinen Leistungen drastisch einbrechen, da ich von einem gewissen Grund-Eigenbedarf ausgehe.
Die DC-Leistung sollte ja relativ einfach mit einem halbwegs vernünftigen Multimeter zu verifizieren sein, AC-seitig wird es schon schwieriger. Ich habe zwar einen True RMS Multimeter, aber den Leistungsfaktor kann ich damit auch nicht messen. Eine Messung bei höherer Leistung (mit vernachlässigbaren Blindleistungsanteil) sollte damit aber möglich sein.
Am genauesten wäre wahrscheinlich eine Messung mit einem alten Ferraris-Zähler (wenn man nicht gerade ein kalibriertes Messgerät zur Hand hat).
Aktuell habe ich eine Shelly installiert. Die zeigt immer eine etwas kleinere Leistung an als der Wert, der vom Wechselrichter kommt. Aber schwer zu sagen, wie genau die misst, eine Toleranz ist nicht spezifiziert (und das Produkt aus Strom mal Spannung ist kleiner als die angezeigte Leistung - das erhöht nicht gerade das Vertrauen in das Gerät).

gone-for-coding, 2023-08-20:
Nachdem ich heute den Issue hier gefunden habe, habe ich ein bisschen mit meinen Daten gespielt und versucht herauszufinden, wie die Effizienz real sein könnte.

Wenn man die Momentanleistung AC aufsummiert und dann durch die Anzahl der Messpunkte pro Stunde teilt sollte das in etwa die Wattstunden ergeben.

Vergleiche ist jetzt pro Tag den Maximalwert von YieldDay mit dem, was ich oben errechnet habe komme ich auf Werte von 95% +- ein paar Prozentpunkte. Man fährt mit 95% wahrscheinlich nicht allzu falsch.

Das natürlich unter der Annahme, dass die Leistung AC hinreichend genau ist und sich der Fehler bei einer Abtastung alle 30 Sekunden in Grenzen hält.

DanielR92, 2023-08-25:
Wird die Effiziens nicht vom WR ausgelesen, oder hat sich was geändert? 
[image]

lumapu, 2023-08-28:
nein steht glaube ich auch weiter oben. Die Effizienz wurde hier in Frage gestellt. Es hat sich aber hausgestellt, dass sie relativ gut stimmt

gone-for-coding, 2023-12-21:
Hier https://github.com/lumapu/ahoy/issues/1243 wurde sich auch noch Mal mit dem Thema Wirkungsgrad beschäftigt, vor allem ob die Yields DC oder AC sind. Ergebnis dort ist, dass die Yields alle AC, also nach Abzug des Wirkungsgrades sind.

BiTy1024, 2024-09-11:
Hey guys,

Where do i find the documentation for the mesuring unit inside the microinverter?

lumapu, 2024-10-07:
I don't know if there is any documentation. Most things are reverse engineered, so there is no official documentation.