lumapu/ahoy #1029: Vorteile 0.7.11 zu 0.6.12

firedp, 2023-07-16:
## Hardware
  * [ ] ESP8266
  * [x ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [x ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [x ] external antenna (SMA)

### Power Stabilization:
* [ x]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: _._.__
Github Hash: _______

### Build & Flash Method:
* [ ] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

knickohr, 2023-07-16:
https://github.com/lumapu/ahoy/blob/development03/src/CHANGES.md

firedp, 2023-07-16:
Vielen Dank!
Hatte ich vergessen :-(

jurbot, 2023-07-25:
Hallo Zusammen, meine 0.6.9 macht massive Probleme. Wie kann ich auf die Zwischenversion updaten bzw wo/wie kann ich die zb v0.7.11 herunterladen? 

Kann man auf den ESP von ahoy DTU auch die openDTU flashen? Ist doch bestimmt die gleiche Hardware.
Danke!

lumapu, 2023-07-25:
Ahoy Development Version: https://nightly.link/lumapu/ahoy/workflows/compile_development/development03/ahoydtu_dev.zip

OpenDTU geht nur mit dem ESP32

closed 2023-08-05 by lumapu