lumapu/ahoy #1033: ahoy doesn't save inverter info

Drakogol, 2023-07-19:
## Hardware
  * [ X] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [ x] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [x ] circuit board
* [ ] external antenna (SMA)

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: 0.6.9
Github Hash: _______

### Build & Flash Method:
* [X ] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

Hi,

I've read several threads here and tried different things but nothing here describes my exact problem.

I've installed everything properly and after a little resoldering the esp and nrf works . 
But when I go to the system settings and want to add my inverter, it doesn't want to save the info. The save button just goes grey/darker blue and nothing happens. I have no idea what's wrong. Is the capacitor really necessary? The dtu is right beneath the inverter

lumapu, 2023-07-19:
It seems like you have some issues with your browser. Have you tried a different one? 
The capacitor is needed for the radio module. The reason behind is the high current during sending of the module. Remember that the NRF is connected to the 3.3V power rail of the ESP, which has an onboard DCDC converter. So the DCDC will have higher load. We got many reports from users where the EPS randomly reboots, without the cap. Since the recomandation it became one of the initial questions because of the higher system stability.

Drakogol, 2023-07-19:
Nice thanks for the tip with the browser. It seems ahoy doesn't like spaces in the name field for the inverter. 
I'll update tomorrow if it works. Sun is down so no inverter ;-)
Meanwhile I'll buy the capacitor and will see how to attach it

Drakogol, 2023-07-20:
Thanks it works like a charm even without the capacitor

closed 2023-07-20 by Drakogol