lumapu/ahoy #1037: Ahoy V0.6.9 doesn't answer after some time

jolo1581, 2023-07-22:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.6.9

### Github Hash

15ec6a0 

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

MQTT broker added

### Error description

After some time my AhoyDTU device turns off. It is not available any more and only a hard reset will turn it back alive.
The time between runnung and broke down is different. It can be more than 7 days or after a few hours.

Maybe there is a solution.

rmayergfx, 2023-07-22:
Tell us which board you are using and what kind of Powersupply. Did it ever run smooth? Is your Network stable? Do you have a seperate SSID for 2.4GHz?
Try newer Version 0.7.6 https://github.com/lumapu/ahoy/actions/runs/5300414822

jolo1581, 2023-07-22:
Board: Wemos D1
SSID:Not separate, it is switch off over the night

Power is stable.

When the error appears, there is no connection for web interface. Seems to me like an endless loop with no WDT timeout

jolo1581, 2023-07-23:
I checked soldering. There was a cold solder joint at 100 uF Capacitor.
I will check if my system works now stable.

Think we can close this issue.

closed 2023-07-23 by jolo1581