lumapu/ahoy #1038: [Bug]

reinhard-brandstaedter, 2023-07-22:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.6.9

### Github Hash

15ec6a0

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Basic setup:
- Inverter HM600
- no pause during night
- sending via MQTT
- setting inverter limit every 20s via MQTT

### Error description

AHOY DTU is running rather stable for a long time, but sporadically it seems to loose it's connection to WiFi, though my AP still is showing it's there. However the webfrontend is not reachable and setting a limit via MQTT doesn't work anymore (assuming it lost connection completely).
I suspect this has something to do with setting the limit rather frequent, as the issues started around the time I started doing that ...

rmayergfx, 2023-07-22:
Tell us which board you are using and what kind of Powersupply. Did it ever run smooth? Is your Network stable? Do you have a seperate SSID for 2.4GHz?
Try newer Version 0.7.6 https://github.com/lumapu/ahoy/actions/runs/5300414822

lumapu, 2023-07-22:
How are your intervals? You said it started once you began to limit via MqTT.
How many inverters do you have configured on your ESP?
Which kind of ESP do you have?

amici, 2023-07-23:
I have the same problem, with the same setup, D3mini, HM800, nRF24L01+, v0.6.9.
I think it also started when I tried to set up the MQTT. I also thought it might be related to too short MQTT period, because I did initially set it to 5s.
In the end, the ahoy became so unresponsive that I reuploaded the same bios and then it worked very stable - until a few days later I tried again setting up some public MQTT. Although, next time I set the period to "0", so it should send data as soon as it gets it from the inverter - which might not be too long too. 

I see Ahoy attached to my router all the time, but when I access it, it is either loading for a while and then fails, or it loads just the basic screen and then all the later, background data calls fail. It can't even load all the submenu items.
At this point not even the reboots help.

What I also find problematic is that sometimes on reboots, it won't nicely start up, unless I unhook the antenna first, reboot, wait a bit, then connect the antenna.

I will try the mentioned new version.

reinhard-brandstaedter, 2023-07-24:
@lumapu @rmayergfx 
The board is a D1mini with a USB mini powersupply 2A.
It's communication via a dedicated IoT WiFi at 2.4G. The WiFi should be rather stable I haven't seen any such disconnects on other devices (I have several ESP-01 8266 based sensors and controllers running mostly esphome). The DTU sitz next to an AP (2m).
Yes it actually ran smooth for a very long time, while I wasn't limiting it and it was still doing pauses between sunset and sunrise.

amici, 2023-07-24:
FYI, it got blocked again now, and when I call it (either by its IP or the name `ahoy-dtu`, on `http`), I get the `NS_ERROR_CONNECTION_REFUSED`:
[image]

Could it be that there is only some problem with the web server process on the ahoy dtu?

mkrasselt1, 2024-10-07:
can you check if the device opens its hotspot or not?
I have a similar problem with my esp32 + cmt setup, but the device just disconnects from the accesspoint, and reverts to setup page

stefan123t, 2024-10-27:
@reinhard-brandstaedter how do you set the limit using the REST API / via MQTT topic ?
And especially how frequently is this executed ?
Do you stop executing this in the night or do you wait until you have a response from the AhoyDTU that the inverter actually accepted your limit, before you send the next Active Power Limit command.

@lumapu do we have an Active Power Limit command queue or do we otherwise make sure that we only send the last APL value to the inverter ?