lumapu/ahoy #1040: DEV 0.7.13 -- fix prometheus build -- Export Settings falsches Datum

rmayergfx, 2023-07-22:
Funktioniert leider nicht so wie gewünscht:
[image]

Kurz nach einem Reboot erscheint das:
na_v0.7.13_ahoy_setup.json
Erst wenn ich manuell auf NTP Server  SYNC NTP gehe
[image]
Wird das Datum und die Uhrzeit mit abgespeichert, leider total verkehrt:

[image]

Und das obwohl im Interface Datum und Uhrzeit richtig sind:
[image]

lumapu, 2023-07-22:
klar, guter Hinweis. Bei meinen Tests hat's funktioniert, ich teste nochmal.

Ollipop030, 2023-07-24:
Guten Morgen,

gerade die 0.7.15 installiert. Beim Export wird leider immer noch nicht die Uhrzeit mit übernommen (na_v0.7.15_ahoy_setup). Zeit ist über NTP synchronisiert und läuft auch richtig.

lumapu, 2023-07-24:
fix kommt heute noch 😉

rmayergfx, 2023-07-25:
Mit Version 0.7.16 habe ich aktuell keine Probleme mehr mit dem Export, Datum und Uhrzeit passt nun bei mir.
@lumapu 
Dankeschön.
Kommt bei SYNC FTP noch ein Update des Status in einer späteren Release? Hier wäre es sinnvoll auch das aktuelle Datum/Uhrzeit einzublenden.

closed 2023-08-05 by lumapu