lumapu/ahoy #1041: [Bug] Display contrast Nokia 5110 eingeschränkt (1-100, aber ~130 notwendig)

You69Man, 2023-07-22:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [X] I will attach/upload an Image of my wiring

### Version

0.7.13

### Github Hash

ac57e169

### Build & Flash Method

VSCode - Platform IO (build & flash)

### Setup

Display config - Luminance -> 130

### Error description

Ich habe heute mein bereits seit ca. 1 Woche problemlos laufendes AhoyDTU (Basis Wemos mini D1 ESP32, nRF24L01+, MQTT ) um ein Nokia 5110 erweitert (Eigenbau, gelötet). 
Die Gesamtschaltung einschließlich Verdrahtung des Displays habe ich zunächst mit Original Examples der u8g2 Library von olikraus getestet und die Demos (Text + Grafik) haben auf Anhieb funktioniert.
Nachdem ich dann den aktuellsten Develop Build (0.7.13) aufgespielt habe, und über die Web-Setting das Display ordnungsgemäß konfiguriert habe, blieb das Display leer (sonst funktioniert weiter alles).
Auf den Suche nach dem Problem bzw. den Unterschieden zwischen dem AHoy-Sourcecode und den u8g2 Examples habe ich festgestellt, dass die u8g2 Examples keine Kontrast-Einstellung vornehmen (.setcontrast()), und dennoch per default gleich einmal ein optimales Bild gegeben ist. Bei AhoyDTU wird hingegen der Luminance Wert der Settings der Display-Library als Kontrast-Wert übergeben. Nun ist der Luminance-Wert aber leider auf 0-100 beschränkt (offenbar als Prozent-Wert gedacht). .setcontrast() der u8g2 Library erwartet aber einen Wert 0-255. Ich habe daher probehalber den eingeschränkten Wertebereich der HTML-pages auf 0-255 erweiter, und siehe da, mit einem Wert von ca. 130 ergibt sich ein klares Bild.

Wenn ich also nicht irgendetwas falsch verstanden habe, fehlt meiner Meinung noch eine Umrechnung des Prozentwertes in den 0-255 Wertebereich für die Kontrasteinstellung.

lumapu, 2023-07-22:
danke für den Hinweis, Nokia Displays werden nur sehr selten mit Ahoy eingesetzt.

You69Man, 2023-07-22:
Dachte ich mir schon, weil man auch kaum Infos zum Anschluss findet. Aber nachdem ich es gerade griffbereit hatte, war es einen Versuch wert, und es funktioniert ja für den Zweck auch gar nicht schlecht, wenn einmal die Settings passen. 
Mit OLEDs habe ich eher schlechte Langzeiterfahrungen gemacht. Ohne Screensaver sind die oft nach kurzer Zeit defekt. Daher benutze ich OLEDs nicht für Daueranzeigen sondern aktiviere sie nur bei Bedarf, z.b. durch Tastendruck, und deaktivieren sie wieder nach einer Minute. So halten sie jahrelang.
EInk habe ich bisher noch nie probiert.

knickohr, 2023-07-23:
Ach spannend !  Endlich mal jemand, der auch den OLED Hype so klasse findet 🤪

Ja, das Nokia wird leider sehr selten verwendet und Dich ist es bis jetzt das dimensional größte Display, auch wenn die Auflösung recht mau ist.

Was ich aber am Nokia festgestellt habe, das hin und wieder der Leitgummi schlechten Kontakt hat und deshalb das Display recht flau wirkt, Ei paar mal draufdrücken brachte aber bei mir den erforderlichen Kontrast. Habe aber schon gehört das es weitere gibt, die den Kontrast nicht stark genug bekommen.

Die Idee mir dem Taster für die OLEDs würde ich sogar auf das Nokia ausweiten. Damit könnte man ebenfalls die Hintergrundbeleuchtung des Displays steuern. Ja, muß natürlich in der Software entsprechend codiert werden. Leider ist das beim ESP8266 ein Problem wegen den fehlenden GPIOs, beim ESP32 aber durchaus machbar.

You69Man, 2023-07-23:
Bei mir ist es definitiv kein Kontaktproblem des Displays, sondern die Kontrastwerte werden von der Software zu niedrig eingestellt (ich glaube per Default auf 60). Durch meinen lokalen Workaround (Aufhebung der 0-100 Grenze in der Settings HTML page und Einstellung eines Wertes von 130 funktioniert es, wie gesagt, prächtig. Vielleicht war das auch bei anderen die es mit diesem Display  versucht haben schon das eigentliche Problem. Vielleicht bin ich aber auch nur der Erste mit Nokia!?

Die Änderung in der Software ist für das Nokia an sich eine Kleinigkeit, aber da ich nicht weiß, wie sich das auf die anderen Displays auswirkt, welche den Luminance-Parameter evt. anders interpretieren, sollte das besser jemand machen, der hier mehr Überblick hat.

knickohr, 2023-07-23:
Ich habe auch recht wenig Ahnung von C, ansonsten würden hier einige mehr Displays supported werden 😅

Vor allem komme ich mit diesem Git und PlatformIO überhaupt nicht zurecht 😢

You69Man, 2023-07-23:
Ich habe gerade noch ein wenig recherchiert. Das Problem dürfte erst ca. vor 3 Monaten (d.h. nach der V0.6.9) durch das Refactoren der DisplayMono Klasse (Auftrennung in mehrere .h File je Displaygröße) entstanden sein.     
In der ursprünglichen DisplayMono Klasse war nämlich das Nokia 5110 von der Kontrasteinstellung ausgenommen, d.h. es wurde gar keine Kontrasteinstellung vorgenommen. Da die Nokia Displays typischerweise ab Werk bereits einen optimalen Kontast haben, gab es vermutlich kein Problem.
```
 if (3 != mType)      //   3=Nokia 5110
            mDisplay->setContrast(mLuminance);
``` 
Erst durch die neue DisplayMono84X48 Klasse kam dann das setContrast auch für das Nokia dazu, und stellt nun leider einen zu niedrigen Kontrast ein, sodass auch bei (den vermutlich wenigen) AhoyDTUs mit Nokia Display beim nächsten Software-Update "das Licht ausgehen wird" (bzw. der Kontrast), falls das nicht vorher behoben wird.
Man könnte also im DisplayMono84X48 evt. auch einfach das setContrast wieder rausnehmen. Oder eben wie gesagt, den Range entsprechend anpassen.

knickohr, 2023-07-23:
Raus nehmen würde ich es nicht. Wir hatten schon den Fall, da war der Kontrast zu flau und wir haben es dann mit rein genommen.  Aber wenn der Wertebereich nicht komplett verwendet werden kann, dann sollte man es ändern. Jetzt stellt sich für mich die Frage ob es bei den OLEDs nicht auch eingeschränkt ist 🤔

You69Man, 2023-07-23:
Das ist durchaus möglich, da bei allen DisplayMono Klassen die gleiche SetContrast Funktion der u8g2 Library verwendet wird. Und die verlangt eben Werte von 0-255.

kiu77, 2023-07-23:
Etwas off topic:
Bei den 1,3" OLEDs (SH1106) kann man mit dem Kontrastwert in einem bestimmten Bereich die Helligkeit einstellen.
Bei meinem 2,42" OLED ist das folgenlos (typisch, wenn man nicht das richtige Display verwendet ;-)
und es reagiert nicht auf diese Werte. Dafür läuft es sowohl mit der Einstellung SH1106 als auch SSD1306 in 128x64 gleich gut (keine Darstellungsfehler).

You69Man, 2023-07-23:
@lumapu 
Danke für den Versuch das Kontrast Problem beim Nokia zu fixen!

Ich fürchte nur, du hast das falsche Plugin erwischt. Das Nokia 5110 wäre das 84x48 mit dem PCD8544 Controller gewesen (Display_Mono_84X48.h), dein fix ist aber im Display_Mono_64X48.h gelandet.

Möglicherweise muss die Umrechnung aber ohnehin für alle Mono-Display Typen gemacht werden, weil, soweit ich es sehe, wird überall die setContrast Funktion der u8g2 Library verwendet, und in der Doku dort steht unabhängig von Display-Typ:
Description: Set the contrast or brightness for the display (if supported). Range for 'value': 0 (no contrast) to 255 (maximum contrast or brightness).

Der Range 0-100 dürfte also überall falsch sein. Für die OLEDs könnte es allerdings bedeuten, dass diese nach dieser Änderung bei gleichen Settings deutlich heller leuchten, und dann der Wert manuell zurückgestellt werden muss (wegen der Lebensdauer).

lumapu, 2023-07-23:
oh 🤫 sorry, wird überprüft 😇

knickohr, 2023-07-23:
Ja, der Wertebereich ist für alle Mono 0 bis 255, oder 0 bis 100%. Kann man für alle Mono machen, dann würde einmal reichen.

lumapu, 2023-07-23:
ich würde es direkt im Setup machen, dann gilt es automatisch für alle, wie @knickohr es richtig angemerkt hat. Die vorherige Änderung der Anpassung von 0 .. 100 auf 0 .. 255 mache ich wieder rückgängig. Das heißt es wird sehr bald eine `0.7.15` geben 😀

You69Man, 2023-08-05:
Für mich ist die Änderung ok und erledigt. Falls nicht noch irgendwelche Einwände von Leuten mit anderen Displays kommen, kann man das Issue closen. Danke!

knickohr, 2023-08-05:
Dann mach zu das Ding 😉

You69Man, 2023-08-05:
Gerne! Drei ... zwei ... eins ... und zu ist der Sack :-)

closed 2023-08-05 by You69Man