lumapu/ahoy #1042: [Bug]

exo1987, 2023-07-25:
### Platform

ESP32

### Assembly

the DTU was already assembled

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.7.16

### Github Hash

aaf685f

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

MQTT intervall = 0
inverter intervall = 5 sec
reset at midnight

### Error description

Hi zusammen,

da ich mit 0.6.9 Probleme mit MQTT hatte (hat nach einer bestimmten Zeit keine Daten mehr übertragen), bin ich auf 0.7.15 bzw. 0.7.16. Hier ist aber nach einer Weile die AhoyDTU nicht mehr erreichbar. Bei 0.7.15 war sie komplett vom WLAN getrennt und der Display war schwarz. Bei 0.7.16 ist die DTU zumindest noch mit dem WLAN verbunden und ist auch morgens (nach dem Reset um Mitternacht) wieder erreichbar. Meistens ist die DTU nach 18-20h nicht mehr erreichbar.

lumapu, 2023-07-25:
evtl. mal mit einem größeren Interval testen, ca. 15s
Normalerweise sind die ESP32 sehr stabil, inzwischen sogar auch die ESP8266.

exo1987, 2023-07-25:
Hab mal auf 10s hochgesetzt. Einen ESP8266 hatte ich davor... war eine Katastrophe was Stabilität angeht. Ich beobachte es mal! Danke!

lumapu, 2023-07-25:
echt interessant, ich habe 1x ESP8266 und 2x ESP32 im Einsatz - alle laufen sehr stabil. Beim ESP8266 ist den Kondensator und das `Power-Level` des NRF auf `MIN` oder `LOW` sehr entscheident.

exo1987, 2023-07-26:
Der ESP8266 war bei mir teilweise nur 3-4h erreichbar, dann musste ich den Stecker ziehen... Der ESP32 lief mit der 0.6.9 super. Sogar bei einer sekündlichen Abfrage sehr stabil. Dafür eben das Problem mit MQTT.

lumapu, 2023-07-26:
ja in der 0.6.9 ist noch ein "work-around" für die MqTT Lib nicht drin. Da kann nach kurzer Zeit der Heap so durchlöchert sein, dass nichts mehr geht. In der Developmentversion sollte das nicht passieren.

theunknownstarman, 2023-07-30:
Wann planst Du die nächste stable release?

lumapu, 2023-07-30:
ich denke die nächsten Tage, ich warte noch bisschen auf Feedback der zuletzt bearbeiteten Themen.
Wird auf jeden Fall höchste Zeit

Karlo49, 2023-07-31:
Ich hatte ähnliche Probleme beim Update von 0.6.15 auf 0.7.11

Hardware: ESP8266pro 16MB mit Antenne, nrf24l01+ mit Antenne, Elko 100µ und Tantal 10µ
Power Level LOW
MQTT an heap_frag auf 25
MQTT aus heap_frag auf 2

Nach dem Update lief alles ca 4 Stunden, dann Absturz, kein W-Lan zugriff.
Reset-Knopf keine Änderung
Strom aus für ca 15 sec danach wieder OK aber mit MQTT heap_frag wieder auf 25

Nach weiteren 4-5 Stunden wieder Absturz.
Reset-Knopf keine Änderung
Strom aus für ca 15 sec keine Änderung!

System für ca 1 Tag stromlos. danach wieder Ok
MQTT an heap_frag auf 2
reboot_reason Exception

Sytem mit 0.7.11 läuft jetzt so 9 Tage

Ähniche Probleme hatte ich schon 2 bis 3 mal bei größeren Softwaresprüngen aber immer mit MQTT an!

lumapu, 2023-08-05:
bitte mit aktuellem development build nochmal testen, siehe auch #1057

exo1987, 2023-08-06:
Danke! Ist aktualisiert. Ich werde berichten

Karlo49, 2023-08-07:
Update auf 0.7.26
MQTT an heap_frag auf 25
Nach ca 2 Stunden stellt MQTT den Dienst ein!
Nochmal über ESPTool  0.7.26 geflashst.
MQTT an heap_frag auf 3-4
Nach ca 2 Stunden hängt sich das gesamte System auf, kein Zugriff mehr!
Zürück auf 0.7.11

lumapu, 2023-08-07:
@Karlo49 wie viele Wechselrichter hast du dran? Verwendest bzw. setzt du das Power-limit per MqTT / REST API?

Karlo49, 2023-08-07:
1 HM1500 mit 4 Module
Kein Powerlimit

lumapu, 2023-08-08:
bitte prüfen:
* NRF24 Power-Limit bei MIN oder LOW?
* Abfrageinterval >= 15s
* Elko nahe dem NRF Modul

bei meinem ESP8266 mit einem HM1200 (auch 4 Module, im Prinzip gleicher Wechselrichter wie deiner nur leistungsschwächer) sieht die Heap-Frag. mit `0.7.26` so aus (meistens 2, in der WebGUI):

[image]

Karlo49, 2023-08-09:
Antwort auf die Fragen:
NRF24 Power-Limit auf LOW?
Abfrageinterval = 30s
Elko nahe dem NRF Modul, ja 100µF

Gedanken zu den Abstürzen bei mir mit 2 Mudulen ESP8266 pro.

Wenn die Sofware in Ordnung ist kann es nur an der Hardware liegen.
Die 3,3Volt wird auf meinen Modulen vom Regler TX6211B (Code DE=A1D) erzeugt.
Laut Datenblatt hat der Regler einen typ. Ausgangsstrom von 400mA bei 1 Volt Differenz.
Bei 1,7Volt Differenz (5V - 3,3V) ergibt das einen Strom von 235 mA.
Das reicht für ESP8266 und dem nRF24L01+ selbst bei voller Sendeleistung.
Wenn es nicht der max. Strom ist, können es nur Ein- und Ausschaltspitzen des Senders sein, die den Prozessor in den Wald schicken.
Ich habe einfach einen 2 Ohm Widerstand in die 3,3 Volt Leitung zwischen ESP und Funkmodul gelötet und jeweils ein 100µF Elko an ESP und Funkmodul befestigt.
Die Spannung fällt von 3,31 auf 3,28 Volt ab. Das ist unproblematisch.
Mit dieser Beschaltung läuft das erste Modul mit Software 0.7.26 und MQTT seit 24 Suden ohne Probleme.
Der heap_frag ist weiterhin auf 25.
Seit ca 3 Sunden läuft das 2. Modul

lumapu, 2024-05-30:
gehen die neueren Versionen besser?

stefan123t, 2024-10-27:
@exo1987 kannst Du bitte mal Feedback geben ob das Problem mit einer aktuellen Version gelöst ist und wir das Issue schließen können ?

exo1987, 2024-10-28:
Hi @stefan123t , ich bin erst letztens dazu gekommen die neue Stable zu installieren. Seitdem läuft die AhoyDTU aber problemlos. Keinerlei Probleme oder Abstürze! Super Arbeit!!

closed 2024-10-29 by stefan123t