lumapu/ahoy #1043: [Bug] 0.7.17  MQTT topic 'available' geht direkt von 1 auf 4

Harald-U, 2023-07-27:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.7.17

### Github Hash

812adce

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

...

### Error description

In 0.7.17 wurden neue Werte für MQTT topic 'available' eingeführt. 
Bei mir geht der Status beim Start von '1 | available but not producing' direkt auf '4 | was available' obwohl die Web UI und alle MQTT Daten anzeigen, dass der Inverter produziert, availble als = 2 sein müsste.

knickohr, 2023-07-27:
Siehe 0.7.18 😉

Harald-U, 2023-07-27:
Das war es, mit 0.7.18 wieder gefixt. Danke Dir!!!

closed 2023-07-27 by Harald-U

knickohr, 2023-07-27:
Das ist brandneu, also noch nicht stabil, wird auch noch ein paar Tage dauern bis man das Feature nutzen kann.