lumapu/ahoy #1045: Frage: Kann man die AhoyDTU Version über versteckten API Endpoint oder versteckten Link auslesen

MetaChuh, 2023-07-27:
## Hardware
  * [ ] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [ ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [ ] external antenna (SMA)

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: _._.__
Github Hash: _______

### Build & Flash Method:
* [ ] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

MetaChuh, 2023-07-27:
Frage: Kann man die AhoyDTU Version über versteckten API Endpoint oder versteckten Link auslesen ?
Bis jetzt habe ich nichts gefunden, ausser die html seite zu scrapen.

thx und greetings,
metachuh

lumapu, 2023-07-27:
ja die API ist bis dato immer offen, da auch die WebUI darauf aufbaut. Prüfe doch mal `/api`

MetaChuh, 2023-07-27:
hi @lumapu 
danke für deine antwort.

hier ist ein dump aller avail_endpoints.
wo finde ich da die ahoydtu versionsnummer, z.b. 0.7.18 oder 0.6.9 ?

[avail_endpoints] => Array
	(
		[system] => http://192.168.0.200/api/system
		[statistics] => http://192.168.0.200/api/statistics
		[inverter/list] => http://192.168.0.200/api/inverter/list
		[index] => http://192.168.0.200/api/index
		[setup] => http://192.168.0.200/api/setup
		[live] => http://192.168.0.200/api/live
		[record/info] => http://192.168.0.200/api/record/info
		[record/alarm] => http://192.168.0.200/api/record/alarm
		[record/config] => http://192.168.0.200/api/record/config
		[record/live] => http://192.168.0.200/api/record/live
	)

[system] => Array
	(
		[ssid] => 3WebCube1E8B
		[device_name] => AHOY-DTU
		[dark_mode] => 1
		[mac] => 34:94:54:81:8D:F9
		[hostname] => AHOY-DTU
		[pwd_set] => 1
		[prot_mask] => 61
		[sdk] => 2.2.2-dev(38a443e)
		[cpu_freq] => 80
		[heap_free] => 11056
		[sketch_total] => 2592768
		[sketch_used] => 538
		[wifi_rssi] => -65
		[ts_uptime] => 185136
		[menu_prot] => 1
		[menu_mask] => 61
		[menu_protEn] => 1
		[esp_type] => ESP8266
		[radio] => Array
			(
				[power_level] => 2
				[isconnected] => 1
				[DataRate] => 2
				[isPVariant] => 1
			)

		[statistics] => Array
			(
				[rx_success] => 19692
				[rx_fail] => 66
				[rx_fail_answer] => 5875
				[frame_cnt] => 44012
				[tx_cnt] => 41110
				[retransmits] => 9486
			)

		[core_version] => 3.1.2
		[flash_size] => 4096
		[heap_frag] => 26
		[max_free_blk] => 7880
		[reboot_reason] => Software/System restart
		[schMax] => 8
	)

[statistics] => Array
	(
		[rx_success] => 19692
		[rx_fail] => 66
[... 998 lines cut ...]
							(
								[fld] => YieldTotal
								[unit] => kWh
								[val] => 21.39
							)

						[16] => Array
							(
								[fld] => P_DC
								[unit] => W
								[val] => 197.10
							)

						[17] => Array
							(
								[fld] => Efficiency
								[unit] => %
								[val] => 95.48
							)

					)

			)

	)

MetaChuh, 2023-07-28:
ps: bisher hole (scrape) ich mir in php die dtu versionsnummer über die html seite via regex so:

```php
<?php
    $html_response=file_get_contents("compress.zlib://http://$dtu_ip/live");
    preg_match("/git sha.*?:: (.*?)</msi", $html_response, $matches_array);
    $dtu_version=$matches_array[1];
    echo "DTU Version: $dtu_version";
?>
```

alternativ wenn die php zlib extension nicht verfügbar ist:

```php
<?php
    $query_timeout=1;
    $html_response=`curl --max-time $query_timeout --compressed "http://$dtu_ip/live"`;
    preg_match("/git sha.*?:: (.*?)</msi", $html_response, $matches_array);
    $dtu_version=$matches_array[1];
    echo "DTU Version: $dtu_version";
?>
```

wäre jedoch weniger hacky, wenns die versionsnummer in der api geben würde.
hab ich was übersehen, oder gibts die wirklich in keinem endpoint ?

greetings & thx,
metachuh

avendretter, 2023-07-28:
@lumapu @MetaChuh 
Ich habe das mal die Version und Build wieder hinzugefügt und einen PR erstellt: https://github.com/lumapu/ahoy/pull/1048
Sollten die beiden Daten aus Gründen nicht mehr in der API auftauchen, bitte ich dich @lumapu den PR einfach zu schließen ;) 

LG Avendretter

MetaChuh, 2023-07-28:
hi @avendretter,

genialer und einfacher turbo pr, danke 😃
... und sehr beruhigend ... dachte schon ich bin blind UND blöd, weil ich's nicht gefunden habe 🙈😂

@lumapu
keine priorität, weil ich ja dann den scraper für die bestehenden alten dtu versionen noch drin lassen muss, solange nicht alle auf (maybe future) latest upgedatet haben.

greetings und thx an alle,
MetaChuh

lumapu, 2023-07-28:
wenn's so einfach geht ist's in nächsten build drin. Habe es damals raus optimiert, da die Werte fest in der UI hinterlegt sind. 
Finde es richtig, die Werte wieder anzubieten.

MetaChuh, 2023-07-29:
@avendretter @lumapu 

hi,
0.7.20 eingespielt und getestet, läuft perfekt 👍
vielen dank euch beiden 

hier noch das php code sniplet, das die version mit scraper fallback ausliest, falls das noch jemand braucht.

```php
<?php
	
	$dtu_ip="10.10.10.68";	// ip der dtu
	// $dtu_ip="10.10.10.93";	// ip der dtu
	// $dtu_ip="192.168.33.1";	// ip der dtu
	
	$query_timeout=0.5;		// timeout für einleseversuch in sekunden (float)
	
	$dtu_endpoint="system";	// dtu endpoint zum einlesen
	$dtu_endpoint_url="http://$dtu_ip/api/$dtu_endpoint";
	
	$json_response=`curl --max-time $query_timeout "$dtu_endpoint_url"`;
	$dtu_data_array[$dtu_endpoint]=json_decode($json_response, true);
	
	// dtu version aus api lesen (ab v0.7.20, @avendretter)
	if (isset($dtu_data_array[$dtu_endpoint]["version"]))
	{
		$dtu_version=$dtu_data_array[$dtu_endpoint]["version"] . " (API)";
	}
	
	// dtu version via regex scraping (legacy workaround, dtu version ist in v0.6.9 bis v0.7.19 nicht via api verfügbar)
	else
	{
		$html_response=file_get_contents("compress.zlib://http://$dtu_ip/live");
		preg_match("/git sha.*?:: (.*?)</msi", $html_response, $matches_array);
		$dtu_version=$matches_array[1] . " (Scraper)";
	}
   
    echo "DTU Version: $dtu_version";
?>

```

greetings,
metachuh

closed 2023-07-29 by MetaChuh

KSNWH, 2024-01-10:
Hallo,
ich hänge mich mal hier rein:
Bisher habe ich keine Erfahrung mit der Ahoi DTU, werde aber demnächst eine für mein BKW einsetzen,
um die Werte in eine bestehende DB und Grafana-Umgebung einzulesen.
Dies möchte ich schon vorbereiten, bevor das BKW läuft.
Meine FRAGE:
a) gibt es ein curl-Kommando, mit dem ich alle relevanten Daten (in Summe) auslesen kann,
z,B. [diese Liste?](https://github.com/lumapu/ahoy/blob/main/User_Manual.md#topicinverter_name_from_setupch0)

b) kann mir jemand das Kommando UND das (ASCII-) Egebnis zur Verfügung stellen.
    PS ich will daraus die relevanten Parameter (Per String-Suche) extrahieren und direkt in die DB
         schreiben.

Herzlichen Dank im Vorraus!
Klaus

commit 53ff04b6 (2023-07-28): 0.7.20
* merge PR #1048 version and hash in API, fixes #1045
* fix: no yield day update if yield day reads `0` after inverter reboot (mostly on evening) #848
* try to fix Wifi override #1047
* added information after NTP sync to WebUI #1040