lumapu/ahoy #1046: [FeatureRequest: HMS Support + Parallele Nutzung von NRF24L01+ und CMT2300A]=> freie Definition von SDIO und SCK zum CMT230A beim ESP 38pin

schluss123, 2023-07-28:
[FeatureRequest: HMS Support + Parallele Nutzung von NRF24L01+ und CMT2300A]=> freie Definition von SDIO und SCK zum CMT230A beim ESP 38pin

Wäre es möglich, alle PinOUT Konfigurationen frei definierbar zu machen? Auf meiner PCB sind die festen Pins, ESP zu CMT:
D23 - SDIO und D18 - SCK als direkte Leiterbahn zum NRF24L01+ ausgestaltet, OpenDTU läuft damit, Ahoy leider nicht mehr. Ich kann die GPIOs 18 und 23 NICHT zum CMT2300A routen.
Ähnliches Problem bei mir - SCK liegt auf dem HSPI GPIO 12 und SDIO liegt ebenfalls auf dem HSPI GPIO 14. Dito 38 Pin ESP32.
## Hardware
  * [ ] ESP8266
  * [X ] ESP32
  * [X ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [ ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

###CMT2300A
* [X ] external antenna (SMA)

### Antenna:
* [X ] circuit board
* [X ] external antenna (SMA)

### Power Stabilization:
* [X ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: _._.__
Github Hash: _______

### Build & Flash Method:
* [ ] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

schluss123, 2023-07-28:
Auf Open DTU läuft es schon problemlos...aber Open DTU unterstützt nicht die älteren TSol-WRs....

Riker18, 2023-09-12:
Gerade gesehen das es hier schon eine Bitte gibt - unterschreib ich so 💯

closed 2023-12-29 by lumapu