lumapu/ahoy #1047: [Bug] assert failed:  tcpip_api_call IDF/components/lwip/lwip/src/api/tcpip.c:497 (Invalid mbox)

avendretter, 2023-07-28:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.6.9

### Github Hash

15ec6a0

### Build & Flash Method

VSCode - Platform IO (build & flash)

### Setup

- used my own `config_override.h`
- not working with the `config_override_example.h` either

- compiled and uploaded to the ESP32 
- ESP keeps crashing with `assert failed: tcpip_api_call IDF/components/lwip/lwip/src/api/tcpip.c:497 (Invalid mbox)`

### Debug Serial Log output

```shell
Executing task: platformio run --target upload --target monitor --environment esp32-wroom32-release --upload-port /dev/cu.usbserial-0001 --monitor-port /dev/cu.usbserial-0001 

Processing esp32-wroom32-release (platform: espressif32@>=6.1.0; board: lolin_d32; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Firmware Revision: g15ec6a0
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/lolin_d32.html
PLATFORM: Espressif 32 (6.3.2) > WEMOS LOLIN D32
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20009.0 (2.0.9) 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - tool-mkfatfs @ 2.0.1 
 - tool-mklittlefs @ 1.203.210628 (2.3) 
 - tool-mkspiffs @ 2.230.0 (2.30) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 44 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ESP Async WebServer @ 1.2.3+sha.1dde9cf
|-- RF24 @ 1.4.7
|-- Time @ 1.6.1
|-- espMqttClient @ 1.4.2+sha.06c150c
|-- ArduinoJson @ 6.21.3
|-- Timezone @ 1.2.4+sha.7cf1425
|-- U8g2 @ 2.35.4
|-- GxEPD2 @ 1.5.2
|-- WiFi @ 2.0.0
|-- Adafruit GFX Library @ 1.11.7
|-- SPI @ 2.0.0
|-- DNSServer @ 2.0.0
|-- LittleFS @ 2.0.0
|-- AsyncTCP @ 1.1.1
|-- Update @ 2.0.0
|-- FS @ 2.0.0
|-- Adafruit BusIO @ 1.14.1
|-- WiFiClientSecure @ 2.0.0
|-- Wire @ 2.0.0
Building in release mode
Compiling .pio/build/esp32-wroom32-release/src/app.cpp.o
Compiling .pio/build/esp32-wroom32-release/src/main.cpp.o
Compiling .pio/build/esp32-wroom32-release/src/wifi/ahoywifi.cpp.o
Linking .pio/build/esp32-wroom32-release/firmware.elf
Retrieving maximum program size .pio/build/esp32-wroom32-release/firmware.elf
Checking size .pio/build/esp32-wroom32-release/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  18.7% (used 61428 bytes from 327680 bytes)
Flash: [========= ]  85.5% (used 1120649 bytes from 1310720 bytes)
Building .pio/build/esp32-wroom32-release/firmware.bin
esptool.py v4.5.1
Creating esp32 image...
Merged 25 ELF sections
Successfully created esp32 image.
Configuring upload protocol...
AVAILABLE: cmsis-dap, esp-bridge, esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Using manually specified: /dev/cu.usbserial-0001
Uploading .pio/build/esp32-wroom32-release/firmware.bin
esptool.py v4.5.1
Serial port /dev/cu.usbserial-0001
Connecting....
Chip is ESP32-D0WD-V3 (revision v3.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: a8:42:e3:ce:f4:c8
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x00122fff...
Compressed 17488 bytes to 12168...
Writing at 0x00001000... (100 %)
Wrote 17488 bytes (12168 compressed) at 0x00001000 in 0.3 seconds (effective 522.6 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 146...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (146 compressed) at 0x00008000 in 0.1 seconds (effective 432.7 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 1051.3 kbit/s)...
Hash of data verified.
Compressed 1126400 bytes to 720798...
Writing at 0x00010000... (2 %)
Writing at 0x00015be8... (4 %)
....
Writing at 0x00117f91... (97 %)
Writing at 0x0011d340... (100 %)
Wrote 1126400 bytes (720798 compressed) at 0x00010000 in 9.8 seconds (effective 923.8 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

Please build project in debug configuration to get more details about an exception.
See https://docs.platformio.org/page/projectconf/build_configurations.html

--- Terminal on /dev/cu.usbserial-0001 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
I: Initializing FS ..
I:  .. done
[    36][E][vfs_api.cpp:105] open(): /littlefs/settings.json does not exist, no permits for creation
W: failed to load json, using default config
I: Settings valid: false
I: RF24 Amp Pwr: RF24_PA_I: LOW
W: WARNING! your NRF24 module can't be reached, check the wiring
I: (#0) resetPayload
I: (#1) resetPayload
I: (#2) resetPayload
I: (#3) resetPayload
I: (#4) resetPayload
I: (#5) resetPayload
I: (#6) resetPayload
I: (#7) resetPayload
I: (#8) resetPayload
I: (#9) resetPayload
I: (#0) resetPayload
I: (#1) resetPayload
I: (#2) resetPayload
I: (#3) resetPayload
I: (#4) resetPayload
I: (#5) resetPayload
I: (#6) resetPayload
I: (#7) resetPayload
I: (#8) resetPayload
I: (#9) resetPayload
W: WARNING! your NRF24 module can't be reached, check the wiring

assert failed: tcpip_api_call IDF/components/lwip/lwip/src/api/tcpip.c:497 (Invalid mbox)

Backtrace: 0x400837b1:0x3ffb1f90 0x4008be49:0x3ffb1fb0 0x40091ba5:0x3ffb1fd0 0x40118465:0x3ffb2100 0x400ec367:0x3ffb2130 0x400f6002:0x3ffb2180 0x400d7649:0x3ffb21a0 0x400e815e:0x3ffb2200 0x400e8cdf:0x3ffb2270 0x40100dfe:0x3ffb2290

  #0  0x400837b1:0x3ffb1f90 in panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:408
  #1  0x4008be49:0x3ffb1fb0 in esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:137
  #2  0x40091ba5:0x3ffb1fd0 in __assert_func at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/assert.c:85
  #3  0x40118465:0x3ffb2100 in tcpip_api_call at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/api/tcpip.c:497 (discriminator 1)
  #4  0x400ec367:0x3ffb2130 in AsyncServer::begin() at .pio/libdeps/esp32-wroom32-release/AsyncTCP/src/AsyncTCP.cpp:511
      (inlined by) AsyncServer::begin() at .pio/libdeps/esp32-wroom32-release/AsyncTCP/src/AsyncTCP.cpp:1264
  #5  0x400f6002:0x3ffb2180 in AsyncWebServer::begin() at .pio/libdeps/esp32-wroom32-release/ESP Async WebServer/src/WebServer.cpp:89
  #6  0x400d7649:0x3ffb21a0 in Web<HmSystem<(unsigned char)10, Inverter<float> > >::setup(IApp*, HmSystem<(unsigned char)10, Inverter<float> >*, settings_t*) at web/web.h:94
  #7  0x400e815e:0x3ffb2200 in app::setup() at app.cpp:76
  #8  0x400e8cdf:0x3ffb2270 in setup() at main.cpp:21
  #9  0x40100dfe:0x3ffb2290 in loopTask(void*) at /Users/jugo/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:42
```

### Error description

I expected to ESP to boot normally, but it keeps crashing.

avendretter, 2023-07-28:
Hey, 

Removing/ commenting out the following lines from the `config_override.h` "resolved" the issue after re-compiling. The ESP is now booting "normal"

```
// override fallback WiFi info
#define FB_WIFI_OVERRIDDEN
``` 

------------------------------------------------------------------------------------------------------------------------

Hey, 

Sobald man folgende Zeilen in der `config_override.h` auskommentiert/löscht und neu kompiliert, startet der ESP wieder "normal".
```
// override fallback WiFi info
#define FB_WIFI_OVERRIDDEN
```

stefan123t, 2023-07-29:
@avendretter schau Dir mal die Änderungen von Lukas im 0.7.20 branch an:
Ggf. mit `git checkout development03` auf den dev-branch wechseln.

https://github.com/lumapu/ahoy/blob/53ff04b6894e0005ea17043889361a530ba75645/src/wifi/ahoywifi.cpp#L48-L68

avendretter, 2023-07-29:
jupp in development03 gehts jetzt! Top!

closed 2023-07-29 by avendretter