lumapu/ahoy #1050: Feature Request :  Signslstärke good/bad von NRF ausgeben

knickohr, 2023-07-29:
Offensichtlich existiert in der RF24.h Library ein Funktionsaufruf mit der man die Signalstärke prüfen kann. Sie gibt ein gut/schlecht zurück wenn die Signalsärke >&%dBm bzw. <&4dBm ist.

bool goodSignal = radio.testRPD();
if(radio.available()){
Serial.println(goodSignal ? "Strong signal > 64dBm : "Weak signal < 64dBm");
radio.read(0,0);

Offenbar schon in OpenDTU integriert, wäre es schön wenn man das auch für Ahoy machen könnte. Bitte auch mit MQTT ausgeben 😉

DanielR92, 2023-08-23:
Hallo zusammen,

das wollt ich schon immer drinnen haben. 
Ich schau mir das mal an, da meine Testumgebung aktuell etwas "umständlich" wurde.

Meld mich sobald ich was neues habe.

knickohr, 2023-08-24:
Von meinen 13 WRs wird wohl einer good ausgeben 😉

knickohr, 2023-08-28:
Siehe hierzu Issue #1129

knickohr, 2023-08-30:
Die Funtion hat leider wenig ynägrwert da die meiste Zeit ein <-64dBm kommt, egal wie nah man am WR dran ist. Schade, wäre eine einfache Möglichkeit gewesen eine wenigsten primitive Empfangsqualität darzustellen.

Vielleicht sollten wir doch auf die heuristische Funktion von „Oberfritze“ setzen ?

Ich laß hier aber mal trotzdem auf, so als Reminder 😉

DanielR92, 2023-09-06:
Hmm, das muss man nochmals anschauen. Danke für die Info schonmal @knickohr

knickohr, 2023-09-06:
@DanielR92 

Schau Dir den PR #1080 mal an. Da wird auch eine Heuristik generiert welche die Qualität der empfangenen Daten nachbildet. Ist zwar alles noch in der Testphase, wird aber meiner Meinung nicht mehr lange her dann fließt das in die nächsten Ahoy-Versionen ein.

Sieht dann so aus :  Quality:  0 -1 -2 4 2, Cnt 3, Fail 0

Was die Werte genau bedeutet muß Dir @oberfritze erklären. Ich weiß nur, je höher der Wert, umso besser. Außerdem müßte man die 5 Werte zu einem zusammen fassen.

knickohr, 2023-09-20:
Könnte es sein das da noch ein Bug drin war ?  Mit der DEV 58 habe ich jetzt mehrere Inverter die auch ein good bringen 😎

lumapu, 2023-09-20:
nein, seit dem changelog mache ich nichts mehr geheim, alles transparent 😊

knickohr, 2023-09-20:
Spannend ! 🤔

knickohr, 2023-09-21:
Bei der 59 ist es auch OK. Da hin und wieder das auch wechselt, gehe ich davon aus das es korrekt funktioniert. Auch wenn es nicht den erhofften Effekt gebracht hat und nur bei sehr guten und kurzen Empfangsbedingungen funktioniert.

knickohr, 2023-09-29:
Nachtrag :  ich glaube herausgefunden zu haben warum es bei einigen NRF geht und bei anderen wiederum nicht :

- Fast alle Billig NRF sind China Clone und da funktioniert es nich
- Auch das EByte Modul macht es nicht
- Einzig das Fusion liefert bei mir diese Werte

Ich mache deshalb hier zu. Es scheint ein reines Chipproblem zu sein, und funktioniert nur mit echten NRF-Chips.

closed 2023-09-29 by knickohr

DanielR92, 2023-09-29:
Danke für die Infos, das sollte man aber im README doch festhalten?

knickohr, 2023-09-29:
Ja, EByte müßte nochmal verifiziert werden. Bin mir aber ziemlich sicher das da auch nichts kam.