lumapu/ahoy #1051: MQTT Home Assistant, keine Anzeige des 2. WR

schlagucke, 2023-07-30:
## Hardware
  * [ X] ESP8266
  * [ ] ESP32
  * [X ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [ ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [X] external antenna (SMA)

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: 0.6.9
Github Hash: _______

### Build & Flash Method:
* [x] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

Hallo,

MQTT funktioniert und die Daten werden auch in Home Assistant angezeigt.

Nur werden die Daten vom ersten Wechselrichter in HA angezeigt und die Daten des zweiten Wechselrichter nicht.

Könnte mir einer eventuell weiterhelfen. 

Habe schon alles zurück gesetzt. Aktualisiert etc. 

Es werden weiterhin nur die Daten vom ersten Wechselrichter und die gesamten Totalwerte übertragen. 

Vielen Dank. 

Gruß

knickohr, 2023-07-30:
Da ich bereits 13 Inverter an der DTU habe und die DTU alle Werte an den Broker liefert, würde ich das Problem eher bei HA suchen.

schlagucke, 2023-07-30:
Alles klar. Das schaue ich da noch einmal. 

Ist der Fehler schon einmal ein wenig eingegrenzt.

knickohr, 2023-07-30:
Hast Du automatisch die Entitäten oder wie sich das nennt angelegt, oder manuell ?

schlagucke, 2023-07-30:
Die Entitäten haben sich automatisch angelegt.

Sobald ich MQTT in HA aktiviert und alles soweit eingerichtet hatte.

knickohr, 2023-07-30:
Dann mach’s manuell 😉

lumapu, 2023-07-30:
das wurde meines Wissens erst in der development Version behoben, bitte diese testen, und Feedback geben.

https://nightly.link/lumapu/ahoy/workflows/compile_development/development03/ahoydtu_dev.zip

schlagucke, 2023-07-30:
Super, vielen Dank für die Info.