lumapu/ahoy #1052: Power limit in Watt

DejanBukovec, 2023-07-30:
I have question about AhoyDTU. Im switch from OpenDTU to AhoyDTU and missing some sensors and input_numbers in HomeAssisntant...

I use MQTT to get data into HA and set limit of 3 inverters to get "zero grid feed".

In HA and Ahoy I can see current Power limit in percent but not in Watts. Is possible somehow get this value in MQTT(Limit in Watts)? 

Is possible show limit in watt instead of % in WEB GUI(Or both % and watt)?

[image]

lumapu, 2023-07-30:
we have several issues with that request, we need to cross link them.
It is planned but was not done until now.

DejanBukovec, 2023-07-31:
Maybe for start it will be ok if there will be shown only calculated value, depend of some value entered into settings, if you can't get model from serial or max absolute value from inverter...

closed 2024-01-03 by lumapu