lumapu/ahoy #1053: zugriff auf ahoy dtu   mal ja, mal nein

golamic, 2023-07-30:
hallo,

ich habe eine Ahoydtu aufgebaut mit ESP8266 mit Antenne und Display, Vers. 6.9... und einen HM1500 mit 4 Modulen.
Alles funktioniert bestens, über das Diplay kann ich ohne weitere Himfsmittel direkt die aktuelle Leistung, den Tages- und den Gesamtertrag ablesen. Auch über Mqtt mit entsprechender Software kann ich mir von überall alle relevanten Daten anzeigen lassen.
NUR... von Zeit zu Zeit ... bekomme ich keine Anzeige wenn ich direkt auf die DTU   via Fritzbox zugreifen möchte. Da steht dann...
"every seconds the values are updated" und es ist nicht möglich sich z.B einzuloggen. Aber hin und wieder, ohne was zu ändern! bekomme ich eine ordentliche Anzeige der Daten und habe auch vollen Zugriff...
Meinem Nachbar geht es genauso... mal ja mal nein, nur hat der keine LCD Anzeige.... daran kann es also nicht liegen.
Was ist da los

dtuuser, 2023-07-30:
Bei meinem ESP8266 auch so. Habe mir eine zweite Ahoy zum Testen mit einem ESP32 aufgebaut. Bis jetzt keine Abfrageprobleme über Web.

GHolli, 2023-07-31:
Ich habe das Verhalten auch schon einmal beobachtet, wenn die AhoyDTU keine Verbindung mehr hat. Die restlichen angezeigten Daten kommen dann offenbar vom Browser-Cache.
Kann es sein, dass die WLAN-Verbindung zu schwach und damit instabil ist?

golamic, 2023-07-31:
nein, die DTU ist nich offline, der LCD zeigt ja alles richtig an und auch via Mqtt sind alle 30sek.  die Daten, wie bei mir eingestellt, zu sehen.Es gibt nur keinen direkten Zugriff  auf die DTU. Irgendwann dann wieder ja,  aber meistens halt nicht.Wer jetzt keinen LCD oder Mqtt eingebunden hat, sieht meistens NICHTS? das ist das Problem!--Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

GHolli, 2023-07-31:
Das Display wird auch ohne WLAN-Verbindung funktionieren (wenn die Zeit einmal synchronisiert wurde), solange eine Funkverbindung mit dem Wechselrichter besteht. MQTT hingegen nicht - werden die Werte via MQTT tatsächlich aktualisiert, oder sind es immer die gleichen Werte (während auch der direkte Zugriff auf die DTU nicht möglich ist)?

golamic, 2023-07-31:
via Mqtt wird definitiv alle 30 Sek. aktualisiert, zugriff auf DTU  mal nein mal ja....  nur checke ich dann nicht zeitgleich den Zugriff auf die DTU sondern entweder schaue ich via Mqtt oder direkt... nur beim umschalten auf Mqtt geht alles immer.....??? beim direkten Zugriff halt nicht immer.....--Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

golamic, 2023-07-31:
egal, was d auch immer nicht richtig funzt, das mit der Ahoydtu ist alles ziemlich instabil.... ein nettes Gadget mehr aber auch nicht! Preiswert ok, keine Frage, aber halt doch irgendwie eine Freak-bastellösung.... werde wohl dann doch auf HM- original umsteigen, Cloud hin oder her, das scheint dann wenigstens stabil zu laufen, das ist es was ich am Ende möchte, irgendwann einfach nicht mehr basteln müssen, ist mir meine Zeit zuschade, sorry, das wars....Adieu Ahoy.....

kiu77, 2023-07-31:
@golamic 
Du hast die Frage von @GHolli nicht beantwortet. Absichtlich?

RufusRed80, 2023-08-01:
Die Probleme kann ich mit meinem Wemos D1 mini bestätigen. Abfrage per MQTT problemlos, das Webinterface hingegen ist extrem langsam, der Login nicht möglich und die Meldung "Every n/a seconds the values are updated". 

Ahoy-Version 0.6.9
WR  Hoymiles 1500
4 Module angeschlossen. 
Netzteil 2A
Display 1,3"
Funkmodul NRF24L01+
aktuell ohne Kondensator (ist noch im Versand)

kiu77, 2023-08-01:
@RufusRed80 
Ist der D1 mini mit ESE8266 oder ESP32?
An einem 2A-Netzteil kann man locker 20 DTUs betreiben ;-)

RufusRed80, 2023-08-01:
@kiu77 Es ist ein ESP8266-12F. Beim Netzteil wollte ich auf der sicheren Seite sein ;-)

GHolli, 2023-08-02:
Da das Kondensator-Thema immer wiederkehrt und offenbar entscheidend für die Stabilität ist, möchte ich auch ein paar Kommentare dazu abgeben.

Die meisten Boards sollten ja die "Standard-Stabilisierung" haben, wie von ningyichensha angesprochen.
@ningyichensha, welches Board verwendest du? Weil ich würde einmal vermuten dass dein Board auch vor- und nach dem Spannungsregler Elkos oder Tantal- sowie keramische Kondensatoren hat.

Ich denke, die Problematik hängt auch sehr stark vom Setup up. Da gibt es einige Parameter:

- Originaler nRF24L01+ Chip (250&nbsp;nm Technologie) oder [Klon](https://hackaday.com/2015/02/23/nordic-nrf24l01-real-vs-fake/) (vielleicht 350&nbsp;nm Technologie)
- Board mit- oder ohne Spannungsregler
    - Mit Spannungsregler: LDO (Linearregler) oder Schaltregler
- Qualität des Netzteils (z. B. Lastregelung, Ripple)
- Verkabelung zum Funkchip (z. B. Länge des Kabels)
- Auf den Modulen/Boards bereits vorhandene Stabilisierungskondensatoren

Im Falle von Instabilitäten einfach den Elko zu vergrößern wäre dabei nicht meine bevorzugte Lösung, denn normalerweise sollte ein kleiner Abblockkondensator die Spitzen abfangen, bis der Spannungsregler nachregelt.
Nun stellt sich für mich die Frage: Wenn Probleme auftreten, regelt der Spannungsregler (oder gar das Netzteil) nicht schnell genug (dynamische Lastregelung), oder kann er/es gar nicht genug Leistung liefern?
Beim Elko muss man auch aufpassen: ein größerer speichert zwar mehr Energie, kann diese aber vielleicht nicht schnell genug abgeben. Daher ist vielleicht ein "[Low ESR](https://de.wikipedia.org/wiki/Elektrolytkondensator#Scheinwiderstand_Z_und_Ersatzserienwiderstand_ESR)" Elko einem größeren Standard-Elko vorzuziehen.
Sollten tatsächlich längere, hohe Stromspitzen auftreten, könnte dies ein großer Elko abfangen. Aber bei Netzteilen, die 500&nbsp;mA oder mehr liefern können, kann ich mir dies fast nicht vorstellen... Aber ich konnte auch schon Stabilitätsunterschiede bei der Verwendung unterschiedlicher USB-Netzteile bei ansonsten gleichen Setup beobachten.

RufusRed80, 2023-08-02:
Heute kamen meine Kondensatoren (6,3V, 100uF). Einen an die 5V / GND vom Wemos Mini und einen fürs Funkmodul. 

Geändert hat sich an der Geschwindigkeit leider nichts, teilweise ist der Seitenaufbau nichtmals möglich. @ningyichensha Welche Version hast Du installiert?

GHolli, 2023-08-02:
@RufusRed80, wie gut ist deine WiFi-Verbindung? Ahoy zeigt wifi_rssi an (habe keine Erfahrung mit dieser Anzeige) und mein AccessPoint zeigt auch das Signal-/Rauschverhältnis und die RX Rate und TX Rate der verbundenen Geräte an. Denn ich gehe einmal stark davon aus, dass der Datentransfer via Webinterface mehr Daten benötigt als ein MQTT Telegramm.
Und wie sieht's mit der Uptime aus? Weil wenn der ESP nicht von selbst neu startet, sollte er eigentlich stabil laufen.

RufusRed80, 2023-08-02:
AhoyDTU:
[image]
FritzBox:
[image]

ningyichensha, 2023-08-02:
Auf Nachfrage noch die Konfiguration meiner Installation:
Wemos D1mini
Funkmodul NRF24L01+
SW Version 4.15
Raspberry Steckernetzteil

Dis Entfernung vom Board zum Wechselrichter ist ca. 5m vom Innenraum zum Balkon.
Der Router (Fritzbox) ist im Erdgeschoss, das Board im 1. OG.

Und hier noch 2 Bilder meines boards mit den 2 Kondensatoren.
[image]
[image]

dtuuser, 2023-08-02:
@ningyichensha 
Ich lade mit meinem VB Programm unter Windows die Seite  http://192.168.178.23/api/inverter/id/0 
und splitte mir dann den Inhalt nach Bedarf in Variablen und schiebe mir diese in meine Registry.
Darauf greifen dann diverse andere Programme zu.  
U.a. meine Nulleinspeisung. Den aktuellen Verbrauch lese ich so auch mit Tasmota vom Stromzähler aus und setze das WR Limit mit einem anderen Programm mit dem Shell Aufruf mit dem folgendem Curl Befehl.

curl -i -X POST -H "Content-Type: application/json" -d "{"id"":"0","cmd":"limit_nonpersistent_absolute","val":"144""}" http://192.168.178.23/api/ctrl

Da ich kein MQTT nutze, habe ich anscheinend auch keine Resets.

dd5xl, 2023-08-04:
> Und hier noch 2 Bilder meines boards mit den 2 Kondensatoren. [image]

Schön kompakter Aufbau, aber: Wenn ESP8266 WLAN-Antenne so dicht an der nRF24 Antenne plaziert ist, wundert mich gar nichts: der nRF24 wird beim Senden den RX des WLAN-Moduls zustopfen und umgekehrt. Ist ja beides 2,4GHz...

Ich habe mittlerweile 5 DTUs aufgebaut und immer darauf geachtet, die beiden Antennen maximal voneinander zu entkoppeln, 10cm sind kein Luxus.

RufusRed80, 2023-08-04:
Genau das selbe wie @dd5xl habe ich gestern Abend auch festgestellt. Nachdem meine DTU gestern fast gar nicht erreichbar war habe ich nochmal einen Dauerping abgesetzt und dann alles aus dem Gehäuse genommen. Je weiter Antenne und ESP von einander entfernt waren, desto besser waren die Werte. Aktuell liegt alles offen aufm Schreibtisch, dank Kabel sind DTU und Antenne gut 15 cm entfernt. 
PS: Ich setze das Funkmodul mit ext. Antenne ein.

ningyichensha, 2023-08-04:
@dd5xl @RufusRed80 
Hallo Zusammen.

in der Theorie habe ich das auch befürchtet, weil die Antennen sehr nah beieinander sind
In der Praxis funktioniert es aber.

Hier mal ein Bild der Statistik. D.h. 2/3 der Anfragen sind erfolgreich. Das war ohne die 2 Kondensatoren wesentlich schlechter.
[image]

kiu77, 2023-08-04:
@dd5xl 
Ja, so eine Überlegung liegt nahe.
Allerdings gibt es ganz viele ähnliche Aufbauten, die klaglos funktionieren.
Um das genau sagen zu können, ob z.B. das Problem gar nicht besteht,
weil die beiden Kommunikationen DTTU->Router und DTU->Inverter eben
zeitlich entkoppelt sind, das kann man nur beantworten, wenn man den Code
versteht (tu ich nicht wirklich). Oder wenn @lumapu diese Frage beantwortet ;-)

Außerdem hat @ningyichensha die Antennen von ESP und NRF24 90° versetzt,
sodass sie nicht die gleiche Polarisationsebene haben. Ich bin mir auch nicht sicher,
ob 10cm wirklich etwas helfen würden, wenn Zustopfen das Problem wäre.
Problematischer finde ich, dass die Antenne des ESP direkt neben dem Quarz
bzw. der Takterzeugung des NRF24 liegt, was den WLAN-Empfang dauerhaft
beeinträchtigen könnte. Aber wenn nicht das das Problem ist, sondern die
Kommunikation mit dem Inverter, dann wäre ein NRF24 mit externer Antenne
vielleicht angebracht.

ningyichensha, 2023-08-04:
@lumapu @dd5xl 
Hallo,
noch als Klarstellung, falls es falsch rüberkam.
Diese Installation mit den Antennen funktioniert wunderbar,
Ich hatte nur mit dem Hinweis auf die 2 Kondensatoren eine Hilfestellung gegeben.

Aber noch eine andere Frage:
Ich möchte das System dazu bringen die aktuelle Leistung ohne Aufruf der web-Seite und ohne MQTT an ein anderes System im Hause regelmäßig zu übertragen.

das hier funktioniert **nur**, wenn ich am PC die Seite 192..../hoymiles öffne, soll aber nicht **immer** funktionieren

Ausschnitt aus app.ccp, in void app::showLiveData

 for(uint8_t fld = 0; fld < 10; fld++) {
                pos = (iv->getPosByChFld(CH0, list[fld]));
                if(0xff != pos) {
                    modHtml += F("<div class=\"subgrp\">");
                    modHtml += F("<span class=\"value\">") + String(iv->getValue(pos));
                    modHtml += F("<span class=\"unit\">") + String(iv->getUnit(pos)) + F("</span></span>");
                    modHtml += F("<span class=\"info\">") + String(iv->getFieldName(pos)) + F("</span>");
                    modHtml += F("</div>");
                    if (String(iv->getFieldName(pos)) == "P_DC") {
                    
                    DPRINTLN(String(iv->getFieldName(pos)) + " " + String(iv->getValue(pos)));
                    WiFiClient client;
                  if (!client.connect(host_licht_roll_kueche_sued, 5585)) {
                  Serial.println("connection failed");
                  }
      
                // This will send a string to the server
                Serial.println("sending data to server");
                if (client.connected()) {
                client.println((String(iv->getFieldName(pos)) + " " + String(iv->getValue(pos))));

closed 2024-01-03 by lumapu