lumapu/ahoy #1054: Requests: Display Anzeige Strom current

blabond, 2023-07-30:
Wäre es möglich auf dem Display, wo aktuell ja der produzierte Strom & täglich produzierte Strom und gesamt produzierte Strom steht eine Zeile weg zu nehmen und dafür einen Wert vom iobroker anzeigen zu lassen ?

Im speziellen habe ich hier an den Wert von meinem ir-lesekopf gedacht, das ich dann auf dem Display den tatsächlichen Strom Bezug bzw Einspeisung sehen kannyaktuell steht. 

Lg

knickohr, 2023-07-30:
Unter Strom meinst Du wahrscheinlich die Leistung oder die Energie ?

Wie willst you diesen Wert in Ahoy rein bekommen ?  MQTT ?

blabond, 2023-07-30:
Ja genau, den aktuell Leistungswert von Stromzähler ob gerade Strom bezogen wird bzw eingespeist.

Das rein bekommen hätte ich 2 Ideen.

Entweder über eine API im iobroker die vom ahoy abgerufen wird oder wie du schon sagtest mit mqtt senden.

knickohr, 2023-07-30:
Über MQTT würde es rein theoretisch sogar gehen, da wäre alles schon da.

Müßte nur jemand programmieren und einen PullRequest einstellen.

Das lohnt sich aber nur für das ePaper Didplay, die anderen wären zu „klein“, Alls zu wenig Pixel um da noch eine Info rein zu quetschen. Ich glaube nicht das die Mehrhrit bereit ist auf irgendeinen Anzeigewert zu verzichten. Beim ePaper könnte man noch quetschen.

blabond, 2023-07-30:
Man kann es ja so gestalten dass man wie bisher 3 Zeilen hat und man nun selbst entscheiden kann welche der möglichen Optionen in welcher Zeile steht. So steht es jedem selbst frei.

lumapu, 2023-07-30:
irgendwann blickt keiner mehr bei den Einstellungen durch, ist ja jetzt schon so mit den ganzen pinout Definitionen 😇

knickohr, 2023-07-31:
Das ist eigentlich das schlimmste, vor allem für Anfänger, an dem ganzen. Aber willst Du es so wie bei OpenDTU mit dieser Pinmapping machen ? 😲

Vielleicht sollte man solche Sachen mit funktionierenden Defaultwerten belegen und nur in einem Expertenmenue die eigentliche Zuorndung frei änderbar machen.

Ollipop030, 2023-07-31:
@lumapu: Vielleicht könnte man die Einstellungen nebeneinander als Reiter positionieren? Dann hätte jeder Menüpunkt eine eigene Seite. Und man könnte evtl ein paar Punkte kominieren, zB Netzwerk, NTP und MQTT auf einer Einstellungseite.

knickohr, 2023-07-31:
Ich denke mir das mehr so :

Du hast auf der Seite die Auswahl, bspw. NRF und CMT, sonst nix. Wenn Du die Pinbelegung von diesen ändern willst, mußt Du auf eine Expertenseite wechseln wo dann die Pins, ja, zB. in Reitern, für jedes Device frei änderbar sind. Dann sollte man aber genau wissen was man tut.

So könnte man es auch mit den bereits 8 ??? Displays machen. Im Setup kann man nur den Typ auswählen, fertig. Dann werden hinterlegte Pins defaultmäßig zugeordnet die zu 90% auch funktionieren. Wer das ändern möchte/muß oder was auch immer, der geht auf die Experteneinstellung und kann das frei ändern. Hier kann man dann auch so Sachen wie welche Zeile im Display wo angezeigt wird und mit welchem Inhalt.

Macht zwar das Webinterface größer, finde ich aber immer noch besser als mit dieser kryptischen Pinmapping. Natürlich sollte der „Experte“ dann sein Hirn einschalten und genau wissen was er tut 😇

lumapu, 2023-07-31:
finde das Brainstorming hier sehr gut, auch die bisherigen Inputs.
Ich finde das JSON von OpenDTU nicht unbedingt erstrebenswert, der Punkt mit Defaultwerten und Experteneinstellungen gefällt mir sehr gut.

kiu77, 2023-07-31:
Finde ich auch. Das klingt plausibel.
Ich habe mich gerade durch OpenDTU durchgebissen. Es ist echt viel von allem und wird letztlich unübersichtlich.
Gerade für nicht so Mikrocontroller-affine Leute kann das echt schwierig werden, denn die Cracks können sich
manchmal offensichtlich nicht recht vorstellen, was andere nicht verstehen ;-)
Der Ansatz von Ahoy ist da ergonomischer.

knickohr, 2023-08-01:
Das war auch der Grund warum ich mich für Ahoy entschieden habe. Das OpenDTU war mit zu Nerd-mäßig. Auch wenn ich einiges an Programmierung, auch für Mikrocontroller, verstehe. Habe selbst Jahrzehnte lang programmiert, aber halt eben nich in C 😞 Das schlimmste an der Programmierung ist das Userinterface. Man glaubt gar nicht auf welche doofe Gedanken Benutzer kommen, der meiste Code geht bei der Prüfung von Dau-Fehlern und unsinnigen Eingaben drauf. Das muß alles wasserfest sein.

OK, mein Vorschlag wäre das jetzige Standard WebGUI wieder etwas zu verschlanken und dafür eine eigene Seite mit Experteneinstellungen zu machen.

- Reiter Ahoy-DTU kann so bleiben
- Die Live-Ansicht sowieso
- Serial Control würde ich schon ins Expertenmenue nehmen, bis auf das Commands absetzen. Dann würde ich aber das auch umbenennen.
- Settings, da geht’s richtig los 😲 Hostname bleibt, bei den Radios würde ich nur noch die Checkboxen NRF und CMT übrig lassen, der Rest wieder ins Expertenmenue.
- Network und Protection bleibt so
- Inverter bis auf den unteren Teil mit General
- NTP, die Sonneneinstellungen und MQTT bleiben
- Display Config nur der obere Teil, Pinout wieder für Experten 😉
- Update, §System, Rest API, Documentation und About sind OK
- Im einem weiteren Reiter dann das Expertenmenue. Dieses wird dann wieder unterteilt so wie in den Settings. Dan kann sich der Experte dann austoben. Standardmäßig werden die dortigen Einstellungen dann mit funktionierenden Defaultwerten abhängig von den vorherigen Einstellungen belegt die auch zum Großteil getestet sind und funktionieren. Selbstverständlich muß eine Plausibilitätsprüfung stattfinden (oh Gott, dieser Programmieraufwand 🫣). Vielleicht noch sinnvoll vorher eine Warnung auszugeben „you know what you are doing !“ 😅

blabond, 2023-08-01:
Oha, richtig was losgetreten hier mit der kleinen Anfrage, oder ist selbst nun untergegangen ^^

knickohr, 2023-08-01:
Nein nein, das war schon länger Wunsch, vor allem im Discord heftig diskutiert :  Welche Werte werden angezeigt ?  😅

MetaChuh, 2023-08-01:
@blabond 
willkommen in der eigendynamik von:
für brainstorming gehijackte issues 😂

ist auch ein gutes zeichen des enthusiasmus aller devs und user, auch die issues anderer zu lesen und mit weiterführenden ideen zu ergänzen.

werbung: selbst für mich und andere devs, ist ahoydtu aus vielen gründen, inkl. den in den bisherigen antworten erwähnte, die erste wahl 👍

thx, greetings an alle, und big kudos an @lumapu,
metachuh

stefan123t, 2024-10-27:
@blabond @lumapu @knickohr eigentlich ging es doch um die Abfrage "eines" MQTT Topics aus dem Home Assistant / MQTT Broker damit die AhoyDTU diesen Wert auch anzeigt.

Die Anforderung ist somit prinzipiell deckungsgleich mit den beiden Features um #606 Tasmota/Shelly Werte per HTTP(S) abzuholen bzw. #431 per Modbus Werte von einem SmartMeter zu lesen, die ich mit #782 verlinkt habe.
Nur eben ohne die Kombination mit dem automatischen Setzen des Power Limit's sondern nur zur Anzeige.

Wollt Ihr die sonstigen hier zusammengetragenen Ideen evtl. in einem eigenen Issue / Feature Request zur Neuorganisation der Settings Seite sammeln ?