lumapu/ahoy #1055: MQTT Additional topics

DejanBukovec, 2023-07-31:
Is possible add additional usefull topics into MQTT HomeAssistant Auto discovery?
For example:

**Active Power Limit:**

[image]

`homeassistant/sensor/PANEL_NAME/ch0_active_PowerLimit/config`

`{    "name":"PANEL_NAME PowerLimit",    "stat_t":"inverter/PANEL_NAME/ch0/active_PowerLimit",    "uniq_id":"INVERTER_SERIAL_ch0_active_PowerLimit",    "unit_of_meas":"%",    "dev":{"name":"PANEL_NAME","ids":"INVERTER_SERIAL","cu":"http://DTU_IP","mf":"Hoymiles","mdl":"PANEL_NAME"} }`

**Restart Inverter:**

[image]

`homeassistant/button/PANEL_NAME/restart_inverter/config`

{    "name":"PANEL_NAME Restart Inverter", "uniq_id":"INVERTER_SERIAL_restart_inverter", "dev_cla":"restart", "ent_cat":"config", "cmd_t":"inverter/ctrl/restart/INVERTER_POSITION", "dev":{"name":"PANEL_NAME","ids":"INVERTER_SERIAL","cu":"http://DTU_IP","mf":"Hoymiles","mdl":"PANEL_NAME"} }

**Turn Off Inverter:**

[image]

`homeassistant/button/PANEL_NAME/turn_inverter_off/config`

`{    "name":"PANEL_NAME Turn Inverter Off",  "uniq_id":"INVERTER_SERIAL_turn_inverter_off", "ic":"mdi:power-plug-off","ent_cat":"config","cmd_t":"inverter/ctrl/power/INVERTER_POSITION","payload_press":"0", "dev":{"name":"PANEL_NAME","ids":"INVERTER_SERIAL","cu":"http://DTU_IP","mf":"Hoymiles","mdl":"PANEL_NAME"} }`

**Turn On Inverter:**

[image]

`homeassistant/button/PANEL_NAME/turn_inverter_on/config`

`{    "name":"PANEL_NAME Turn Inverter On",  "uniq_id":"INVERTER_SERIAL_turn_inverter_on", "ic":"mdi:power-plug","ent_cat":"config","cmd_t":"inverter/ctrl/power/INVERTER_POSITION","payload_press":"1", "dev":{"name":"PANEL_NAME","ids":"INVERTER_SERIAL","cu":"http://DTU_IP","mf":"Hoymiles","mdl":"PANEL_NAME"} }`

**Producing:**

[image]

`homeassistant/binary_sensor/PANEL_NAME/producing/config`

```
{
  "name": "PANEL_NAME Producing",
  "stat_t": "inverter/PANEL_NAME/available",
  "uniq_id": "INVERTER_SERIAL_Producing",
  "value_template": "{{ 1 if value | int == 2 else 0 }}",
  "pl_on": "1",
  "pl_off": "0",
  "dev": {
    "name": "PANEL_NAME",
    "ids": "INVERTER_SERIAL",
    "cu": "http://DTU_IP",
    "mf": "Hoymiles",
    "mdl": "PANEL_NAME"
  }
}
```

**Reachable:**

[image]

`homeassistant/binary_sensor/PANEL_NAME/reachable/config`

```
{
  "name": "PANEL_NAME Reachable",
  "stat_t": "inverter/PANEL_NAME/available",
  "uniq_id": "INVERTER_SERIAL_Reachable",
  "value_template": "{{ 0 if value | int == 0 else 0 }}",
  "pl_on": "1",
  "pl_off": "0",
  "dev": {
    "name": "PANEL_NAME",
    "ids": "INVERTER_SERIAL",
    "cu": "http://DTU_IP",
    "mf": "Hoymiles",
    "mdl": "PANEL_NAME"
  }
}
```

**Limit NonPersistent Relative:**

[image]

`homeassistant/number/PANEL_NAME/ch0_limit_nonpersistent_relative/config`

`{    "name":"PANEL_NAME Limit NonPersistent Relative", "stat_t":"inverter/PANEL_NAME/ch0/active_PowerLimit", "cmd_t":"inverter/ctrl/limit/INVERTER_POSITION",  "uniq_id":"INVERTER_SERIAL_ch0_limit_nonpersistent_relative","unit_of_meas":"%","ic":"mdi:speedometer","min":1,"max":100, "dev":{"name":"PANEL_NAME","ids":"INVERTER_SERIAL","cu":"http://DTU_IP","mf":"Hoymiles","mdl":"PANEL_NAME"} }`

lumapu, 2023-08-16:
I think about that. A frist look into the code says: not that easy but possible 😉

stefan123t, 2024-10-27:
Finally the MQTT Additional Topics issue I have been looking for :see_no_evil: 
I think we should attach all the other MQTT Topics related issues to this one and try to come up with a concise list of all the topics that we want to support via MQTT.

This list should then be cross-checked with the OpenDTU MQTT Topic tree in order to kind of synchronize / harmonize the two projects in terms of what data is sent and how to Home Assistant and other recipients via MQTT.
This would have the added benefit that the any visualizations or other third-party add-ons (e.g. Node-Red flows, etc.) could be made to work with both projects and our users gain a mutual benefit.